lwfinger/rtw88 #304: Injection doesn't work on VIF when AP mode is started on main interface (only affects kernel 6.9+)

i486, 2025-02-06:
**Description**:
When starting the AP mode on the main interface (`wlan0`), creating a virtual interface (`wlan0mon`) in monitor mode, and testing deauthentication injection using tools like `aireplay-ng`, the injection fails to work. The problem began after kernel 6.9.0, as it worked on previous kernel versions.

**Steps to Reproduce**:
1. **Start AP Mode** on `wlan0` interface using `hostapd`. I've used [linux-wifi-hotspot](https://github.com/lakinduakash/linux-wifi-hotspot/) with just No Virt and 2.4GHz checked, which used following configuration:
   ```
   beacon_int=100
   ssid=Test
   interface=wlan0
   driver=nl80211
   channel=1
   ctrl_interface=/tmp/create_ap.wlan0.conf.P66RQgOm/hostapd_ctrl
   ctrl_interface_group=0
   ignore_broadcast_ssid=0
   ap_isolate=0
   hw_mode=g
   wpa=2
   wpa_passphrase=12345678
   wpa_key_mgmt=WPA-PSK
   wpa_pairwise=CCMP
   rsn_pairwise=CCMP   
   ```

2. **Create Virtual Interface**:
   ```
   iw dev wlan0 interface add wlan0mon type monitor addr 12:34:56:78:AB:CD
   ```
3. **Test Injection** using `aireplay-ng`:
   ```
   aireplay-ng -9 wlan0mon
   ```
   Output on 6.8 or earlier:
   ```
   23:58:12  Trying broadcast probe requests...
   23:58:14  xx:xx:xx:xx:xx:xx - channel: 1 - 'xxxxxxx'
   23:58:20   30/30: 100%

   23:58:20  xx:xx:xx:xx:xx:xx - channel: 1 - 'xxxxxxx'
   23:58:21  Ping (min/avg/max): 4.066ms/27.308ms/170.734ms Power: -24.87
   23:58:21  30/30: 100%
   ```

   Output on 6.9 or later:
   ```
   23:58:12  Trying broadcast probe requests...
   23:58:14  xx:xx:xx:xx:xx:xx - channel: 1 - 'xxxxxxx'
   23:58:20   0/30: 0%

   23:58:20  xx:xx:xx:xx:xx:xx - channel: 1 - 'xxxxxxx'
   23:58:21  Ping (min/avg/max): 4.066ms/27.308ms/170.734ms Power: -24.87
   23:58:21  0/30: 0%
   ```

**Expected Behavior**:  
Deauthentication injection should work as expected on the `wlan0mon` interface after starting AP mode on `wlan0`.

**Actual Behavior**:  
Deauthentication injection does not work, even though it previously worked on all kernel versions before 6.9.0.

**Tested Working On These Kernels**:
- `linux-image-6.1.0-kali9-amd64` (6.1.27-1kali1)
- `linux-image-6.1.127-custom` (6.1.127-1)
- `linux-image-unsigned-6.2.0-060200-generic`
- `linux-image-unsigned-6.2.0-060200rc1-generic`
- `linux-image-unsigned-6.3.0-060300-generic`
- `linux-image-unsigned-6.4.0-060400-generic`
- `linux-image-unsigned-6.5.0-060500-generic`
- `linux-image-unsigned-6.6.63-060663-generic`
- `linux-image-unsigned-6.7.10-060710-generic`
- `linux-image-unsigned-6.8.12-060812-generic`
[... 573 lines cut ...]

##### dmesg #############################

[    3.854913] systemd[1]: Starting systemd-timesyncd.service - Network Time Synchronization...
[    4.624738] rtw_8812au 2-6:1.0: Firmware version 52.14.0, H2C version 0
[    5.041123] rtw_8812au 1-10:1.0: Firmware version 52.14.0, H2C version 0
[    8.308919] e1000e 0000:00:1f.6 eth0: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
[ 5552.540415] rtw_8812au 1-10:1.0: Firmware version 52.14.0, H2C version 0 (repeated 4 times)
[ 6487.946491] warning: `aireplay-ng' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
[ 6487.946581] rtw_8812au 1-10:1.0 wlan0mon: entered promiscuous mode
[ 6487.946583] rtw_8812au 1-10:1.0 wlan0mon: entered allmulticast mode
[ 7036.094245] rtw_8812au 1-10:1.0: error beacon valid
[ 7036.094325] rtw_8812au 1-10:1.0: failed to download drv rsvd page

########## wireless info END ############
```
</details>

**Additional Notes**:  

- The issue continues to occur across various tools, including `aireplay-ng`, `mdk`, and `hcxdumptool`.
- There are similar reports from other devices, though I'm uncertain if they're directly related:  
 https://github.com/ZerBea/hcxdumptool/discussions/465
 https://github.com/morrownr/USB-WiFi/issues/390
- I considered bisecting, but it's just not feasible with my current PC setup and I am fairly new to Linux.

dubhater, 2025-02-08:
Please report it to the linux-wireless mailing list.

ZerBea, 2025-02-09:
hcxdumptool won't work as expected if running on a virtual interface.
This is mentioned in "--help":
```
Important recommendation:
-------------------------
Do not set monitor mode by third party tools or third party scripts!
Do not use virtual interfaces (monx, wlanxmon, prismx, ...)!
Do not use virtual machines or emulators!
Do not run other tools that take access to the interface in parallel (except: tshark, wireshark, tcpdump)!
Do not use tools to change the virtual MAC (like macchanger)!
Do not merge (pcapng) dump files, because this destroys assigned hash values!
```

 Exclusive access to the physical interface is mandatory. If these conditions are met, everything is fine:
```
$ uname -r
6.13.1.arch1-1

$ hcxdumptool -l
  0	  3	706655781d9f	dc7014b24c2a	+	wlp4s0          	rtw_8821ce	NETLINK

$ sudo hcxdumptool --rcascan=active
...
81 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
33 PROBERESPONSE(s) captured

exit on sigterm
```

dubhater, 2025-06-12:
@i486 You have a better chance to reach the right people on the mailing list. I'm betting the problem with rtw88 appeared in this commit: https://github.com/torvalds/linux/commit/0a44dfc070749514b804ccac0b1fd38718f7daa1