lwfinger/rtw88 #305: AP mode broken for rtw_8822cu

henkv1, 2025-02-07:
When I put my rtw_8822cu device in AP mode, the connection stops after a while.
I observed the following error in the kernel log:
[ 1612.157086] rtw_8822cu 1-1.3:1.0: unused phy status page (10)
[ 1612.158301] rtw_8822cu 1-1.3:1.0: unused phy status page (10)
[ 1612.159094] rtw_8822cu 1-1.3:1.0: unused phy status page (10)

After some bisecting I found out that this commit caused this behavior:
https://github.com/lwfinger/rtw88/commit/386382ace137f5209e8e83a4cf2e044bf52e8b38 : wifi: rtw88: usb: Prevent TX URB use after free

dubhater, 2025-02-07:
That's strange. I reverted that commit.

briansune, 2025-02-07:
@dubhater 

I also encountered some very puzzling issue after WIFI 7 with 6G band gear updated.

All the WIFI 5 devices are become very bad on TRX rating.

Do you have any knowledge or idea that WIFI 7 could introduce issue on WIFI 5?

All band are separated into individual SSID aka xxx and  xxx_5G and xxx_6G

dubhater, 2025-02-08:
@briansune I'm not aware of wifi 7 causing problems like that for wifi 5 devices. The wifi 7 drivers are still new, maybe they didn't get enough testing.

briansune, 2025-02-08:
@dubhater 

I guess the router or AC need firmware updates to coverups all generation. Most likely people uses WIFI 6(E) or WIFI7 now adays.

There is s.t. messing around on the "OFDMA+MU-MIMO" =[

briansune, 2025-02-08:
@dubhater 

Not sure this will help.

If you still recall that RTL8822BS module had some puzzling 5G STA issue before.

And I compiled the repository that you suggested. I also tested the AP as well.

Modified repository: https://github.com/briansune/dr1-rtl8822bs

During AP test I saw a very interesting behavior.

If AP is working 100% good on 2.4G band it can run up to >100Mbps (direct connect case) and any Bluetooth or 2.4G band devices are all completely dead.

It feels like 2.4G band is undergo a serious of "tug of war". Who wins who can use it.

henkv1, 2025-02-20:
More bisecting pointed to the following commit as the first bad one: https://github.com/lwfinger/rtw88/commit/ce76fa323d563f047f65aad75362e0c40dae1a6d
I don't see how this commit could affect stability, but with this applied the throughput stops after a while. The AP is still visible, but the clients are unable to connect.

I also noticed that this commit is different from the upstream one, which does not seem to have this problem: https://git.kernel.org/pub/scm/linux/kernel/git/wireless/wireless-next.git/commit/drivers/net/wireless/realtek/rtw88?id=4b6652bc6d8d5fb0648b3a7a16ef8af4e0345bcd

Is it possible to sync this repo with upstream?

dubhater, 2025-02-20:
What if you disable the blinking by clearing the trigger? ``echo none > /sys/class/leds/rtw88-*/trigger``

I will sync with upstream at some point.

henkv1, 2025-02-21:
Disabling the blinking seems to work. Thank you.

closed 2025-02-21 by henkv1

Samt43, 2025-03-19:
Hi @henkv1 @dubhater : using latest master 11d29443525eca394e69d8adab065dc33cce23d6, even if the code has been synced with linux kernel, i have the same stability issue as @henkv1 using rtw88 with my 8822bu device (Bus 001 Device 014: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]).
As soon as i don't disable leds using :
`echo none > /sys/class/leds/rtw88-*/trigger`
The hotspot stop working (emitting beacons) after some time, without any logs.
**If i disable the led (that is by the way not blinking at all on my device), the hotspot as never be so stable (no crash for more than two days !!)**, and gives very good performances !!
Thanks a lot for all the work done, this is crazy but i could remove now the old driver from my server, finally this happened :)

Samt43, 2025-03-19:
@dubhater : I played with blink_time values without any sucess (i tried to x10 the values). The led stays black all time. Maybe registers are not correct ?
Using https://github.com/morrownr/88x2bu-20210702, led is not working too...
At a moment i was using one of the out of source tree that had the led working, unfortunatly i can't find which one it was anymore.

Samt43, 2025-03-19:
Maybe this one : https://github.com/cilynx/rtl88x2bu

dubhater, 2025-03-19:
@Samt43 The point of playing with the blink_time values is just to see if AP mode doesn't break anymore.

If you know for sure which driver makes the LED blink, I can look at the code.

Samt43, 2025-03-21:
@dubhater : i confirm that x10 the values of blink_time fix instabilities issue. But do not fix the led.

Samt43, 2025-03-22:
@dubhater : finally, my hotspot stopped emitting beacons today afetr 4 hours, even with values of blink_time x10. The driver does not compile if i disable led support by commenting DCONFIG_RTW88_LEDS.

dubhater, 2025-04-29:
Kernel 6.14 has the LED blinking code. Does it have this problem with the AP mode dying? Does this repository still have the problem? I did sync the code at some point.

dubhater, 2025-07-18:
This may fix the problem with the blinking:

```diff
diff --git a/drivers/net/wireless/realtek/rtw88/led.c b/drivers/net/wireless/realtek/rtw88/led.c
index fc3817b55b27..17364b8af704 100644
--- a/drivers/net/wireless/realtek/rtw88/led.c
+++ b/drivers/net/wireless/realtek/rtw88/led.c
@@ -11,8 +11,12 @@ static int rtw_led_set_blocking(struct led_classdev *led,
 {
 	struct rtw_dev *rtwdev = container_of(led, struct rtw_dev, led_cdev);
 
+	mutex_lock(&rtwdev->mutex);
+
 	rtwdev->chip->ops->led_set(led, brightness);
 
+	mutex_unlock(&rtwdev->mutex);
+
 	return 0;
 }
 ```

Can you test it, please?

henkv1, 2025-07-20:
It looks like the connection is stable now with led blinking enabled. Thank you.