lwfinger/rtw88 #306: rtw_8821ce slow performance

enricotagliavini, 2025-02-08:
Hi there,

I've just got a new laptop and it comes with a 

> 01:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. RTL8821CE 802.11ac PCIe Wireless Network Adapter [10ec:c821]

I soon discovered the poor performance of this adapter on Linux. I tried the rtw_8821ce from your repo, but it has the same issue. I also tried the alt driver, but it doesn't build on kernel 6.12.11-200.fc41.x86_64

Wavemon reports a performance around 50 Mbit/s, but with other laptops I get 600 - 1200. I tried disabling ASPM on the driver, but this panics the kernel, and then in the BIOS (no difference), I also tried to disable msi in the driver, but no difference.

There is not much in the logs, other than the occasional 

`kernel: rtw_8821ce 0000:01:00.0: unhandled firmware c2h interrupt `

Is there any chance I can get a decent speed out of this, or is this hardware just a pile of garbage?

Thank you.

dubhater, 2025-02-08:
You can try this one: https://github.com/tomaspinho/rtl8821ce

It's an older version of the "alt" driver but updated to compile with newer kernels.

enricotagliavini, 2025-02-08:
> You can try this one: https://github.com/tomaspinho/rtl8821ce
> 
> It's an older version of the "alt" driver but updated to compile with newer kernels.

Worth a try. Unfortunately it doesn't work at all, it doesn't associate and then throws kernel stack traces. Thank you for the suggestion.

dubhater, 2025-02-09:
By the way, were you looking at RX or TX in wavemon?

enricotagliavini, 2025-02-09:
Both RX and TX are low and it looks like the RX is correct, the download speed I reach is consistent with that is displayed. I get 1 - 4 MB/s downloading from a local computer also connected to the wireless. When I switch to wire I get 100 MB/s for the same data (which is the speed of the network card).

Wavemon is also useful to see the link quality reported, which is much lower than what it should be.

enricotagliavini, 2025-02-09:
[image]

this is what wavemon looks like. It's also interesting that the reported link quality start bouncing to 100% and back very quickly during a download, and so does the signal level, however the TX and RX stay the same.

enricotagliavini, 2025-02-09:
[image]

This is from my other laptop (with Intel WiFi), from the exact same location

dubhater, 2025-02-09:
Hmm, so you're kind of far from the AP. -82 is a very low signal. wpa_supplicant will constantly trigger scans, looking for networks with better signal strength. You should get more speed if you turn off the bg_scan thing in wpa_supplicant.

The Intel card is faster because apparently it has better reception, but also because it's more advanced: wifi 6 versus 5, 160 MHz channel width versus 80 MHz, 2 spatial streams versus 1. The maximum real speed you can get (in a browser, wget, torrents, etc) with RTL8821CE is ~300 Mbps.

What is the signal strength when you move the Realtek card close to the AP?

> It's also interesting that the reported link quality start bouncing to 100% and back very quickly during a download, and so does the signal level, however the TX and RX stay the same.

I think that will be fixed when you update to kernel 6.13 or newer.

enricotagliavini, 2025-02-10:
[image]
This is what wavemon reports when I'm 25-30 cm from the AP. The signal is now -52 dBm, which is quite low for such a short distance. So close it could be -40 or even in the -30 dBm, no? I get around 100 TX and 160 RX Mbit/s tested with a speed test.

It could also be the placement of the antenna in this laptop (BMAX Y14) is not optimal or has a small gain. With that being said, even when I'm far away (well I'm not that far away, but there is a wall in between unfortunately), -80 vs. -70 still feels quite a lot, it's 10 times less power in RX.

Worst case, if this is totally an HW fault, I can try swap the wireless card for a different model and see what happens. If there is anything I can try, please let me know.

Thank you for taking the time to answer me, I really appreciate it.

And about kernel 6.13: good to know, thank you! It's around the corner and will get into Fedora soon.

dubhater, 2025-02-10:
Yes, I would expect -40..-30 at that distance. The antenna is connected, right?

If other drivers don't work, can you try Windows?

enricotagliavini, 2025-02-11:
I'll have to open up the laptop to check the antenna config. I can also try to swap the card for another brand / model, if it happens to be the same form factor and interface of a spare one I have.

As for Windows I'd rather try ndiswrapper (just kidding of course). I have to see, wiping the SSD and reinstalling twice is a long task, especially with the WiFi being slow. If I can find a spare NVME drive somewhere it would save time. Let's see if I'm lucky.

enricotagliavini, 2025-02-13:
I found an old spare NVME disk at work, so no need to shrink. The manufacturer of the laptop, BMAX, was also very kind and sent me a picture of what the antenna configuration should look like, so I have a reference. This will probably take me a while, please keep the issue open ^_^.

enricotagliavini, 2025-02-14:
[image]
Well in the end I managed to find some time now. I opened the laptop and I can say the antenna configuration is fine, there is no hardware damage.

I can also say I found out how to get a better signal after checking the layout. Due to the way the antennas are placed I can get a better signal if the laptop is oriented in a certain way. Awkward to be fair, but it does the job for the test. At this angle at 30 cm distance from the AP I get max signal strength, but I still get only ~260 Mbit/s reported by wavemon and 120 - 170 Mbit/s speed tested with a benchmark.

Obviously this speed is good enough at this distance, but it still doesn't sound as fast as it should be, or is it?

Is there anything I can do to improve the link quality a bit, other than disabling the bgscan as you mentioned?

dubhater, 2025-02-15:
Good to know the antenna is okay. It should be faster. You should see VHT-MCS 9.

It's probably a driver problem. There are at least 4 types of RTL8821CE cards in the wild, but the team at Realtek who wrote rtw88 I think only had one of them.

I'm curious what results you get with Windows.

enricotagliavini, 2025-03-02:
[image]
So this was so much work, you have no idea. Windows is so hard to install. I had to install it via a VM on a USB attached disk adapter, then boot it, complete the installation and install all the drivers, all while Windows was happily crashing or restarting every less than 5 minutes. But I digress...

Once I got it working I could see that the wireless signal, from the same position where I get ~-80 dBm from Linux, is much stronger. It reports -64 dBm. When up close to the AP there is no difference in the dBm reported and also the speed test, using Ookla, so online, gives the same result.

enricotagliavini, 2025-03-02:
I opened a bug upstream: https://bugzilla.kernel.org/show_bug.cgi?id=219830

enricotagliavini, 2025-03-02:
Now that I started using the laptop for daily tasks (e.g. watching videos and so on), I also noticed that these messages are popping up periodically in dmesg:

- rtw_8821ce 0000:01:00.0: failed to send h2c command
- rtw_8821ce 0000:01:00.0: firmware failed to leave lps state

they seem to only trigger if I actually make some traffic happen.

BTW I also switched to kernel 6.13 and indeed the wireless signal strength is now reported always correctly in wavemon, there is no more jumping to 100% randomly or the like.

dubhater, 2025-03-02:
rtw88 reports the signal strength pretty much as it comes from the hardware, whereas the Windows driver processes the number a bit, so I'm not sure we can compare them. We should compare the speed test numbers. How fast is the Windows driver in that position where you have low signal strength?

You could put this in ``/etc/modprobe.d/no-lps-deep.conf``:

```
options rtw_core disable_lps_deep=1
options rtw88_core disable_lps_deep=1
```

Maybe it will help with those messages in dmesg.

Neo-29, 2025-03-21:
Can confirm the same issues on archlinux
Only kernel ok is 6.13.5.
Above are not.
Hope a fix soon.

vozoid, 2025-04-19:
Same issue on linux mint
Current kernel is 6.8.0-58-generic

EvilBoi123, 2025-11-04:
can confirm on my side as well on archlinux 6.17.7-arch1-1