lwfinger/rtw88 #307: rtw_8821au slows down after 15 minutes

giopera, 2025-02-08:
While I'm connected to a network in the first minutes i have 1/3 minutes with low ping and rates that are similar to the ones i have on another computer but after those minutes the ping goes to ~1000ms and 100 kB/s rates.
When i disconnect and reconnect the speed goes back up to the normal one.
Here a video with a demonstration of what happens (my normal speed rate is 2/3 MB/s).

https://github.com/user-attachments/assets/822b5213-3412-4e76-be65-8f4c0301a042

When i opened journalctl I noticed that every 2/3 seconds there is wpa_supplicant and i don't know if it could be a part of the issue:
```
feb 08 21:21:34 PC-Desktop wpa_supplicant[582]: wlp0s26u1u3: CTRL-EVENT-BEACON-LOSS
feb 08 21:21:35 PC-Desktop wpa_supplicant[582]: wlp0s26u1u3: CTRL-EVENT-BEACON-LOSS
feb 08 21:21:36 PC-Desktop wpa_supplicant[582]: wlp0s26u1u3: CTRL-EVENT-BEACON-LOSS
feb 08 21:21:39 PC-Desktop wpa_supplicant[582]: wlp0s26u1u3: CTRL-EVENT-BEACON-LOSS
feb 08 21:21:59 PC-Desktop wpa_supplicant[582]: wlp0s26u1u3: CTRL-EVENT-BEACON-LOSS
feb 08 21:22:08 PC-Desktop wpa_supplicant[582]: wlp0s26u1u3: CTRL-EVENT-BEACON-LOSS
feb 08 21:22:09 PC-Desktop wpa_supplicant[582]: wlp0s26u1u3: CTRL-EVENT-BEACON-LOSS
feb 08 21:22:10 PC-Desktop wpa_supplicant[582]: wlp0s26u1u3: CTRL-EVENT-BEACON-LOSS
feb 08 21:22:11 PC-Desktop wpa_supplicant[582]: wlp0s26u1u3: CTRL-EVENT-BEACON-LOSS
```

This is my USB dongle info from lshw, it is a TP-Link Archer T2U v3 Wireless USB Dual Band AC600:
```
description: Generic USB device
product: 802.11ac WLAN Adapter
vendor: Realtek
physical id: 3
bus info: usb@4:1.3
version: 2.00
serial: 00e04c000001
capabilities: usb-2.10
configuration: driver=rtw_8821au maxpower=500mA speed=480Mbit/s
```

Currently I'm using Arch Linux x86_64 and Linux 6.13.1-arch1-1 with the driver installed from the AUR.

dubhater, 2025-02-08:
Is your network in the 2.4 GHz band? Do you have Bluetooth speakers or headphones nearby?

giopera, 2025-02-09:
My network is in both 2.4 GHz and 5 GHz but the adapter picks up only the 2.4 GHz Band, i have a pair of Bluetooth headphones which i use daily to listen to videos while i program.

dubhater, 2025-02-09:
Unfortunately this chip has a problem with the Bluetooth interference. I don't know what you can do about it other than switching to the 5 GHz band.

Give the two networks different names, then you can more easily select the 5 GHz one.

closed 2025-05-03 by dubhater