lwfinger/rtw88 #308: doesn't work on Jetson Orin NX

handshow888, 2025-02-09:
Environment:
```bash
nvidia@orin-210:~$ uname -r
5.15.148-tegra
nvidia@orin-210:~$ uname -a
Linux orin-210 5.15.148-tegra #1 SMP PREEMPT Tue Jan 7 17:14:38 PST 2025 aarch64 aarch64 aarch64 GNU/Linux
nvidia@orin-210:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.5 LTS
Release:	22.04
Codename:	jammy
nvidia@orin-210:~$ lsusb
Bus 002 Device 002: ID 0451:8240 Texas Instruments, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0bda:c82c Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 002: ID 0451:8242 Texas Instruments, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
I used NVIDIA SDKManager to flash JetPack 6.2 for my device.
`ID 0bda:c82c` I think my wifi module is RTL8822CU, and this driver should be suitable for my device, but it is not working properly.

after these steps, I still can't view my WiFi network card through `ifconfig`
```bash
git clone https://github.com/lwfinger/rtw88
cd rtw88
make
sudo make install
sudo make install_fw
```

Here are some output by `sudo dmesg |grep rtw`
```bash
[   11.015240] rtw_core: disagrees about version of symbol ieee80211_connection_loss
[   11.015248] rtw_core: Unknown symbol ieee80211_connection_loss (err -22)
[   11.015253] rtw_core: disagrees about version of symbol ieee80211_report_wowlan_wakeup
[   11.015254] rtw_core: Unknown symbol ieee80211_report_wowlan_wakeup (err -22)
[   11.015268] rtw_core: disagrees about version of symbol ieee80211_txq_get_depth
[   11.015269] rtw_core: Unknown symbol ieee80211_txq_get_depth (err -22)
[   11.015306] rtw_core: disagrees about version of symbol ieee80211_probereq_get
[   11.015307] rtw_core: Unknown symbol ieee80211_probereq_get (err -22)
[   11.015314] rtw_core: disagrees about version of symbol ieee80211_iterate_stations_atomic
[   11.015315] rtw_core: Unknown symbol ieee80211_iterate_stations_atomic (err -22)
[   11.015317] rtw_core: disagrees about version of symbol ieee80211_start_tx_ba_session
[   11.015318] rtw_core: Unknown symbol ieee80211_start_tx_ba_session (err -22)
[   11.015408] rtw_core: disagrees about version of symbol regulatory_hint
[   11.015410] rtw_core: Unknown symbol regulatory_hint (err -22)
[   11.015422] rtw_core: disagrees about version of symbol ieee80211_nullfunc_get
[   11.015422] rtw_core: Unknown symbol ieee80211_nullfunc_get (err -22)
[   11.015428] rtw_core: disagrees about version of symbol ieee80211_pspoll_get
[   11.015429] rtw_core: Unknown symbol ieee80211_pspoll_get (err -22)
[   11.015436] rtw_core: disagrees about version of symbol ieee80211_register_hw
[   11.015437] rtw_core: Unknown symbol ieee80211_register_hw (err -22)
[   11.015443] rtw_core: disagrees about version of symbol __ieee80211_create_tpt_led_trigger
[   11.015444] rtw_core: Unknown symbol __ieee80211_create_tpt_led_trigger (err -22)
[   11.015449] rtw_core: disagrees about version of symbol ieee80211_restart_hw
[   11.015450] rtw_core: Unknown symbol ieee80211_restart_hw (err -22)
[   11.015472] rtw_core: disagrees about version of symbol cfg80211_calculate_bitrate
[   11.015473] rtw_core: Unknown symbol cfg80211_calculate_bitrate (err -22)
[   11.015476] rtw_core: disagrees about version of symbol ieee80211_find_sta
[   11.015477] rtw_core: Unknown symbol ieee80211_find_sta (err -22)
[   11.015479] rtw_core: disagrees about version of symbol ieee80211_iter_keys
[   11.015479] rtw_core: Unknown symbol ieee80211_iter_keys (err -22)
[   11.015484] rtw_core: disagrees about version of symbol ieee80211_tx_status_irqsafe
[   11.015484] rtw_core: Unknown symbol ieee80211_tx_status_irqsafe (err -22)
[   11.015497] rtw_core: disagrees about version of symbol ieee80211_stop_tx_ba_cb_irqsafe
[   11.015498] rtw_core: Unknown symbol ieee80211_stop_tx_ba_cb_irqsafe (err -22)
[   11.015503] rtw_core: disagrees about version of symbol wiphy_to_ieee80211_hw
[   11.015504] rtw_core: Unknown symbol wiphy_to_ieee80211_hw (err -22)
[   11.015508] rtw_core: disagrees about version of symbol ieee80211_queue_delayed_work
[... 211 lines cut ...]
[  210.749167] rtw_core: disagrees about version of symbol ieee80211_iter_keys_rcu
[  210.749168] rtw_core: Unknown symbol ieee80211_iter_keys_rcu (err -22)
[  210.749174] rtw_core: disagrees about version of symbol ieee80211_free_txskb
[  210.749175] rtw_core: Unknown symbol ieee80211_free_txskb (err -22)
[  210.749192] rtw_core: disagrees about version of symbol ieee80211_tx_dequeue
[  210.749193] rtw_core: Unknown symbol ieee80211_tx_dequeue (err -22)
[  210.749207] rtw_core: disagrees about version of symbol ieee80211_stop_queues
[  210.749209] rtw_core: Unknown symbol ieee80211_stop_queues (err -22)
[  210.749225] rtw_core: disagrees about version of symbol ieee80211_scan_completed
[  210.749226] rtw_core: Unknown symbol ieee80211_scan_completed (err -22)
[  210.749232] rtw_core: disagrees about version of symbol ieee80211_iterate_active_interfaces_atomic
[  210.749233] rtw_core: Unknown symbol ieee80211_iterate_active_interfaces_atomic (err -22)
[  210.749247] rtw_core: disagrees about version of symbol ieee80211_unregister_hw
[  210.749248] rtw_core: Unknown symbol ieee80211_unregister_hw (err -22)
[  210.749260] rtw_core: disagrees about version of symbol ieee80211_beacon_get_tim
[  210.749261] rtw_core: Unknown symbol ieee80211_beacon_get_tim (err -22)
[  210.749269] rtw_core: disagrees about version of symbol ieee80211_channel_to_freq_khz
[  210.749270] rtw_core: Unknown symbol ieee80211_channel_to_freq_khz (err -22)
[  210.749279] rtw_core: disagrees about version of symbol ieee80211_cqm_rssi_notify
[  210.749281] rtw_core: Unknown symbol ieee80211_cqm_rssi_notify (err -22)
[  210.749290] rtw_core: disagrees about version of symbol ieee80211_request_smps
[  210.749292] rtw_core: Unknown symbol ieee80211_request_smps (err -22)
[  210.749296] rtw_core: disagrees about version of symbol ieee80211_queue_work
[  210.749297] rtw_core: Unknown symbol ieee80211_queue_work (err -22)
```

handshow888, 2025-02-09:
Secure boot if off
All prerequisites have been installed, except `linux-headers-$(uname -r)` , but no errors were encountered during the installation process.
```bash
sudo apt install linux-headers-$(uname -r) build-essential git
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-headers-5.15.148-tegra
E: Couldn't find any package by glob 'linux-headers-5.15.148-tegra'
```
Here are some supplements and attempts
```bash
nvidia@orin-210:~$ sudo modprobe -r rtw_8822cu
nvidia@orin-210:~$ sudo modprobe -r rtw_core
nvidia@orin-210:~$ sudo modprobe rtw_8822cu
modprobe: ERROR: could not insert 'rtw_8822cu': Invalid argument
```
```bash
nvidia@orin-210:~$ sudo modprobe cfg80211
nvidia@orin-210:~$ sudo modprobe rtw_core
modprobe: ERROR: could not insert 'rtw_core': Invalid argument
nvidia@orin-210:~$ sudo modprobe rtw_8822cu
modprobe: ERROR: could not insert 'rtw_8822cu': Invalid argument
```
```bash
sudo insmod rtw_8822cu.ko 
insmod: ERROR: could not insert module rtw_8822cu.ko: Unknown symbol in module
```
```bash
nvidia@orin-210:~$ grep cfg80211 /proc/kallsyms > cfg80211.txt
nvidia@orin-210:~$ grep ieee80211 /proc/kallsyms > ieee80211.txt
```
It can be seen that the symbols mentioned in `dmesg` do exist
[cfg80211.txt](https://github.com/user-attachments/files/18724335/cfg80211.txt)
[ieee80211.txt](https://github.com/user-attachments/files/18724337/ieee80211.txt)

dubhater, 2025-02-09:
You need the correct headers for the kernel. I don't know how to acquire them in your case. Check NVidia/Jetson/Jetpack/ forums.

handshow888, 2025-02-10:
> You need the correct headers for the kernel. I don't know how to acquire them in your case. Check NVidia/Jetson/Jetpack/ forums.

Thank you for your response. It seems that the kernel header has already been installed before installing rtw88, and its name is different from other Ubuntu systems. If I have made any mistakes, please feel free to correct me.
```bash
# List the Module.symvers file in the build directory for the current kernel version.
# This checks if the Module.symvers file exists and is accessible at the expected location.
nvidia@orin-210:~/environment/rtw88$ ls /lib/modules/$(uname -r)/build/Module.symvers
/lib/modules/5.15.148-tegra/build/Module.symvers

# Search installed packages for 'nvidia-l4t-kernel-headers' to verify its installation and version.
# This command uses dpkg to list all installed packages and filters the output to show only those containing 'nvidia-l4t-kernel-headers'.
nvidia@orin-210:~/environment/rtw88$ dpkg -l | grep nvidia-l4t-kernel-headers
ii  nvidia-l4t-kernel-headers                  5.15.148-tegra-36.4.3-20250107174145        arm64        NVIDIA Linux Tegra Kernel Headers Package

# List details about the symbolic link that points from the kernel's build directory to the actual headers path.
# It shows the symbolic link information, confirming where the build directory points to.
nvidia@orin-210:~/environment/rtw88$ ls -l /lib/modules/$(uname -r)/build
lrwxrwxrwx 1 root root 102  1月  8 09:41 /lib/modules/5.15.148-tegra/build -> /usr/src/linux-headers-5.15.148-tegra-ubuntu22.04_aarch64/3rdparty/canonical/linux-jammy/kernel-source

# List all files and directories in the current directory.
nvidia@orin-210:/usr/src/linux-headers-5.15.148-tegra-ubuntu22.04_aarch64/3rdparty/canonical/linux-jammy/kernel-source$ ls
arch    debian         include   Kbuild   Makefile        samples   tools
block   Documentation  init      Kconfig  mm              scripts   ubuntu
certs   drivers        io_uring  kernel   Module.symvers  security  usr
crypto  fs             ipc       lib      net             sound     virt
```

dubhater, 2025-02-10:
I still don't know why the symbol versions don't match.

briansune, 2025-02-11:
@dubhater 

Just purely guessing.

But I do see this before.

If you use Makefile with make + module.

The current Makefile is missing

> export CONFIG_RTW88 = m

So this will create all kinds of issue if people are not modifying the kernel Kconfig and Makefile.

And using cross-compile outside the kernel build.

handshow888, 2025-02-12:
> [@dubhater](https://github.com/dubhater)
>
> Just purely guessing.
>
> But I do see this before.
>
> If you use Makefile with make + module.
>
> The current Makefile is missing
>
>
> So this will create all kinds [...]

Thank you for your reply. I tried to add this line in the Makefile of rtw88, but it still doesn't work. :(

briansune, 2025-02-12:
@handshow888 

Adding that line isn't enough if you didn't complete the story and also in the correct place of the Makefile

You got to make with  module from first place on RTW88 with:

all: module
.....

And the trunk of the kernel itself is either on [y] or [m] with necessary export symbols with those ieee packages etc (assume they didn't modify the kernel).

So complicated my friend you go to ask Jetson to guide you more on how to compile the out-of-tree kernel module.

briansune, 2025-02-12:
@handshow888 

Long story short:

https://stackoverflow.com/questions/33734133/disagrees-about-version-of-symbol-symbol-name-after-insmod

It sound like you are using a different kernel version and compiling the out-of-tree RTW88 with another kernel version.

a5a5aa555oo, 2025-02-13:
[Another user reported the same problem yesterday](https://github.com/a5a5aa555oo/rtw89/issues/2), I guess... this is caused by the mistakes in Makefile... maybe..

Does it work if this patch is applied?

```patch
diff --git a/Makefile b/Makefile
index 8a606e2..99bb595 100644
--- a/Makefile
+++ b/Makefile
@@ -1,6 +1,7 @@
-SHELL := /bin/sh
-KVER ?= $(if $(KERNELRELEASE),$(KERNELRELEASE),$(shell uname -r))
-KSRC ?= $(if $(KERNEL_SRC),$(KERNEL_SRC),/lib/modules/$(KVER)/build)
+SHELL := /bin/bash
+KVER ?= $(shell uname -r)
+KSRC ?= /lib/modules/$(KVER)/build
+PWD := $(shell pwd)
 FWDIR := /lib/firmware/rtw88
 MODLIST := rtw_8723cs rtw_8723de rtw_8723ds rtw_8723du \
           rtw_8812au rtw_8814ae rtw_8814au rtw_8821au rtw_8821ce rtw_8821cs rtw_8821cu \
@@ -185,7 +186,7 @@ rtw_usb-objs        := usb.o
 ccflags-y += -D__CHECK_ENDIAN__
 
 all: 
-       $(MAKE) -j`nproc` -C $(KSRC) M=$$PWD modules
+       $(MAKE) -j`nproc` -C $(KSRC) M=$(PWD) modules
        
 install: all
        @install -D -m 644 -t $(MODDESTDIR) *.ko
@@ -232,7 +233,7 @@ uninstall:
        @echo "The rtw88 drivers were removed successfully."
 
 clean:
-       $(MAKE) -C $(KSRC) M=$$PWD clean
+       $(MAKE) -C $(KSRC) M=$(PWD) clean
        @rm -f MOK.*
 
 sign:
```

briansune, 2025-02-13:
@a5a5aa555oo 

If I under the make stuff correctly (high wrong from first place). W/O the below line

> export CONFIG_xxxxx = m 

will generate all kinds of out-of-tree compiled kernel module issue aka the some symbol stuffs I don't know what SW guy calls it.

I only know how to make things work I didn't have any education on these kind of stuffs my deep sorry.

I suggest you see the Makefile under some of my modified driver.

https://github.com/briansune/dr1-rtl8822bs/blob/main/Makefile

handshow888, 2025-02-13:
@briansune 
As you said, those steps are too complicated. So I made an attempt, flashing the device's system again, flashing JetPack 6.1, and **the driver worked normally** (without making any modifications to rtw88)

Although JetPack 6.1 and JetPack 6.2 (latest version) have the same version number of the system kernel (5.15.148-tegra), compiling and installing RTW88 directly in the 6.1 environment can work properly, but errors will be encountered in the 6.2 environment. I don't know what subtle differences there are

handshow888, 2025-02-13:
@a5a5aa555oo 

> [@briansune](https://github.com/briansune) As you said, those steps are too complicated. So I made an attempt, flashing the device's system again, flashing JetPack 6.1, and **the driver worked [...]

I rewrote my device using JetPack 6.1 and directly compiled rtw88 (without any modifications), and the driver worked properly.

It feels a bit like avoiding the problem, but I did spend a lot of time trying to install the wifi driver. And after the driver worked normally, I installed other work environment dependencies and used the device for my work.

In about a week, I will have a second device of the same type, and at that time, I will try the modification plan you provided. I feel that it is highly likely to be useful in the JetPack 6.2 environment. Thank you for your reply.

briansune, 2025-02-13:
@handshow888 

Why the internal RTW88 works is because the make process is completely different.

If you read the trunk of the kernel ./driver/net/wireless/..../rtw88

Makefile and Kconfig you will understand how to convert it to a in-tree-driver.

I suggest this conversion method, all you need is remove the Makefile from this repository replace it will the RTW88 Make and Kconfig 

Make sure the C files -o is correctly cloned w/o missing any package.

And also do not mix up out-off-tree kernel module and try remove version kernel from first place.

So many point to fail. Good luck. My friend.

Again <-

You can use a kernel module on 6.2 when your rootfs is 6.1 or even different compiled rootfs 6.1 this create all headache from first place.

Make sure you only doing things on the same Linux trunk, too easy to make mistake.

handshow888, 2025-02-20:
@briansune 

Thank you for your detailed guidance. To clarify, the system image and JetPack I am using are officially provided by NVIDIA as a bundled package. This means that both the kernel and the root filesystem, along with other components, are consistent and validated by NVIDIA. Therefore, there is no mixing of versions like 6.1 and 6.2; everything comes from the same official release.

handshow888, 2025-02-20:
@briansune @a5a5aa555oo 
Something magical happened. I flashed the system and JetPack6.2 on another identical device, pulled the latest code from the RTW88 repository, and successfully installed it directly. Now I can use WiFi and Bluetooth on the new device. I didn't even make any modifications to the Makefile and kernel.

The two devices only have different production batches, but their hardware is the same. The system image and JetPack installation package were both downloaded locally a long time ago, using the exact same image. I tried reinstalling the system several times when using my old device a few weeks ago, but it didn't work.

briansune, 2025-02-20:
@handshow888 

Didn't play with too much on Jetson it is my colleague work on image recognition.

But good to know it is now working.

I guess you also learn new thing from this problem =]

Close issue to enjoy~

closed 2025-02-21 by handshow888

6thday, 2025-03-11:
Jetson Orin Nano Super + Jetpack 6.2 b77 + RTL8922AE
Not working.....I use 6.6-lts and did clean modules and  cleanup_target_system before install instructions 

keihofung@C000:~$ uname -r
5.15.148-tegra
keihofung@C000:~$ uname -a
Linux C000 5.15.148-tegra #1 SMP PREEMPT Tue Jan 7 17:14:38 PST 2025 aarch64 aarch64 aarch64 GNU/Linux
keihofung@C000:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.5 LTS
Release:	22.04
Codename:	jammy
keihofung@C000:~$ lsusb
Bus 002 Device 002: ID 0bda:0489 Realtek Semiconductor Corp. 4-Port USB 3.0 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0bda:8922 Realtek Semiconductor Corp. Bluetooth Radio
Bus 001 Device 002: ID 0bda:5489 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
keihofung@C000:~$ sudo dmesg |grep rtw

0bda:8922 recongize as bluetooth and dmesg return nothing
thank you!

a5a5aa555oo, 2025-03-11:
Hi @6thday 

[I'm the owner of the a5a5aa555oo/rtw89 repo](https://github.com/a5a5aa555oo/rtw89). RTL8922AE is not driven by the driver here, 
so I think this is not the right place to discuss this issue. [I have replied your question here](https://github.com/a5a5aa555oo/rtw89/issues/2#issuecomment-2714682471).