lwfinger/rtw88 #309: "Newest" now deprecated rtl8188eus repo points here, but this doesn't have the chipset yet

Anonymous-User47, 2025-02-10:
Forgive me if this is the wrong 'place', but the (formerly) most updated RTL8188EUS repo points here but but this doesn't have the chipset yet: https://github.com/aircrack-ng/rtl8188eus?tab=readme-ov-file#these-drivers-is-deprecated

The other most 'popular' repo is https://github.com/lwfinger/rtl8188eu, of which its last commit was Jul 29, 2023 by the now dead developer https://github.com/lwfinger/rtl8188eu/issues/458

Was this some type of miscoordination between you two or what?

morrownr, 2025-02-11:
> Was this some type of miscoordination between you two or what?

Which `you two` are you talking about?

I can recommend the following site:

https://github.com/morrownr/USB-WiFi

Scroll down and read menu items 1 and 2.

The Plug and Play List shows many good adapters and you don't have to run around looking for a driver.

Mastiffen, 2025-04-13:
@morrownr I would assume he means this repository and Aircrack. I ran into the same snag for an 8188eu card, and for me it's not really easy to replace the card. I have the USB wifi glued in place in my boat, as the AP for ESP32's and other stuff. You may or may not remember that I am using an Asus 8814 based card as the link to my cabin, but the 8188eu card where I had some signal issues. But this is something else, I do not get the card up at all. Does that mean that it is not supported, or is there a change I need to do somewhere?

morrownr, 2025-04-23:
> Does that mean that it is not supported, or is there a change I need to do somewhere?

There is no support for the rtl8188eus in this rtw88 repo unless I have totally overlooked it. I'm not sure why any site would be pointing to this repo. There may be a rtl8188eus repo somewhere here on this github site but the only repo that I pay attention to is this one.