lwfinger/rtw88 #310: [BUG] RTW_8814AU FAILED TO WORK ON USB 2.0 EXTENSION CABLE

D357R0Y3R, 2025-02-11:
I HAVE TRIED TESTING THE RTW_8814AU WITH AND WITHOUT AN USB 2.0 EXTENSION CABLE. NORMALLY THE DRIVER IS WORKING AS EXPECTED WHEN CONNECTED ON BOTH USB 2.0 AND USB 3.0 PORTS. BUT WHEN I TRIED TO CONNECT IT VIA AN USB 2.0 EXTENSION CABLE ON USB 2.0 AND USB 3.0 PORTS NONE OF THEM WORKS.

```
[ 1039.208296] usb 1-2: new high-speed USB device number 19 using xhci_hcd
[ 1039.332067] usb 1-2: New USB device found, idVendor=0bda, idProduct=8813, bcdDevice= 0.00
[ 1039.332074] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1039.332076] usb 1-2: Product: 802.11ac NIC
[ 1039.332078] usb 1-2: Manufacturer: Realtek
[ 1039.332080] usb 1-2: SerialNumber: 123456
[ 1039.337922] rtw_8814au 1-2:1.0: Firmware version 33.6.0, H2C version 6
[ 1039.443388] usb 1-2: USB disconnect, device number 19
[ 1039.777719] usb 2-2: new SuperSpeed USB device number 6 using xhci_hcd
[ 1039.788672] usb 2-2: Int endpoint with wBytesPerInterval of 512 in config 1 interface 0 altsetting 0 ep 133: setting to 64
[ 1039.788929] usb 2-2: New USB device found, idVendor=0bda, idProduct=8813, bcdDevice= 0.00
[ 1039.788936] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1039.788938] usb 2-2: Product: 802.11ac NIC
[ 1039.788941] usb 2-2: Manufacturer: Realtek
[ 1039.788943] usb 2-2: SerialNumber: 123456
[ 1039.792539] rtw_8814au 2-2:1.0: Firmware version 33.6.0, H2C version 6
[ 1053.085825] usb 2-2: USB disconnect, device number 6
[ 1054.064362] xhci_hcd 0000:00:14.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[ 1065.091020] usb 1-2: new high-speed USB device number 20 using xhci_hcd
[ 1065.213935] usb 1-2: New USB device found, idVendor=0bda, idProduct=8813, bcdDevice= 0.00
[ 1065.213941] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1065.213943] usb 1-2: Product: 802.11ac NIC
[ 1065.213945] usb 1-2: Manufacturer: Realtek
[ 1065.213947] usb 1-2: SerialNumber: 123456
[ 1065.220364] rtw_8814au 1-2:1.0: Firmware version 33.6.0, H2C version 6
[ 1065.327212] usb 1-2: USB disconnect, device number 20
[ 1065.677981] usb 1-2: new high-speed USB device number 21 using xhci_hcd
[ 1065.801239] usb 1-2: New USB device found, idVendor=0bda, idProduct=8813, bcdDevice= 0.00
[ 1065.801246] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1065.801248] usb 1-2: Product: 802.11ac NIC
[ 1065.801250] usb 1-2: Manufacturer: Realtek
[ 1065.801251] usb 1-2: SerialNumber: 123456
[ 1065.804608] rtw_8814au 1-2:1.0: Firmware version 33.6.0, H2C version 6
[ 1066.575093] rtw_8814au 1-2:1.0: mac power on failed
[ 1066.575098] rtw_8814au 1-2:1.0: failed to power on mac
[ 1066.589302] rtw_8814au 1-2:1.0: mac power on failed
[ 1066.589308] rtw_8814au 1-2:1.0: failed to power on mac
[ 1066.602318] rtw_8814au 1-2:1.0: mac power on failed
[ 1066.602323] rtw_8814au 1-2:1.0: failed to power on mac
[ 1077.243566] rtw_8814au 1-2:1.0: mac power on failed
[ 1077.243571] rtw_8814au 1-2:1.0: failed to power on mac
[ 1077.256390] rtw_8814au 1-2:1.0: mac power on failed
[ 1077.256395] rtw_8814au 1-2:1.0: failed to power on mac
[ 1087.241261] rtw_8814au 1-2:1.0: mac power on failed
[ 1087.241267] rtw_8814au 1-2:1.0: failed to power on mac
[ 1087.254608] rtw_8814au 1-2:1.0: mac power on failed
[ 1087.254613] rtw_8814au 1-2:1.0: failed to power on mac
[ 1094.006888] usb 1-2: USB disconnect, device number 21
```

dubhater, 2025-02-11:
Have you tried a different USB 2.0 extension cable?

Izodn, 2025-02-16:
I just tested my 8814AU with my own USB extension cable.

As far as I can see, the cable is indeed a USB 2 extension cable and it does totally work.

```
root@nix:~# lsusb | grep 8814AU
Bus 001 Device 013: ID 0bda:8813 Realtek Semiconductor Corp. RTL8814AU 802.11a/b/g/n/ac Wireless Adapter

root@nix:~# lsusb -D /dev/bus/usb/001/013
Device: ID 0bda:8813 Realtek Semiconductor Corp. RTL8814AU 802.11a/b/g/n/ac Wireless Adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x8813 RTL8814AU 802.11a/b/g/n/ac Wireless Adapter
  bcdDevice            0.00
  iManufacturer           1 Realtek
  iProduct                2 802.11ac NIC
  iSerial                 3 123456
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0035
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0016
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000002
      HIRD Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat        1023 micro seconds
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```

My recommendation is that you attempt to use a different cable and/or port.

MR-Diamond, 2025-02-16:
I regularly use my wifi adapter TP-Link Archer T9UH v1 [Realtek RTL8814AU] with an extension cable (USB 3.0); for verifying, I also tried another cable, USB 2.0: all fine.