lwfinger/rtw88 #311: RTL8822CE hot plug PCIe support?

briansune, 2025-02-12:
@dubhater 

I finally fixed the PCIe slot not detecting some PCIe cards (I guess many before the HW modifications).

So I see a hot plug error on dmesg.

```
01:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device c822
        Subsystem: Realtek Semiconductor Co., Ltd. Device c822
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 50
        Region 0: I/O ports at <unassigned> [disabled]
        Region 2: Memory at e0000000 (64-bit, non-prefetchable) [size=64K]
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=375mA PME(D0+,D1+,D2+,D3hot+,D3cold+)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
                Address: 00000000fd480000  Data: 0000
        Capabilities: [70] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 128 bytes, PhantFunc 0, Latency L0s <4us, L1 <64us
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 0.000W
                DevCtl: Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop-
                        MaxPayload 128 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr+ TransPend-
                LnkCap: Port #0, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <4us, L1 <64us
                        ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
                LnkCtl: ASPM Disabled; RCB 64 bytes Disabled- CommClk-
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Not Supported, TimeoutDis+, LTR+, OBFF Via message/WAKE#
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
                LnkCtl2: Target Link Speed: 2.5GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance De-emphasis: -6dB
                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-
                         EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
        Capabilities: [100 v2] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
                AERCap: First Error Pointer: 00, GenCap+ CGenEn- ChkCap+ ChkEn-
        Capabilities: [148 v1] Device Serial Number 00-e0-4c-ff-fe-c8-22-01
        Capabilities: [158 v1] Latency Tolerance Reporting
                Max snoop latency: 0ns
                Max no snoop latency: 0ns
        Capabilities: [160 v1] L1 PM Substates
                L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
                          PortCommonModeRestoreTime=30us PortTPowerOnTime=60us
                L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
                           T_CommonMode=0us LTR1.2_Threshold=0ns
                L1SubCtl2: T_PwrOn=10us
        Kernel driver in use: rtw_8822ce
        Kernel modules: rtw_8822ce
```

```
root@zynqmp:/home/ubuntu/ko# echo 1 /sys/bus/pci/devices/0000\:01\:00.0/remove
1 /sys/bus/pci/devices/0000:01:00.0/remove
root@zynqmp:/home/ubuntu/ko# [  250.901380] nwl-pcie fd0e0000.pcie: Slave error
[  250.905917] Internal error: synchronous external abort: 96000210 [#1] SMP
[  250.912693] Modules linked in: rtw_8822ce(O) rtw_8822c(O) rtw_pci(O) rtw_core(O) [last unloaded: rtw_core]
[  250.922340] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G           O      5.4.0 #1
[  250.929549] Hardware name: ZYNQ_MP (DT)
[  250.933371] pstate: 60000005 (nZCv daif -PAN -UAO)
[  250.938150] pc : rtw_pci_read32+0x10/0x28 [rtw_pci]
[  250.943017] lr : rtw8822c_led_set+0x2c/0x68 [rtw_8822c]
[  250.948229] sp : ffff800010003cf0
[  250.951527] x29: ffff800010003cf0 x28: ffff00006fbaea00
[  250.956822] x27: dead000000000122 x26: 00000000000000e0
[  250.962117] x25: ffff00006fbaea28 x24: ffff800010003e70
[  250.967412] x23: 0000000000000060 x22: ffff000066081ba8
[  250.972707] x21: ffff800010003db8 x20: 0000000000000001
[  250.978001] x19: ffff000066081e20 x18: 0000000000000000
[  250.983296] x17: 0000000000000000 x16: 0000000000000000
[  250.988591] x15: 0000000000000000 x14: 0000000000000000
[  250.993886] x13: 003d090000000000 x12: 00003d08ffff8ff4
[  250.999181] x11: ffff00006fbaea70 x10: 0000000000000001
[  251.004476] x9 : 0000000000000001 x8 : ffff00006b640450
[  251.009771] x7 : ffff00006fbaea00 x6 : ffff00006fbaed68
[  251.015065] x5 : 0000000080000000 x4 : ffff800008c2e468
[  251.020360] x3 : 0000000000000000 x2 : 0000000000007ec8
[  251.025655] x1 : 000000000000004c x0 : ffff80001143004c
[  251.030950] Call trace:
[  251.033385]  rtw_pci_read32+0x10/0x28 [rtw_pci]
[  251.037909]  led_set_brightness_nopm+0x14/0x40
[  251.042341]  led_blink_setup+0x90/0x118
[  251.046160]  led_blink_set+0x80/0x90
[  251.049721]  tpt_trig_timer+0x138/0x178
[  251.053548]  call_timer_fn.isra.0+0x20/0x78
[  251.057722]  __run_timers.part.0+0x1c0/0x2a8
[  251.061974]  run_timer_softirq+0x38/0x70
[  251.065881]  __do_softirq+0x118/0x22c
[  251.069527]  irq_exit+0x98/0xc0
[  251.072653]  __handle_domain_irq+0x64/0xb8
[  251.076739]  gic_handle_irq+0x5c/0xb8
[  251.080385]  el1_irq+0xb8/0x140
[  251.083511]  arch_cpu_idle+0x10/0x18
[  251.087079]  do_idle+0x200/0x280
[  251.090298]  cpu_startup_entry+0x20/0x28
[  251.094204]  rest_init+0xb0/0xbc
[  251.097416]  arch_call_rest_init+0xc/0x14
[  251.101417]  start_kernel+0x4f8/0x524
[  251.105066] Code: d28fd902 8b020000 f955f400 8b214000 (b9400000)
[  251.111151] ---[ end trace 8257ba2626063971 ]---
[  251.115756] Kernel panic - not syncing: Fatal exception in interrupt
[  251.122092] SMP: stopping secondary CPUs
[  251.126000] Kernel Offset: disabled
[  251.129479] CPU features: 0x0002,20002004
[  251.133470] Memory Limit: none
[  251.136512] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---
```

dubhater, 2025-02-13:
I don't know, but you can ask the official maintainers of rtw88 at the linux-wireless mailing list. They know about PCI stuff.

briansune, 2025-02-13:
@dubhater 

I am using this repository not the official maintainer one.

[  250.912693] Modules linked in: rtw_8822ce(O) rtw_8822c(O) rtw_pci(O) rtw_core(O) [last unloaded: rtw_core]

briansune, 2025-02-13:
@dubhater 

I guess most user only use it as fixed case so don't care those hot-plug stuff.

closed 2025-02-13 by briansune