lwfinger/rtw88 #315: rtl8814ae EFUSE corrupted and cannot recover.

briansune, 2025-02-18:
See below dmesg and report for more info.

https://github.com/briansune/rtw88-hw/tree/main/rtl8814ae_arm_a53

briansune, 2025-02-18:
Added Windows OS comparison 

https://github.com/briansune/rtw88-hw/blob/main/rtl8814ae_arm_a53/README.md#additional-windows-11-os-debug

dubhater, 2025-02-18:
That was my fault. Fixed now.

briansune, 2025-02-19:
Same situation like 8723DS before you patched and EFUSE is blanked and completely dead on STA (no AP are listed in the nmcli or nmtui).

New report see iw info and lspci after driver is able to load and LED indication is on:

https://github.com/briansune/rtw88-hw/tree/main/rtl8814ae_arm_a53

dubhater, 2025-02-19:
I added something. Try again?

closed 2025-02-20 by briansune

dubhater, 2025-02-20:
Quoting from [the other issue](https://github.com/lwfinger/rtw88/issues/251#issuecomment-2670358695):

> 5G STA have hard time to connect need to reconnect multiple times to success, 2.4G shows no such issue.

Is there anything in dmesg when it's trying to connect?

I noticed a difference in the signal strength, -44 with 2.4 GHz and -70 with 5 GHz. Was the distance to the AP much greater when you tested in 5 GHz?

Thank you for buying this and for all the testing.

briansune, 2025-02-21:
@dubhater 

You are welcome.

The router is located at the same location.

So I don't see how could the path loss is so much compared with 2.4G and 5G.

As for AP it is so near to host PC standalone network card to test with so this is why the AP 5G so good.

I will try to grep some dmesg, I run those test using auto script so the report didn't include much.

Meanwhile I do see the EFUSE is really blank and vendor driver do tell using default settings.

briansune, 2025-02-21:
@dubhater 

Antenna issue. those default antenna is bad on 5G band.

Same distance to the router with different antenna on 2.4G band still far better than 5G:

2.4G = Link Quality=62/70  Signal level=-48 dBm
5G = Link Quality=44/70  Signal level=-66 dBm

Almost 20dB different.

Host-Router-DUT iperf3 new result:

```
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
Accepted connection from 192.168.1.252, port 59166
[  5] local 192.168.1.199 port 5201 connected to 192.168.1.252 port 59167
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  5]   0.00-1.00   sec  17.1 MBytes   144 Mbits/sec    0   1024 KBytes
[  5]   1.00-2.00   sec  25.0 MBytes   210 Mbits/sec    0   1.06 MBytes
[  5]   2.00-3.00   sec  25.0 MBytes   210 Mbits/sec    0   1.06 MBytes
[  5]   3.00-4.00   sec  23.8 MBytes   199 Mbits/sec    0   1.06 MBytes
[  5]   4.00-5.00   sec  21.2 MBytes   178 Mbits/sec    0   1.06 MBytes
[  5]   5.00-6.00   sec  17.5 MBytes   147 Mbits/sec    0   1.06 MBytes
[  5]   6.00-7.00   sec  23.8 MBytes   199 Mbits/sec    0   1.06 MBytes
[  5]   7.00-8.00   sec  26.2 MBytes   220 Mbits/sec    0   1.06 MBytes
[  5]   8.00-9.00   sec  23.8 MBytes   199 Mbits/sec    0   1.06 MBytes
[  5]   9.00-10.00  sec  15.0 MBytes   126 Mbits/sec    0   1.06 MBytes
[  5]  10.00-11.00  sec  20.0 MBytes   168 Mbits/sec    0   1.06 MBytes
[  5]  11.00-12.00  sec  21.2 MBytes   178 Mbits/sec    0   1.06 MBytes
[  5]  12.00-13.00  sec  22.5 MBytes   189 Mbits/sec    0   1.06 MBytes
[  5]  13.00-14.00  sec  20.0 MBytes   168 Mbits/sec    0   1.06 MBytes
[  5]  14.00-15.00  sec  16.2 MBytes   136 Mbits/sec    0   1.06 MBytes
[  5]  15.00-16.00  sec  20.0 MBytes   168 Mbits/sec    0   1.06 MBytes
[  5]  16.00-17.00  sec  20.0 MBytes   168 Mbits/sec    0   1.06 MBytes
[  5]  17.00-18.00  sec  22.5 MBytes   189 Mbits/sec    0   1.06 MBytes
[  5]  18.00-19.00  sec  10.0 MBytes  83.9 Mbits/sec    0   1.06 MBytes
[  5]  19.00-20.00  sec  20.0 MBytes   168 Mbits/sec    0   1.06 MBytes
[  5]  20.00-21.00  sec  23.8 MBytes   199 Mbits/sec    0   1.81 MBytes
[  5]  21.00-22.00  sec  23.8 MBytes   199 Mbits/sec    0   1.81 MBytes
[  5]  22.00-23.00  sec  23.8 MBytes   199 Mbits/sec    0   1.81 MBytes
[  5]  23.00-24.00  sec  25.0 MBytes   210 Mbits/sec    0   1.81 MBytes
[  5]  24.00-25.00  sec  23.8 MBytes   199 Mbits/sec    0   1.81 MBytes
[  5]  25.00-26.00  sec  25.0 MBytes   210 Mbits/sec    0   1.81 MBytes
[  5]  26.00-27.00  sec  26.2 MBytes   220 Mbits/sec    0   1.81 MBytes
[  5]  27.00-28.00  sec  15.0 MBytes   126 Mbits/sec    0   1.81 MBytes
[  5]  28.00-29.00  sec  18.8 MBytes   157 Mbits/sec    0   1.81 MBytes
[  5]  29.00-30.00  sec  25.0 MBytes   210 Mbits/sec    0   1.81 MBytes
[  5]  30.00-30.04  sec  1.25 MBytes   287 Mbits/sec    0   1.81 MBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  5]   0.00-30.04  sec   642 MBytes   179 Mbits/sec    0             sender
[  5]   0.00-30.04  sec  0.00 Bytes  0.00 bits/sec                  receiver
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
```

5G HOSTAP - gain around 60 Mbps

```
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
Accepted connection from 192.168.175.86, port 61967
[  5] local 192.168.175.1 port 5201 connected to 192.168.175.86 port 61968
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  5]   0.00-1.00   sec  51.7 MBytes   434 Mbits/sec    0    525 KBytes
[... 14 lines cut ...]
[  5]  15.00-16.01  sec  60.0 MBytes   500 Mbits/sec    0   1.03 MBytes
[  5]  16.01-17.00  sec  58.8 MBytes   496 Mbits/sec    0   1.03 MBytes
[  5]  17.00-18.00  sec  58.8 MBytes   493 Mbits/sec    0   1.03 MBytes
[  5]  18.00-19.00  sec  58.8 MBytes   493 Mbits/sec    0   1.03 MBytes
[  5]  19.00-20.00  sec  60.0 MBytes   503 Mbits/sec    0   1.03 MBytes
[  5]  20.00-21.00  sec  57.5 MBytes   482 Mbits/sec    0   1.03 MBytes
[  5]  21.00-22.00  sec  45.0 MBytes   377 Mbits/sec    0   1.03 MBytes
[  5]  22.00-23.00  sec  50.0 MBytes   419 Mbits/sec    0   1.03 MBytes
[  5]  23.00-24.00  sec  61.2 MBytes   514 Mbits/sec    0   1.03 MBytes
[  5]  24.00-25.00  sec  60.0 MBytes   503 Mbits/sec    0   1.03 MBytes
[  5]  25.00-26.00  sec  62.5 MBytes   524 Mbits/sec    0   1.03 MBytes
[  5]  26.00-27.00  sec  43.8 MBytes   367 Mbits/sec    0   1.03 MBytes
[  5]  27.00-28.00  sec  55.0 MBytes   461 Mbits/sec    0   1.03 MBytes
[  5]  28.00-29.00  sec  47.5 MBytes   398 Mbits/sec    0   1.03 MBytes
[  5]  29.00-30.00  sec  52.5 MBytes   440 Mbits/sec    0   1.03 MBytes
[  5]  29.00-30.00  sec  52.5 MBytes   440 Mbits/sec    0   1.03 MBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  5]   0.00-30.00  sec  1.69 GBytes   484 Mbits/sec    0             sender
[  5]   0.00-30.00  sec  0.00 Bytes  0.00 bits/sec                  receiver
iperf3: the client has terminated
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
```

briansune, 2025-02-26:
@dubhater 

GRIS - 2.4G: Link Quality=62/70  Signal level=-48 dBm
GRIS - 5G Link Quality=46/70  Signal level=-64 dBm

Not much different.

I tried a router completely closest less than 20CM.

GRIS - 5G: Link Quality=70/70  Signal level=-14 dBm

For 4x4 800Mbps I will retry after time is allowed need to reconnect many gear.

GRIS card shows a very significant different on connection.

If connect to 2.4G STA it is instant that I can't even finish blinking my eye.

dubhater, 2025-02-27:
Hmm, maybe the signal strength is normal.

briansune, 2025-02-28:
@dubhater 

TRX test had done, full report: https://github.com/briansune/rtw88-hw/tree/main/rtl8814ae_arm_a53_5g_trxmax#test-index

The peaks can reach 800mbps but very rarely it can. Most cases are 700 range.

I am not sure should 4x4 can reach > 1000 mbps??

I think the driver still have s.t. not turned on?

dubhater, 2025-03-02:
It is possible the driver is still missing something, but it's good enough for now.

I think you can reach over 1000 Mbps with 4 spatial streams. This chip only has 3 streams, even though it has 4 antennas.

briansune, 2025-03-02:
@dubhater 

Yes, forgot the actual stream num.

Usually TxR:S should be same but due to stability design S-1 is used.

But still T x R：S = 4x4:3

[image]

64 QAM 5/6 coding should be higher?