lwfinger/rtw88 #318: RPi5 with Archer T2U PLUS [RTL8821AU] - Monitor mode - "command failed: Device or resource busy (-16)"

jhgarrison, 2025-02-21:
RPi5 with latest OS updates, cannot set monitor mode, I get 

```
jhg@rpi51:~ $ sudo iw wlan1 set type monitor
command failed: Device or resource busy (-16)
```

System details

```
jhg@rpi51:~ $ uname -a
Linux rpi51 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux

jhg@rpi51:~ $ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU]
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

jhg@rpi51:~ $ lsmod|grep rtw
rtw_8821au             49152  0
rtw_8821a              65536  1 rtw_8821au
rtw_88xxa              65536  1 rtw_8821a
rtw_usb                49152  1 rtw_8821au
rtw_core              245760  3 rtw_88xxa,rtw_usb,rtw_8821a
mac80211             1064960  2 rtw_usb,rtw_core
cfg80211             1015808  3 rtw_core,brcmfmac,mac80211

jhg@rpi51:~ $ iw dev
phy#1
        Interface wlan1
                ifindex 4
                wdev 0x100000001
                addr 98:48:27:e9:99:ad
                type managed
                txpower 30.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
phy#0
        Unnamed/non-netdev interface
                wdev 0x2
                addr 2e:cf:67:5c:25:a3
                type P2P-device
                txpower 31.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 2c:cf:67:5c:25:a3
                ssid quasimodem
                type managed
                channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
                txpower 31.00 dBm

```

Suggestions?

briansune, 2025-02-22:
@jhgarrison 

Compared with my report: https://github.com/briansune/rtw88-hw/tree/main/rtl8821au_arm_a53

I think your USB entered the driver section, you need to switch it back?

>/usr/sbin/usb_modeswitch

Just pure guess

jhgarrison, 2025-02-23:
I got past the initial problems by using the script from https://github.com/morrownr/Monitor_Mode.  I adjusted the channel width after starting the script with

    sudo iw dev wlan1mon set channel 40 80MHz

This gets the adapter running in monitor mode, but I'm having trouble seeing any data packets.  Problems are similar but not exactly the same as what I documented in https://github.com/morrownr/8821au-20210708/issues/167.

I suspect some of the issue is in wireshark, since it displays far fewer packets than I see in tcpdump.

Rather than stumbling around in the dark I'll wait for suggestions on what to do next to pinpoint the problem.

morrownr, 2025-02-27:
Hi @jhgarrison 

> I got past the initial problems by using the script from https://github.com/morrownr/Monitor_Mode.

I know the guy that wrote that script. He he. It was written because the out-of-kernel Realtek drivers do not work as they should. However, for folks that are using instructions that they have learned over the years for the old Realtek drivers, you may have some new learning to do because these new drivers should work the way they are supposed to.

What I need to do is take some time to update the script for these modern rtw88 drivers.

> sudo iw dev wlan1mon set channel 40 80MHz

I tried to document the script such that you can edit it to make it do what you want it to do. The documentation is inside the script. However, I have not tested everything.

Right now, I do not have a good answer for you. I'd need to dig my rtl8811au adapter out and do some monitor mode stuff and I do not have the time today. Therefore, stumbling around in the dark may work best in the short term. I know that in the longer term more monitor mode users will show up, some of which are really good coders, and we can get to the bottom of this and other issues.

@morrownr