lwfinger/rtw88 #319: driver installed but no wireless conection

odyn666, 2025-02-28:
`[odyn@archlinux rtw88]$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 258a:1006 SINO WEALTH USB KEYBOARD
Bus 001 Device 004: ID 04e8:4001 Samsung Electronics Co., Ltd PSSD T7
Bus 001 Device 005: ID 3299:4e3a SM900 Microphon SM900 Microphone 
Bus 001 Device 006: ID 0951:16a4 Kingston Technology HyperX 7.1 Audio
Bus 001 Device 007: ID 09da:1c02 A4Tech Co., Ltd. USB Device 
Bus 001 Device 009: ID 05c6:9024 Qualcomm, Inc. YUPIK-QRD _SN:9A9C9EE1
Bus 001 Device 012: ID 0bda:1a2b Realtek Semiconductor Corp. RTL8188GU 802.11n WLAN Adapter (Driver CDROM Mode)
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
[odyn@archlinux rtw88]$ iwconfig
lo        no wireless extensions.

enp0s20f0u7  no wireless extensions.

enp2s0    no wireless extensions.

[odyn@archlinux rtw88]$ 
`

a5a5aa555oo, 2025-02-28:
Your wifi adapter is in Driver CD-ROM mode, you need to install `usb_modeswitch`, it will turn your adapter into Wi-Fi mode.

odyn666, 2025-03-01:
> Your wifi adapter is in Driver CD-ROM mode, you need to install `usb_modeswitch`, it will turn your adapter into Wi-Fi mode.

i figured that out later when i switched it back from cdrom to usb ls usb showed me 

Bus 001 Device 003: ID 0b05:1997 ASUSTek Computer, Inc. 802.11ac WLAN Adapter

but still no wifi connection

a5a5aa555oo, 2025-03-01:
I saw your post [here](https://github.com/lwfinger/rtl8852au/issues/112).
Realtek Wi-Fi 6 USB-based adapters are not well supported on Linux, so It is suggested to find another replacement.

https://github.com/morrownr/USB-WiFi

odyn666, 2025-03-01:
i used this adapter before on fedora with [this](https://github.com/lwfinger/rtl8852au) driver but since kernel udpate it just stoped i was trying build it on arch but it also failed as you ware able to see on previous post you mentioned before

odyn666, 2025-03-01:
thats good idea i use mostly fedora (arch was temporary, just to check if drivers work) tbh where i can find this kind info

morrownr, 2025-03-03:
Hi @odyn666 

> Realtek Wi-Fi 6 USB-based adapters are not well supported on Linux, so It is suggested to find another replacement.

@a5a5aa555oo is giving you very good advice. I understand about trying to make what you have work, but that will likely lead to a lot of frustration over the long term. Here is a post that gives more details:

https://github.com/morrownr/USB-WiFi/issues/449

If you read menu items 1 and 2 at the following link, you will find a lot of good information:

https://github.com/morrownr/USB-WiFi

Maybe you can sell your adapter.

odyn666, 2025-03-05:
hi @morrownr i know its good advice but you know how it is if i have something i want to made it work but in this case i had  enough so currently i must decide if i want a pci one or usb it also must be active monitor capable

morrownr, 2025-03-05:
G'day @odyn666 

> ...  it also must be active monitor capable

This is good information to know. Sometimes there is confusion regarding `monitor mode` vs. `active monitor mode`. Do you understand that `active monitor mode` is just a capability within `monitor mode`?

If you really do want `active monitor mode` then there are only 3 drivers that I am aware of that support it:

mt7610u
mt7612u
mt7921u and mt7921e (this one had a bug a few months ago so I would have to do some testing)
mt 7925 may support it but I have not tested as no USB adapters with that chip are available yet

If you go to my site Main Menu:

https://github.com/morrownr/USB-WiFi

You can see the Plug and Play List and Recommended Adapters for Kali Linux. They will have information on adapters that use those drivers.

odyn666, 2025-03-05:
> G'day [@odyn666](https://github.com/odyn666)
> 
> > ...  it also must be active monitor capable
> 
> This is good information to know. Sometimes there is confusion regarding `monitor mode` vs. `active monitor mode`. Do you understand that `active monitor mode` is just a capability within `monitor mode`?

yes its self explenatory you cannot go into active monitor mode without monitor mode how it will work otherwise.
Maby i misspelled something if yes sorry english is not my first language

morrownr, 2025-03-06:
> Maby i misspelled something if yes sorry english is not my first language

Your English is fine. I have discovered many users that think `active monitor mode` is `monitor mode` but in discussions, it is discovered that they actually do not need `active monitor mode` which means that adapters supported by the drivers here should work for them. It can be confusing.