lwfinger/rtw88 #320: Bluetooth Dropping Connection

AdhipShukla, 2025-03-02:
I followed the installation instructions in the main description, and the driver successfully supports TP LINK AC600 5GHz. However, it interferes with Bluetooth, frequently dropping and stuttering the connection. To validate, I unloaded `sudo modprobe -r rtw_8723de` and `sudo modprobe -r rtw_core`, which resolved the Bluetooth connection issue. Do I somehow need to explicitly enable Bluetooth coexist?

dubhater, 2025-03-02:
Bluetooth coexistence is always on, you don't need to do anything for that.

I don't understand what devices you are using. "TP LINK AC600 5GHz" has nothing to do with rtw_8723de. That driver is for a N150 card which only supports 2.4 GHz.

AdhipShukla, 2025-03-02:
I followed the two-step process in the description, which asks to unload rtw_8723de before rtw_core.

I noticed that if I connect to the 2.5GHz, along the 5GHz SSID from the same router, the Bluetooth connection is also stable. Only when 5GHz is connected alone do I see issues.

[image]

Realtek RTL8723BE is connected to 2.5GHz and AC600 RTL8811AU to 5Ghz.
[image]

dubhater, 2025-03-02:
I see. Well, you would unload rtw_8723de only if that's the module in use. In your case, rtw_8723de doesn't seem to be in use, because you don't have the RTL8723DE card. RTL8723BE is handled by a different driver.

If you are connecting to ItHasToBeCool_5G, that should not interfere with the Bluetooth at all.

Let's start from the beginning: why did you plug in the TP-Link adapter when you already have wifi from the RTL8723BE card?

AdhipShukla, 2025-03-02:
I am using AC600 as the Realtek RTL8723BE card does not support 5GHz.

dubhater, 2025-03-02:
In that case you should not need the old wifi card's driver. What if you unload it? ``modprobe -r rtl8723be`` Is the bluetooth working better?

AdhipShukla, 2025-03-02:
When I remove rtl8723be, it leads to the same connection issue, but as soon as I load it back and also connect to a network on rtl8723be, the BT issue resolves. Also, just realised the BT issue comes even when wifi is turned off altogether. Here are some more logs, I believe the problem is coming as Bluetooth is using **btcoexist rtl8723be**, how to force it to use rtw88?

```
$ sudo modprobe -r rtl8723be
$ lsmod | grep bt
btusb                  77824  0
btrtl                  32768  1 btusb
btintel                57344  1 btusb
btbcm                  24576  1 btusb
btmtk                  12288  1 btusb
bluetooth            1028096  48 btrtl,btmtk,btintel,btbcm,bnep,btusb,rfcomm
$ sudo modprobe rtl8723be
$ lsmod | grep bt
btcoexist             290816  1 rtl8723be
btusb                  77824  0
btrtl                  32768  1 btusb
btintel                57344  1 btusb
btbcm                  24576  1 btusb
btmtk                  12288  1 btusb
bluetooth            1028096  48 btrtl,btmtk,btintel,btbcm,bnep,btusb,rfcomm
rtlwifi               139264  4 rtl_pci,rtl8723be,btcoexist,rtl8723_common
$ lsmod | grep rtw
rtw_8821au             12288  0
rtw_8821a              45056  1 rtw_8821au
rtw_88xxa              45056  1 rtw_8821a
rtw_usb                28672  1 rtw_8821au
rtw_core              344064  3 rtw_88xxa,rtw_usb,rtw_8821a
mac80211             1720320  5 rtl_pci,rtw_usb,rtw_core,rtl8723be,rtlwifi
cfg80211             1323008  3 rtw_core,rtlwifi,mac80211
$ lsmod | grep rtl
rtl8723be             176128  0
btcoexist             290816  1 rtl8723be
rtl8723_common         24576  1 rtl8723be
rtl_pci                40960  1 rtl8723be
btrtl                  32768  1 btusb
bluetooth            1028096  48 btrtl,btmtk,btintel,btbcm,bnep,btusb,rfcomm
rtlwifi               139264  4 rtl_pci,rtl8723be,btcoexist,rtl8723_common
mac80211             1720320  5 rtl_pci,rtw_usb,rtw_core,rtl8723be,rtlwifi
cfg80211             1323008  3 rtw_core,rtlwifi,mac80211
```

dubhater, 2025-03-02:
You can't force it to use rtw88 because rtw88 doesn't know what to do with RTL8723BE.

One more thing to try: create ``/etc/modprobe.d/blacklist-rtl8723be.conf`` and put this line in it:
```
blacklist rtl8723be
```

Then shut down the computer, wait a minute to make sure the RTL8723BE card is powered off, then turn it on again and check the bluetooth.

briansune, 2025-03-03:
@dubhater  @AdhipShukla 

Why not make it more simpler?

Just simply remove the PCIe via

sudo echo 1 > /sys/bus/pci/,,,,/remove

Then the pcie is sure kill during operation for testing purpose?

Need no blacklist and the driver simply states their doing nothing and the PCIe RC just simply stop.

dubhater, 2025-03-03:
@briansune I didn't know about that method.

But blacklisting rtl8723be is my only idea. If it works, then that's the permanent solution. If it doesn't work, I can only suggest replacing the RTL8723BE card. This is an old chip, I doubt it's getting any support from Realtek.

Also, if blacklisting works, it means the problem is in the wifi driver. If it doesn't work, it means the problem is in the bluetooth driver or firmware.

briansune, 2025-03-03:
@dubhater 

The method aka the command is just tell the PCIe to disable the device.
Which this is similar to Windows OS device manager right click disabling devices.

sudo echo 1 > /sys/bus/pci/,,,,/remove

If the PCIe card is disable and works then it is also same as telling the driver is having issue.
Otherwise when the PCIe card is disable the BT driver still having problem the BT driver / firmware is having issue.

RTW88 and BT driver is separately thing if I am correct. 

The only different is that the above disable method never touch any settings on the kernel driver just like turning off the PCIe RC node.

Linux search via google: See: https://superuser.com/questions/541854/disable-specific-pci-device-at-boot