lwfinger/rtw88 #321: Can't install need some help

Slayerwen, 2025-03-04:
I got this when trying to install:

`pi@ender3:~` $ cd rtw88
`pi@ender3:~/rtw88` $ make
make -j1 -C /lib/modules/6.12.16+/build M=$PWD modules
make[1]: *** /lib/modules/6.12.16+/build: No such file or directory.  Stop.
make: *** [Makefile:189: all] Error 2
`pi@ender3:~/rtw88` $ sudo make install
make -j1 -C /lib/modules/6.12.16+/build M=$PWD modules
make[1]: *** /lib/modules/6.12.16+/build: No such file or directory.  Stop.
make: *** [Makefile:189: all] Error 2
`pi@ender3:~/rtw88` `$`

Im on raspberry pi 1, OS Lite

morrownr, 2025-03-05:
Hi @Slayerwen 

> OS Lite

Can you be more specific? Is this RasPiOS Lite? 32 bit?

Slayerwen, 2025-03-05:
Yes OS Lite 32 bits I think. And I update to latest kernel version to 6.12. And stuffs.

When I follow your guide and trying update kernel headers I do get this message:

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'raspberrypi-kernel-headers' instead of 'linux-headers'
raspberrypi-kernel-headers is already the newest version (1:1.20230405-1).
The following package was automatically installed and is no longer required:
  libcamera0.3
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

morrownr, 2025-03-06:
@Slayerwen 

> Yes OS Lite 32 bits I think. 

Well, I really need to know so if you can be sure, that would help.

https://www.raspberrypi.com/software/operating-systems/

Scroll down to see:

Raspberry Pi OS Lite
Release date: November 19th 2024
System: 32-bit
Kernel version: 6.6
Debian version: 12 (bookworm)

Is that it?

> And I update to latest kernel version to 6.12. And stuffs.

RasPiOS has not been updated to kernel 6.12. This makes it hard for me to offer any advice because I do not know what you have done.

> raspberry pi 1

This is not very specific. 1A+ or 1B+ or exactly what model?

> make[1]: *** /lib/modules/6.12.16+/build: No such file or directory. Stop.

This makes it look like the header files are not installed. The changes that you have made make it very difficult to help you. My recommendation is for you to do a clean installation with the current version of RasPiOS.

morrownr, 2025-03-06:
> I just typed in update so it updated to 6.12 automatic.

$ `sudo apt update` won't do that. There are additional commands that will but they can destabilize your system. Not recommended unless you are highly skilled. 

> My Pi is a 1B+ 2014

Okay, This would have been nice to know. That is a 64 bit chip. Here is a copy and paste from one of my FAQs:

 Raspberry Pi

Problem: no header files detected even though they are supposedly installed

Description: If you are running the 32 bit version of RasPiOS on RasPi
hardware that is 64 bit capable, you may run into a situation were
header files are not detected when compiling a driver. The problem is
that if RasPiOS detects a 64 bit capable system, then it will use a
64 bit kernel even if a 32 bit version of RasPiOS was installed. This
confuses the situation and results in problems.

> When running `sudo sh install-driver.sh` on my 64 bit RasPi hardware,
> I see the following:
>
>> Your kernel header files aren't properly installed.
>> Please consult your distro documentation or user support forums.
>> Once the header files are properly installed, please run...

The RasPiOS now prefers the 64-bit kernel if one exists so even if you
installed the 32 bit version of the RasPiOS, you may now have the 64 bit
kernel active.

To fix, add the following line to `/boot/config.txt` or
`/boot/firmware/config.txt` depending on your version of
RasPiOS:

``` text
arm_64bit=0
```
Save the file and reboot.

Reference: <https://forums.raspberrypi.com/viewtopic.php?p=2091532&hilit=Tp+link#p2091532>

> [!NOTE]
> If your RasPi is capable of using a 64 bit version of the RasPiOS, it is recommended that you use a 64 bit version of the RasPiOS as this will eliminate this problem and possible others.

Note to RasPiOS devs:
We really really wish you would consider the consequences of the changes you make.
Thank you.

Slayerwen, 2025-03-06:
I think sudo apt install --reinstall raspi-firmware fix my problem

Slayerwen, 2025-03-06:
Updates: I have installet now and every thing is working great!

Only one thing i dont understand is Blacklisting, when do i need to do that?

morrownr, 2025-03-06:
> Only one thing i dont understand is Blacklisting, when do i need to do that?

With this rtw88 repo, you don't need to do it. The Makefile that controls things like the compilation, installation and uninstallation is written to automatically blacklist and unblacklist the possible conflicts. You can look in the following file to see what is blacklisted for you:

blacklist-rtw88.conf

I was looking back to see what chip is in your adapter but I cannot see where you mentioned it. So what driver are you using so that we know where the problem is in case you have a problem?

Let us know if you have any problems.