lwfinger/rtw88 #323: [Archer T2U Nano] RTL8811au driver not working under elementaryOS 8 (Circle)

MishterKirby, 2025-03-08:
On a fresh install of elementaryOS 8, I installed the upstream 6.14rc5 kernel and the rtw88 driver, but upon trying to connect to a network, it just eternally keeps retrying without success.

dmesg hasn't been too helpful, either, and I noticed after installing the driver and replugging the dongle, commands such as lsusb take a long time to output
```
mishy@elderberry:~/rtw88$ sudo dmesg | grep rtw
[   72.915410] rtw_core: loading out-of-tree module taints kernel.
[   72.915418] rtw_core: module verification failed: signature and/or required key missing - tainting kernel
[   72.933627] rtw_8821au 1-7:1.0: Firmware version 42.4.0, H2C version 0
[  100.418100] usbcore: registered new interface driver rtw_8821au
[  100.434380] rtw_8821au 1-7:1.0 wlx34e894e4e8e8: renamed from wlan0
[  143.556147] rtw_8821au 1-7:1.0: write register 0x122c failed with -71
[  173.987558] rtw_8821au 1-7:1.0: write register 0x1230 failed with -110
[  175.454181] rtw_8821au 1-5:1.0: Firmware version 42.4.0, H2C version 0
[  202.967476] rtw_8821au 1-5:1.0 wlx34e894e4e8e8: renamed from wlan0
[ 2145.481112] usbcore: deregistering interface driver rtw_8821au
[ 2376.848820] rtw_8821au 1-5:1.0: Firmware version 42.4.0, H2C version 0
[ 2404.338597] usbcore: registered new interface driver rtw_8821au
[ 2404.352845] rtw_8821au 1-5:1.0 wlx34e894e4e8e8: renamed from wlan0

```

briansune, 2025-03-08:
@MishterKirby 

Can you post a simple picture or image that dongle you are using?

It feels like you are using the same one I got.
Indeed it is bad.

The antenna is so badly designed.

Report: https://github.com/briansune/rtw88-hw/tree/main/rtl8811au_arm_a9
Report: https://github.com/briansune/rtw88-hw/tree/main/rtl8811au_arm_a35
Report: https://github.com/briansune/rtw88-hw/tree/main/rtl8811au_arm_a53

MishterKirby, 2025-03-08:
This is the one I'm using (straight from TP-Link's website)
[image]

briansune, 2025-03-08:
@MishterKirby 

Did you try cross comparison on Windows ?

With that same device the antenna is sure going to be even short range.

Despite the reconnection issue. The connection to STA is more important.

"Todo" Try move it closer to the router or AP and cross compared on Windows and Linux.

If the issue stays then it is somehow the driver is missing s.t.

MishterKirby, 2025-03-08:
Under Windows it works correctly and I get full strength signal with no dropouts on the same machine on the same network. 

Under elementaryOS, it fails to connect to any network, including my mobile hotspot with my phone just inches away from the receiver, so I genuinely think it's something driver related, maybe (though I doubt it, considering there's other working antenna with the same chipset) it is missing station support like you said.

I did also notice lsusb refers to it as an 8811au, but the system tries loading the 8812au driver, so maybe that could also be an issue?

briansune, 2025-03-08:
@MishterKirby 

No you should use 8821au driver for 8811au.

Can you manually try to use 8821au driver?

MishterKirby, 2025-03-08:
> No you should use 8821au driver for 8811au.
> Can you manually try to use 8821au driver?

The `rtw_8821au` module is already loaded, and I don't see any modules labeled `8811au` so I'm guessing that's the only one.
 

> I found that the driver you loaded calls out
> "rtw_core: module verification failed: signature and/or required key missing - tainting kernel"
> Did you do the steps as:
> https://github.com/lwfinger/rtw88?tab=readme-ov-file#installation-with-secureboot-for-all-distros-

I have secure boot disabled so I didn't think I'd need to do that, but I'll try again and see if it helps or not.

> If you still having trouble on RTW88 simply use, it is documented it is tested on the dongle you are trying out.
> https://github.com/morrownr/8821au-20210708

Yeah, I used to use that one back on Ubuntu 16.04, but not only did I want to switch because it's unmaintained, but because this one is supposed to be in the mainline kernel so I was expecting it would work out of the box from kernel 6.13 onwards.

briansune, 2025-03-08:
> I have secure boot disabled so I didn't think I'd need to do that, but I'll try again and see if it helps or not.

My understanding to this is that "CONFIG_MODULE_SIG"  either on 

kernel build is set to disable

or 

during the RTW88 build the CONFIG_MODULE_SIG is set to disable.

But either both of that should not return the message that is shown on the dmesg.

> Yeah, I used to use that one back on Ubuntu 16.04, but not only did I want to switch because it's unmaintained,

I am not aware that it is stop maintained. I still see there are merges few days ago. 

> I was expecting it would work out of the box from kernel 6.13 onwards.

But of cause, any driver can make things work should make your day.

Does default ubuntu RTW88_CORE not working and you need to pull from this downstream out-of-tree driver?

Latest kernel LTS already included RTW88 inside. I also compiled latest kernel days ago on ROCKCHIP with longterm: 6.12.18 also using Ubuntu as well.

MishterKirby, 2025-03-08:
> I am not aware that it is stop maintained. I still see there are merges few days ago.

Maybe I misunderstood this, but on *that* driver's README:
```The new in-kernel driver is Linux Standards compliant and is a much better driver than this one. This driver will no longer get API related updates beyond kernel 6.13.```

> Does default ubuntu RTW88_CORE not working and you need to pull from this downstream out-of-tree driver?

Ubuntu 24.04 LTS (what elementaryOS 8 is based off of) ships with kernel 6.8, which as far as I know doesn't have RTW88.

> Latest kernel LTS already included RTW88 inside. I also compiled latest kernel days ago on ROCKCHIP with longterm: 6.12.18 also using Ubuntu as well.

I'll try downgrading my kernel to 6.12, but as far as I remember none of the kernels I've tried before worked out-of-box and I always had to use a DKMS module.

briansune, 2025-03-08:
> I'll try downgrading my kernel to 6.12, but as far as I remember none of the kernels I've tried before worked out-of-box and I always had to use a DKMS module.

My work flow is different I build u-boot, kernel, rootfs for ARM-based platform so desktop based Ubuntu might be different.

But sound different on desktop case:

[image]

I do see all rtw88 drivers "Out of the box"

MishterKirby, 2025-03-08:
Just tried reinstalling it, and I still get:

```
mishy@elderberry:~/rtw88$ sudo dmesg | grep rtw
[    7.061930] rtw_core: loading out-of-tree module taints kernel.
[    7.061938] rtw_core: module verification failed: signature and/or required key missing - tainting kernel
[    7.247231] rtw_8821au 1-5:1.0: Firmware version 42.4.0, H2C version 0
[   34.735398] usbcore: registered new interface driver rtw_8821au
[   34.752582] rtw_8821au 1-5:1.0 wlx34e894e4e8e8: renamed from wlan0
```

Additionally, I also noticed:

```
mishy@elderberry:~/rtw88$ sudo dmesg | grep wlx34e894e4e8e8
[   34.752582] rtw_8821au 1-5:1.0 wlx34e894e4e8e8: renamed from wlan0
[  131.427986] wlx34e894e4e8e8: authenticate with e8:9f:80:6f:19:2a (local address=34:e8:94:e4:e8:e8)
[  133.294176] wlx34e894e4e8e8: send auth to e8:9f:80:6f:19:2a (try 1/3)
[  133.294369] wlx34e894e4e8e8: aborting authentication with e8:9f:80:6f:19:2a by local choice (Reason: 3=DEAUTH_LEAVING)
```

briansune, 2025-03-08:
@MishterKirby 

What WAP you are using? WAP2

MishterKirby, 2025-03-08:
Yes, my network is WPA2

briansune, 2025-03-08:
@MishterKirby 

Then should be normal to connect.

Router WIFI 5G and 2,4G is separated  not combined SSID?

I also give it a try on Ubuntu x64 with down-stream RTW88 on virtual machine it is able to connect but really not stable due to that very bad antenna.

[image]

[image]

briansune, 2025-03-08:
@MishterKirby 

Nan, somehow combined SSID is not doing good on the rtw88 for trying purpose split it.

Then retry connection, I suggest you try 2.4G with no Bluetooth device nearby and if it is ok then the day is saved.

Let see, just one setting in the router.

MishterKirby, 2025-03-08:
> Nan, somehow combined SSID is not doing good on the rtw88 for trying purpose split it.
> Then retry connection, I suggest you try 2.4G with no Bluetooth device nearby and if it is ok then the day is saved.

Even after splitting it, it refuses to connect.

briansune, 2025-03-08:
@MishterKirby 

Both 2.4G and 5G?

insert the driver of the rtw_core:

>sudo insmod rtw_core debug_mask=0x20

Post the demsg and eFUSE table etc out.

@dubhater 

Maybe the driver have some issue.

dubhater, 2025-03-08:
@briansune Yes, it looks like the driver has some problem.

@MishterKirby You said you installed kernel 6.14, which has the driver for this device, but it looks like you also installed the driver from this repository. Which modules are actually loaded? ``lsmod | grep rtw``

What kind of computer do you have? Is the T2U plugged into an add-on card or hub?

a5a5aa555oo, 2025-03-08:
> > Router WIFI 5G and 2,4G is separated  not combined SSID?
> 
> No, my SSIDs are combined.
> 
> Maybe elementaryOS does something weird with the kernel module because I've tried different kernel versions and none of them seem to work well.

I tested this driver with 8822BU on ElementaryOS 8 and it works, so elementaryOS should not be the cause.

briansune, 2025-03-08:
@dubhater @a5a5aa555oo 

I might be still not understanding the distro and kernel enough.

https://www.kernel.org/

The mainline do included rtl8821au code but not LTS.

And instead of special distro changed / modified the kernel for purpose most distro should follow that website.

So distro itself should not introduce any inherent issue from first place as whatever it calls the bottom kernel are all still the same trunk?

MishterKirby, 2025-03-08:
> [@MishterKirby](https://github.com/MishterKirby) You said you installed kernel 6.14, which has the driver for this device, but it looks like you also installed the driver from this repository. Which modules are actually loaded? `lsmod | grep rtw`
> 
Yes, I tried installing 6.14 but when that didn't work I went back to 6.8 and installed the driver, which yielded the same results

```mishy@elderberry:~/rtw88$ lsmod | grep rtw
rtw_8821au             12288  0
rtw_8821a              45056  1 rtw_8821au
rtw_88xxa              45056  1 rtw_8821a
rtw_usb                28672  1 rtw_8821au
rtw_core              344064  3 rtw_88xxa,rtw_usb,rtw_8821a
mac80211             1720320  2 rtw_usb,rtw_core
cfg80211             1323008  2 rtw_core,mac80211
```

> What kind of computer do you have? Is the T2U plugged into an add-on card or hub?

It's a desktop, and the T2U is plugged in straight to the back USB2 ports on the motherboard.

> I tested this driver with 8822BU on ElementaryOS 8 and it works, so elementaryOS should not be the cause.
> kernel version: 6.8.0-55-generic

Yeah, I think it's a problem with the driver not communicating correctly with the 8811au chipset (or maybe my dongle has some unknown quirk or something that doesn't play nicely.)

dubhater, 2025-03-08:
Okay. Can you show the output of ``lspci -nn`` and a full dmesg with the errors?

MishterKirby, 2025-03-08:
Alright
`lspci -nn` shows
```
mishy@elderberry:~$ lspci -nn
00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Root Complex [1022:1480]
00:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse IOMMU [1022:1481]
00:01.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:01.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse GPP Bridge [1022:1483]
00:01.3 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse GPP Bridge [1022:1483]
00:02.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:03.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:03.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse GPP Bridge [1022:1483]
00:04.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:05.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:07.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:07.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Internal PCIe GPP Bridge 0 to bus[E:B] [1022:1484]
00:08.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:08.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Internal PCIe GPP Bridge 0 to bus[E:B] [1022:1484]
00:14.0 SMBus [0c05]: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller [1022:790b] (rev 61)
00:14.3 ISA bridge [0601]: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge [1022:790e] (rev 51)
00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 0 [1022:1440]
00:18.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 1 [1022:1441]
00:18.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 2 [1022:1442]
00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 3 [1022:1443]
00:18.4 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 4 [1022:1444]
00:18.5 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 5 [1022:1445]
00:18.6 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 6 [1022:1446]
00:18.7 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 7 [1022:1447]
01:00.0 Non-Volatile memory controller [0108]: Solid State Storage Technology Corporation CL1-3D256-Q11 NVMe SSD M.2 [1e95:9100] (rev 03)
03:00.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset USB 3.1 xHCI Compliant Host Controller [1022:43d5] (rev 01)
03:00.1 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset SATA Controller [1022:43c8] (rev 01)
03:00.2 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Bridge [1022:43c6] (rev 01)
20:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Port [1022:43c7] (rev 01)
20:01.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Port [1022:43c7] (rev 01)
20:04.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Port [1022:43c7] (rev 01)
20:05.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Port [1022:43c7] (rev 01)
20:06.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Port [1022:43c7] (rev 01)
20:07.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Port [1022:43c7] (rev 01)
25:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8211/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 15)
29:00.0 VGA compatible controller [0300]: NVIDIA Corporation GA106 [GeForce RTX 3060 Lite Hash Rate] [10de:2504] (rev a1)
29:00.1 Audio device [0403]: NVIDIA Corporation GA106 High Definition Audio Controller [10de:228e] (rev a1)
2a:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Function [1022:148a]
2b:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Reserved SPP [1022:1485]
2b:00.1 Encryption controller [1080]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Cryptographic Coprocessor PSPCPP [1022:1486]
2b:00.3 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Matisse USB 3.0 Host Controller [1022:149c]
2b:00.4 Audio device [0403]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse HD Audio Controller [1022:1487]
```

And dmesg (attached as a text file as it is huge)

[dmesg.txt](https://github.com/user-attachments/files/19145942/dmesg.txt)

dubhater, 2025-03-08:
Hmm, does this help?

```diff
diff --git a/rtw8821au.c b/rtw8821au.c
index a01744b..d52a3af 100644
--- a/rtw8821au.c
+++ b/rtw8821au.c
@@ -70,6 +70,7 @@ static struct usb_driver rtw_8821au_driver = {
 	.id_table = rtw_8821au_id_table,
 	.probe = rtw_usb_probe,
 	.disconnect = rtw_usb_disconnect,
+	.disable_hub_initiated_lpm = 1,
 };
 module_usb_driver(rtw_8821au_driver);
 
```

MishterKirby, 2025-03-08:
No dice, it still refuses to connect.

dubhater, 2025-03-08:
Okay, then back to that dmesg: it did not have any errors in it.

MishterKirby, 2025-03-08:
Any other suggestions where I might be able to look for a possible root cause?

briansune, 2025-03-08:
@MishterKirby 

Do a USB list. 

> sudo lsusb

> sudo lsusb -t

dubhater, 2025-03-08:
> rtw_8821au 1-7:1.0: write register 0x122c failed with -71

@MishterKirby Does it always die with an error like that? Specifically, is the register always in the range 0x1000...0x1fff?

briansune, 2025-03-08:
@dubhater 

Will there any blocking on the USB when the signature key is missing?

[    7.395470] rtw_core: module verification failed: signature and/or required key missing - tainting kernel

dubhater, 2025-03-08:
@briansune No, that's harmless. I get that message all the time.

briansune, 2025-03-08:
@dubhater 

Do you need eFUSE dump from my side?

I got two RTL8811AU.

dubhater, 2025-03-08:
@briansune Thank you, but I have one Archer T2U Nano. It works for me.

dubhater, 2025-03-08:
@MishterKirby Another patch for you to try: [chunks.diff.txt](https://github.com/user-attachments/files/19146371/chunks.diff.txt)

morrownr, 2025-03-09:
Just now catching up with this issue. Whew!

I think the best thing I can do right now is add a little history to help sort things out...

> Ubuntu 24.04 LTS (what elementaryOS 8 is based off of) ships with kernel 6.8, which as far as I know doesn't have RTW88.

Yes, Ubuntu 24.04 LTS does have rtw88. rtw88 goes back further than kernel 6.2 but USB support was added in kernel 6.2. 

The USB support in kernel 6.2 was in need of a lot of work. Late in 2023, Dubhater, Larry Finger and I started testing and work to improve the USB support in rtw88. A lot of testers were very helpful during 2024 and numerous patches went in to improve rtw88 USB support. Thank you Dubhater for most of the hard work. Larry Finger passed away in May of 2024. Work continued. Work was started on rtl8821/11au and rtl8812au support and those modules started going into kernel 6.13 and completed with kernel 6.14. Currently rtl8814au support is going into kernel 6.15.

Things that may be confusing things or otherwise causing problems:

- Ubuntu has a had an installable driver for rtl8812au/rtl8821/11au in its repos for years. It is based on a very old version of the Realtek out of kernel driver for the rtl8812au (which, at that time, also supported the rtl8821/11au chips). If that driver has been installed, it needs to be removed so that it does not cause problems. It appears that this potential conflict has been cleaned up for Ubuntu 25.04 but any distro using the last LTS, 24.04, will have this potential conflict for many years.

- Any other out-of-kernel drivers that may have been installed could cause problems as well.

- There are not different drivers for the rtl8821 and rtl8811 chips. They are the same chip except that the rtl8821 has bluetooth support. The driver is call 8821au or rtw_8821au or maybe other names depending on where you got it. The end name for the in-kernel driver should be rtw88_8821au. It is different in this repo so as to avoid confusion.

Now to address some comments in this thread:

>  I installed the upstream 6.14rc5 kernel and the rtw88 driver, but upon trying to connect to a network, it just eternally keeps retrying without success.

I would avoid 6.14 rcx like the plague right now. It has numerous wireless issues. Recommend you use 6.8 or 6.11.

> dmesg hasn't been too helpful, either, and I noticed after installing the driver and replugging the dongle, commands such as lsusb take a long time to output

Welcome to kernel 6.14. Back off to something earlier.

> $ sudo dmesg | grep rtw

I would recommend you also run the following to help make sure no conflicts are in there:

$ sudo dmesg | grep 8821
$ sudo dmesg | grep 8812
$ dkms status

> I did also notice lsusb refers to it as an 8811au, but the system tries loading the 8812au driver, so maybe that could also be an issue?

> 8812au

That indicates to me that an out-of-kernel driver is being loaded. Have you or Elementary OS tried to install the 8812au driver that is in the Ubuntu repos... the Elementary devs may have made this more automatic.

briansune, 2025-03-09:
> Yes, Ubuntu 24.04 LTS does have rtw88. rtw88 goes back further than kernel 6.2 but USB support was added in kernel 6.2.

This is not true and did you even try install Ubuntu desktop 24.04 LTS image?

It does have rtw88 but only that the RTL8821AU is not included and the kernel version is also not correct is it 6.11.0-19

See upper comments

> > I'll try downgrading my kernel to 6.12, but as far as I remember none of the kernels I've tried before worked out-of-box and I always had to use a DKMS module.
> 
> My work flow is different I build u-boot, kernel, rootfs for ARM-based platform so desktop based Ubuntu might be different.
> 
> But sound different on desktop case:
> 
> [image]
> 
> I do see all rtw88 drivers "Out of the box"
> 
> Maybe what you mean is that RTL8821AU is not included some rtw88 driver is not included.
> 
> [@MishterKirby](https://github.com/MishterKirby)
> 
> I feel so confused from your case.
> 
> But really got off track from the ticket.
> 
> Can you just redo the sign install can see if it is still having issue?

briansune, 2025-03-09:
@morrownr 

I recall the last response it is not helping this ticket and no point to discuss in this direction.

What @MishterKirby trying to achieve is Out-of-the-box (no extract works is needed).

So any options do able him to install w/o manually reinstalling the RTL8821AU driver?

This is more important.

What I saw is that longterm: | 6.12.18 also missing some RTW drivers.

a5a5aa555oo, 2025-03-09:
> What I saw is that longterm: | 6.12.18 also missing some RTW drivers.

@briansune 
kernel 6.12 will not have the driver for 8821AU/8812AU because the stable kernels accept small patches only. [1]
[Here is also some FAQ](https://www.kernel.org/category/releases.html) about how kernel development is working, hope it helpful to you.

[1] https://github.com/torvalds/linux/blob/master/Documentation/process/stable-kernel-rules.rst

briansune, 2025-03-09:
@a5a5aa555oo 

Okay but how about distro itself it might not necessary uses LTS right?

So the question still opens that for this ticket creator what it is seeking is distro that have RTL8821AU stable from first place.

briansune, 2025-03-09:
@a5a5aa555oo 

Appreciate the answer with clean and clear.

So MishterKirby should kept using the driver from morrownr on chipset RTL8821AU / RTL8811AU.

Until dubhater working out the patches and distro have out-of-the-box support.

Time are more reasonable as such at least morrownr driver is well test on that dongle.

MishterKirby, 2025-03-09:
> [@MishterKirby](https://github.com/MishterKirby) Another patch for you to try: [chunks.diff.txt](https://github.com/user-attachments/files/19146371/chunks.diff.txt)

Cool, I'll apply this patch and report back if this patch works or not in a bit.

> Yes, Ubuntu 24.04 LTS does have rtw88. rtw88 goes back further than kernel 6.2 but USB support was added in kernel 6.2.
> The USB support in kernel 6.2 was in need of a lot of work. Late in 2023, [...]

I see, I understood kernel 6.14 had support for the rtl8821/11au, but was missing the rest of the context, hence my decision to install a prerelease kernel in the first place. Originally I tried on 6.8 but I don't remember exactly what happened (other than it not working) that lead me to try 6.14

> Things that may be confusing things or otherwise causing problems:
>
>     * Ubuntu has a had an installable driver for rtl8812au/rtl8821/11au in its repos for years. It is based on a very old version [...]
I see, I kinda suspected that they were the same but I wasn't too sure.

I'll also double-check because at one point earlier I did try to install the other driver that was suggested earlier in the thread, but while I thought I uninstalled it after that didn't work either, I may be misremembering and causing a driver conflict.

> Now to address some comments in this thread:
>
>
> I would avoid 6.14 rcx like the plague right now. It has numerous wireless issues. Recommend you use 6.8 or 6.11.

Yes, I'm currently back on the 6.8 kernel Elementary ships with, albeit with the same end result as of now with 6.14.

> That indicates to me that an out-of-kernel driver is being loaded. Have you or Elementary OS tried to install the 8812au driver that is in the Ubuntu repos... the Elementary devs may have made this more automatic.

As I said earlier I'll make sure to check, I don't recall installing any driver packages myself (other than NVIDIA drivers but that's got nothing to do with wireless), but maybe Elementary did it automatically and I just didn't notice it.

morrownr, 2025-03-09:
> Yes, I'm currently back on the 6.8 kernel Elementary ships with, albeit with the same end result as of now with 6.14.

Did you install the driver in this repo? Remember that wireless drivers in Linux consist of multiple files. Driver files and one or more firmware files. Following the instructions in this repo will install the firmware files for you and well as compiling and installing the driver files.

Just updating to kernel 6.14 rcx is hit or miss. Does your distro have compiling the rtw88_8821au driver in kernel 6.14 turned on? Distros have to turn it on. Are the firmware files installed?

Having several years experience maintaining some out-of-kernel drivers has led to me having a lot of experience with users of many different distros. The amount of ideas that are not so good that are used by distro maintainers is large. I knew distros would have to do a lot of cleaning up with the new rtw88 rtl8812au and rtl8821/11au support coming online. How this cleanup goes is going to depend on each distro. Many distros do not have enough helpers to do even basic things let alone something like this so it may not happen until bug reports flood in. I have no experience with the distro you are using so can't be specific with help.

MishterKirby, 2025-03-09:
Alright, I tried the patch but it didn't work (still won't connect). I also checked but I don't seem to have any other drivers installed, either.

> Did you install the driver in this repo? Remember that wireless drivers in Linux consist of multiple files. Driver files and one or more firmware files. Following the instructions in this repo will install the firmware files for you and well as compiling and installing the driver files.

Yes, I compiled it myself from this repo for both 6.8 and 6.14 when I was using it.

> Did you install the driver in this repo? Remember that wireless drivers in Linux consist of multiple files. Driver files and one or more firmware files. Following the instructions in this repo will [...]

I'll have to check the kernel's config, but I installed a mainline 6.14 kernel so I assume whatever is default on the mainline one was set on the one I installed (which based on your statement, I'm assuming that would be disabled by default).

> I would recommend you also run the following to help make sure no conflicts are in there:
> 
> $ sudo dmesg | grep 8821 $ sudo dmesg | grep 8812 $ dkms status

Here's the output of all 3 commands (as far as I see I don't see anything that looks wrong other than the second one perhaps)

```
mishy@elderberry:~$ sudo dmesg | grep 8821
[    7.517582] rtw_8821au 1-5:1.0: Firmware version 42.4.0, H2C version 0
[   35.000057] usbcore: registered new interface driver rtw_8821au
[   35.011875] rtw_8821au 1-5:1.0 wlx34e894e4e8e8: renamed from wlan0
[ 2877.826707] usbcore: deregistering interface driver rtw_8821au
[ 2951.261499] rtw_8821au 1-5:1.0: Firmware version 42.4.0, H2C version 0
[ 2978.744047] usbcore: registered new interface driver rtw_8821au
[ 2978.762782] rtw_8821au 1-5:1.0 wlx34e894e4e8e8: renamed from wlan0
```

```
mishy@elderberry:~$ sudo dmesg | grep 8812
[ 2918.333715] rtw_8812au: Unknown symbol rtw8812a_hw_spec (err -2)
```

```
mishy@elderberry:~$ sudo dkms status
nvidia/550.120, 6.8.0-49-generic, x86_64: installed
```

briansune, 2025-03-09:
@MishterKirby 

How about the driver you had used before?

https://github.com/morrownr/8821au-20210708

Do this in the current environment works as before?

MishterKirby, 2025-03-09:
Can confirm that it works properly with that driver, yes.

dubhater, 2025-03-09:
@MishterKirby All right, then try to connect to a network and attach the full dmesg again.

MishterKirby, 2025-03-09:
Alright, here's the dmesg with the other driver

[dmesg.txt](https://github.com/user-attachments/files/19151226/dmesg.txt)

dubhater, 2025-03-09:
Oh, I meant with rtw88.

MishterKirby, 2025-03-09:
Oh I'm sorry, I misunderstood.
Here's the output with rtw88.

[dmesg-rtw88.txt](https://github.com/user-attachments/files/19151292/dmesg-rtw88.txt)

dubhater, 2025-03-09:
```
[  131.649011] wlx34e894e4e8e8: authenticate with e8:9f:80:6f:19:2a (local address=34:e8:94:e4:e8:e8)
[  133.479179] wlx34e894e4e8e8: send auth to e8:9f:80:6f:19:2a (try 1/3)
[  133.479444] wlx34e894e4e8e8: aborting authentication with e8:9f:80:6f:19:2a by local choice (Reason: 3=DEAUTH_LEAVING)
```

I have never seen that before ("aborting authentication").

Seven years ago it was blamed on the interface name: https://unix.stackexchange.com/questions/386925/aborting-authentication-by-local-choice-reason-3-deauth-leaving-when-trying

This makes no sense to me, but who knows what's going on in Debian.

MishterKirby, 2025-03-09:
Hmmm, I will try the proposed solution over at StackExchange later (as I'm not at my PC at the moment), and see if that works.

If it does, I'm guessing it's nothing that can be worked around driver-side, but rather something that has to be fixed on the distro, correct?

dubhater, 2025-03-09:
As far as I know, the driver doesn't do anything with the interface name, so there is nothing to fix here.

briansune, 2025-03-09:
@dubhater 

Do you able to repeat what @MishterKirby facing on the same dongle?

It is completely make no sense unless the hardware is fault from first place.

You do have the same dongle and yet the register errors are not reviewing on your side?

briansune, 2025-03-09:
@MishterKirby 

Maybe it is not driver from first place.

The network manager could be the root issue.

See the link explains: https://bbs.archlinux.org/viewtopic.php?id=233365

dubhater, 2025-03-09:
@briansune No, my Archer T2U Nano works. I'm using it right now. But I don't have a B450 motherboard or Elementary OS.

briansune, 2025-03-09:
@dubhater 

I guess you had done what all you can.

On the same dongle, the same driver and yet somehow have two different results.

We all learnt sometime from that.

MishterKirby, 2025-03-09:
Tried the suggested StackExchange fix, still doesn't work

> [@MishterKirby](https://github.com/MishterKirby) Does it always die with an error like that? Specifically, is the register always in the range 0x1000...0x1fff?

Also forgot to answer this yesterday but yes, it always seems to log an error like that on dmesg
I'll try rebooting and I'll attach a dmesg log once my system's back up just to be completely sure.

briansune, 2025-03-10:
@MishterKirby 

This is caused by unplug the RTL8811AU right?

So this is different issue apart from the WIFI connection.

This is not necessary critical because you can unload the driver via terminal.

But the WIFI connection itself is considered as more critical issue.

Can you see the link: https://bbs.archlinux.org/viewtopic.php?id=233365

Do the OS have somehow messed up network manager?

Although it is archlinux but this is still valid what that post is trying to solve.

a5a5aa555oo, 2025-03-10:
@MishterKirby

Your 8811AU dongle, external HDD and Bluetooth dongle are all connected to the same USB 3 host controller, does your wifi dongle work if your remove another two devices (external HDD and BT dongle)?

Or does it work if you plug the Wi-Fi dongle in an USB 2.0 port?

briansune, 2025-03-10:
@a5a5aa555oo 

https://github.com/lwfinger/rtw88/issues/323#issuecomment-2708708721

> [@MishterKirby](https://github.com/MishterKirby)
> 
> How about the driver you had used before?
> 
> https://github.com/morrownr/8821au-20210708
> 
> Do this in the current environment works as before?

https://github.com/lwfinger/rtw88/issues/323#issuecomment-2708721034

> Can confirm that it works properly with that driver, yes.

MishterKirby, 2025-03-12:
> Your 8811AU dongle, external HDD and Bluetooth dongle are all connected to the same USB 3 host controller, does your wifi dongle work if your remove another two devices (external HDD and BT dongle)?
> 

I apologize for the late reply, I've been dealing with a few situations and haven't had time to troubleshoot. 
But I just tested and I still get the same result as with the other devices plugged in.

[4 more replies]