lwfinger/rtw88 #329: rtw_8822ce connectivity issues

josepepito-upm, 2025-03-15:
Hi everyone.

I'm experiencing some problems with realtek driver rtw_8822ce on arch linux. I can connect to wireless networks, but there is no internet.

I came to the conclusion that is a driver's fault due to the kernel messages. Here are the messages after booting: [https://pastebin.com/ic4K1dEk](url). 

I've tried making a configuration file with some configurations to disable power saving on the module: 
``` 
[ganxo@ganxotatil modprobe.d]$ cat rtw88_8822ce.conf 
blacklist rtw_8822ce
options rtw_pci disable_aspm=y disable_msi=y
options rtw_core disable_lps_deep=y
```

I've also installed the `rtw88-dkms-git` package from the AUR, but it includes a blacklist of all the modules and makes my card unusable for internet connection.

I've seen many posts ans issues about this card but I don't really know where to start due to all the differences and information gathered around the internet. Please, can someone help me with this? I'm really new into this kind of problems.

For more information, here is a link to my post on the arch linux forum which contains more information about the problem: [https://bbs.archlinux.org/viewtopic.php?id=304196](url)

Thank a lot beforehand :)

morrownr, 2025-03-16:
Hi @josepepito-upm 

What kernel are you using?

uname -r

josepepito-upm, 2025-03-16:
Hi @morrownr, thank for your help.

I'm using `6.13.6-arch1-1`, the latest one.

morrownr, 2025-03-16:
> 6.13.6-arch1-1

Why are you installing a driver? That kernel has the `rtw88_8822ce` driver. Your device should be plug and play.

josepepito-upm, 2025-03-17:
Yeah I know, it was because I saw a post where they installed the driver appart top see if they worked, but it didn't so I uninstalled them.

I think it's worth mentioning that over these days I tried disabling the power options in the drivers options, but it didn't do anything: 

in /etc/modprobe.d/rtw88_8822ce.conf:
```
options rtw_pci disable_aspm=y disable_msi=y
options rtw_core disable_lps_deep=y

options rtw88_pci disable_aspm=y disable_msi=y
options rtw88_core disable_lps_deep=y
```

Then i powered off and started the laptop again

morrownr, 2025-03-17:
G'day @josepepito-upm 

>  I'm really new into this kind of problems.

Before we continue, let's discuss a couple of things:

It sounds like you are fairly new to Linux? If so, is there a reason that you decided to use Arch?

I've run a github site for several years. The mission is primarily to help Linux users with information regarding the use of USB-WiFi but we get into wifi cards at times as well. Here is the Main Menu to the site:

https://github.com/morrownr/USB-WiFi

The reason I am telling you this is that over the years I have developed some opinions about the ease of use of many distros. In addition to the information that is available, I maintain some Realtek out-of-kernel drivers. I try to support as many Linux distros as possible but that is a challenge. Distros based on Red Hat are a challenge because of how Red Hat handles their kernels so I allow Red Hat devs to maintain the drivers if they want. Android more or less falls in the same category as does OpenWRT. Of the more mainstream desktop distros, my opinion is that Arch is probably not a good distro for beginners. I have had a few discussions with Arch folks about how AUR causes me problems. It had no effect of course. The problem is that I have carefully documented my drivers and made them as easy to install as I can. What AUR does is repackage my drivers, which basically hides my documentation, and then they post my repo as the source. What do Arch users do when they have a problem? They come to my site. I have no idea what their mess is because they did not use the instructions I prepared nor did they use my installer. I understand that Arch developed AUR to solve a problem but there is a core problem with AUR, it is a workaround instead of an effort to actually solved the underlying problem.

The core problem is that out-of-kernel drivers do not work well at all with Linux due to the development model. The development model is design to best take advantage of the resources that are available, Companies working with the appropriate kernel maintainers to include their drivers in the Linux kernel is best way to do drivers. My opinion is that Arch could best spend its time providing information to its users about what companies are doing driver development the way it should be. That way users could purchase hardware that works really well with Linux and makers of the products would notice what is selling and adjust accordingly. So, my opinion is that Arch is using a lot of resources toward something that is actually counter to supporting how things should be. That is my opinion and some may not agree with me. That is fine. My undergrad and grad degrees are in economics. I know how to influence markets and what Arch does really is counter to where we need to go.

Anyway, probably more than you wanted to know. Here is my opinion. New or inexperienced users would probably be better off starting with a distro like Linux Mint. Over time, if you want to get into programming or other more technical pursuits, you can progress on to other distros.

Now, back to the real topic: (I am starting at the top again as this thread is confusing)

> I'm experiencing some problems with realtek driver rtw_8822ce on arch linux.

This tells me that you have installed the driver in this repo or the AUR repackaged version of it or both as you seem to indicate later. The default that should have already been in place and working would be named rtw88_8822ce. The change of name for this repo is to avoid conflicts and allow the ability to blacklist.

> I came to the conclusion that is a driver's fault due to the kernel messages.

> ganxotatil kernel

I have no idea what that means. I would prefer something like 6.12.12-amd64. That would actually tell me something useful.

> rtw_8822ce 0000:03:00.0: WOW Firmware version 9.9.4

> rtw_8822ce 0000:03:00.0: Firmware version 9.9.15

That seems odd to me. I am used to seeing both firmware files have the same version number if there is more than one firmware file.

Can I get you to post the results of:

$ ls -l /lib/firmware/rtw88

$ rfkill list all

$ ls -l /usr/lib/modules/6.12.12+bpo-amd64/kernel/drivers/net/wireless/realtek/rtw88

Let's see what those 3 commands show and then we can proceed.

josepepito-upm, 2025-03-17:
Hi @morrownr, let's answer everything you asked me for:

> It sounds like you are fairly new to Linux? If so, is there a reason that you decided to use Arch?

Not really, I've been using Linux in my personal laptop for university for a year or so, and I'm also into daily driving it into my personal computer around half a year. I was trying to say that all of these problems of kernel and firmware and digging into logs and stuff was kind of new. Indeed, I reinstalled arch bc my other distro (fedora, which I installed after another arch installation broke due to a bad timeshift) broke and I decided to gave it another chance.

> I've run a github site for several years. The mission is primarily to help Linux users with information regarding the use of USB-WiFi but we get into wifi cards at times as well. Here is the Main [...]

Ok' I'll take a look at it and see if works and all the thing, mainly because some arch users on the forum told me to make a downgrade of the kernel and I really want to avoid that for the moment. 

> Anyway, probably more than you wanted to know. Here is my opinion. New or inexperienced users would probably be better off starting with a distro like Linux Mint. Over time, if you want to get into programming or other more technical pursuits, you can progress on to other distros.

I know, I've used other ones as well, but I mainly use arch die to all the customization and lightweight it is, because other alternatives like NixOS, Void Linux or other ones are kind of disappealing bc of thier philosophy and complexity, so ftm I'll try to stick with it.

> I have no idea what that means. I would prefer something like 6.12.12-amd64. That would actually tell me something useful.
I don't know where did you get that but that is the hostname, and as far as I know you are telling me about a kernel for the ARM architecture, which is different from mine (x86_64). Also, I uninstalle

> That seems odd to me. I am used to seeing both firmware files have the same version number if there is more than one firmware file.

I haven't notices, but it's strange bc these logs were copied after a fresh installation, so I don't know why there are 2 versions...

> Can I get you to post the results of:
```
[ganxo@ganxotatil network]$ ls -l /lib/firmware/rtw88
total 376
-rw-r--r-- 1 root root   599 Mar 11 23:06 README.zst
-rw-r--r-- 1 root root 13117 Mar 11 23:06 rtw8703b_fw.bin.zst
-rw-r--r-- 1 root root 14647 Mar 11 23:06 rtw8703b_wow_fw.bin.zst
-rw-r--r-- 1 root root 17918 Mar 11 23:06 rtw8723d_fw.bin.zst
-rw-r--r-- 1 root root 16756 Mar 11 23:06 rtw8812a_fw.bin.zst
-rw-r--r-- 1 root root 19570 Mar 11 23:06 rtw8821a_fw.bin.zst
-rw-r--r-- 1 root root 53954 Mar 11 23:06 rtw8821c_fw.bin.zst
-rw-r--r-- 1 root root 77712 Mar 11 23:06 rtw8822b_fw.bin.zst
-rw-r--r-- 1 root root 85925 Mar 11 23:06 rtw8822c_fw.bin.zst
-rw-r--r-- 1 root root 64058 Mar 11 23:06 rtw8822c_wow_fw.bin.zst
```
```
[ganxo@ganxotatil ~]$ rfkill list all 
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
```
```
[ganxo@ganxotatil ~]$ ls -l /usr/lib/modules/6.13.7-arch1-1/kernel/drivers/net/wireless/realtek/rtw88
total 936
-rw-r--r-- 1 root root  33332 Mar 13 19:12 rtw88_8703b.ko.zst
-rw-r--r-- 1 root root  18544 Mar 13 19:12 rtw88_8723cs.ko.zst
-rw-r--r-- 1 root root  18288 Mar 13 19:12 rtw88_8723de.ko.zst
-rw-r--r-- 1 root root  35534 Mar 13 19:12 rtw88_8723d.ko.zst
-rw-r--r-- 1 root root  18583 Mar 13 19:12 rtw88_8723ds.ko.zst
-rw-r--r-- 1 root root  18231 Mar 13 19:12 rtw88_8723du.ko.zst
-rw-r--r-- 1 root root  25029 Mar 13 19:12 rtw88_8723x.ko.zst
-rw-r--r-- 1 root root  31234 Mar 13 19:12 rtw88_8812a.ko.zst
-rw-r--r-- 1 root root  18062 Mar 13 19:12 rtw88_8812au.ko.zst
-rw-r--r-- 1 root root  29109 Mar 13 19:12 rtw88_8821a.ko.zst
-rw-r--r-- 1 root root  18076 Mar 13 19:12 rtw88_8821au.ko.zst
-rw-r--r-- 1 root root  18188 Mar 13 19:12 rtw88_8821ce.ko.zst
-rw-r--r-- 1 root root  40399 Mar 13 19:12 rtw88_8821c.ko.zst
-rw-r--r-- 1 root root  18439 Mar 13 19:12 rtw88_8821cs.ko.zst
-rw-r--r-- 1 root root  18352 Mar 13 19:12 rtw88_8821cu.ko.zst
-rw-r--r-- 1 root root  18181 Mar 13 19:12 rtw88_8822be.ko.zst
-rw-r--r-- 1 root root  48043 Mar 13 19:12 rtw88_8822b.ko.zst
-rw-r--r-- 1 root root  18424 Mar 13 19:12 rtw88_8822bs.ko.zst
-rw-r--r-- 1 root root  18630 Mar 13 19:12 rtw88_8822bu.ko.zst
-rw-r--r-- 1 root root  18207 Mar 13 19:12 rtw88_8822ce.ko.zst
-rw-r--r-- 1 root root  75299 Mar 13 19:12 rtw88_8822c.ko.zst
-rw-r--r-- 1 root root  18430 Mar 13 19:12 rtw88_8822cs.ko.zst
-rw-r--r-- 1 root root  18191 Mar 13 19:12 rtw88_8822cu.ko.zst
-rw-r--r-- 1 root root  33304 Mar 13 19:12 rtw88_88xxa.ko.zst
-rw-r--r-- 1 root root 158971 Mar 13 19:12 rtw88_core.ko.zst
-rw-r--r-- 1 root root  38876 Mar 13 19:12 rtw88_pci.ko.zst
-rw-r--r-- 1 root root  38615 Mar 13 19:12 rtw88_sdio.ko.zst
-rw-r--r-- 1 root root  33244 Mar 13 19:12 rtw88_usb.ko.zst
```
Here is everthing you asked for. Again, thanks for your time and dedication to this problem :)

morrownr, 2025-03-17:
Okay, so you posted the above from a clean Arch installation, right?

> -rw-r--r-- 1 root root 85925 Mar 11 23:06 rtw8822c_fw.bin.zst

> -rw-r--r-- 1 root root 64058 Mar 11 23:06 rtw8822c_wow_fw.bin.zst

Those are the 2 firmware files that are needed.

No blocking issues.

> -rw-r--r-- 1 root root  18207 Mar 13 19:12 rtw88_8822ce.ko.zst

> -rw-r--r-- 1 root root  75299 Mar 13 19:12 rtw88_8822c.ko.zst

There are your two driver files.

Is the 8822ce card working? It should be. I know, famous last words. 

Don't install any drivers. Let's figure out why it is not working, if it is not.

Post the results of:

$ iw dev

josepepito-upm, 2025-03-17:
> Okay, so you posted the above from a clean Arch installation, right?
> 
> >

Indeed, I have only changed what you told me, nothing more.

> Is the 8822ce card working? It should be. I know, famous last words.

As far as I concern yeah, when I try to connect to internet from a live ISO from any distribution it works flawlessly.

Here are the result of `iw dev`:
```
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr 74:97:79:fe:bf:13
		ssid pelukass
		type managed
		channel 11 (2462 MHz), width: 20 MHz, center1: 2462 MHz
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```

morrownr, 2025-03-18:
> when I try to connect to internet from a live ISO from any distribution it works flawlessly.

Great. The inability to connect to the internet was your original problem. I think we can call this issue fixed. Lets don't close it as others might learn from it.

> Interface wlan0 ...

That is a good looking wireless interface.

Run:

$ lsmod | grep 8822

Your should see `rtw88_8822ce` in use as that is the name of the primary driver that is in rtw88 in the kernel. As long as you are using perhaps kernel 6.12 or later, you should see good performance without the need to even install a driver. Enjoy.

josepepito-upm, 2025-03-18:
> Great. The inability to connect to the internet was your original problem. I think we can call this issue fixed. Lets don't close it as others might learn from it.

Why? I mean, the main problem is that I can't connect to internet after the installation, and I'd like to after that, not only on the installation medias. Do you think that if I change the kernel to an older version or a LTS kernel the problem would be solved?

Here is the output of `lsmod | grep 8822`
```
[ganxo@ganxotatil ~]$ lsmod | grep 8822
rtw88_8822ce           12288  0
rtw88_8822c           495616  1 rtw88_8822ce
rtw88_pci              40960  1 rtw88_8822ce
rtw88_core            331776  2 rtw88_pci,rtw88_8822c
```

morrownr, 2025-03-18:
I am confused again. You are saying that the original problem is still there but only on your fully installed laptop, right?

If so, what kernel and distro are you using on that laptop?

josepepito-upm, 2025-03-18:
> I am confused again. You are saying that the original problem is still there but only on your fully installed laptop, right?

Yes, the problem only happens when using arch linux, and as far as I am concerned there ain't any problems with other distros like fedora.

I'm  using this kernel: `6.13.6-arch1-1`, the same you asked me before, with arch linux

josepepito-upm, 2025-03-18:
> This can be confusing on my side. Remember I am not there with an eyes on view. Also, remember that on any given day, I may help several people and it can be a challenge to keep everything separate.

Yeah I can figure that out, I didn't mean to be rude. I thank you infinitely for all the help you are giving me to solve this.

> You have a laptop. It has a clean installation of Arch running kernel 6.13, right? If so, 6.13 is the kernel that came with the img you installed, right?

As far as a user from the arch forum told me , the img came with the 6.13.5 kernel, but when installing the latest one it installed the 6.13.6, as you can see on the uname -r command:

> I'm using this kernel: `6.13.6-arch1-1`, the same you asked me before, with arch linux.

> Kernel 6.13 has some problems with wifi. I have no idea what the status is now as I am only using and testing 6.12 and 6.14.
> 
> Is there another version of Arch with maybe kernel 6.12? If so, would it be okay to give it a go.

Yeah, I can downgrade the kernel with no issue at all. As a matter of fact, I've downgraded the kernel from 6.13.6 to 6.13.5, which was the one the installation media used.

> This does not look like a problem with the driver but rather something else in Arch or with kernel 6.13. You can tell with your testing that the rtw88_8822ce driver is good.

I think it's a problem with the new rev of the kernel 6.13 bc the same one worked on the live iso.

> Did any of the distros that you ran from the flash drive, use kernel 6.13? The ones from flash driver all worked, right?

Oh I'm sorry I forgot this one. I used a live iso of fedora and NixOS. As far as I know the fedora installation media right now uses the 6.12 I think, but I'm not sure. On the NixOS part, I have no clue.

Again, thanks for all the work you're putting on this issue to solve it.

morrownr, 2025-03-18:
Hmmm... there are a lot of rabbit holes we could go down but I think the next thing to try to for you to backoff to the last LTS kernel - 6.12)

Here is a video that goes over installing different kernels in arch:

https://www.youtube.com/watch?v=-1Ebx3eKpuU&t=196s

It looks like the term Longterm is what you are looking for.

josepepito-upm, 2025-03-18:
Ok, I bring an update for that: 

I tried with the kernel 6.13.5 instead of 6.13.6 and now everything works as expected, no issues at all. 
I'll try to lock the updates for the kernel and when the issue is resolved in the next release, I'll try to see if everything works. 

Thank a lot for your help and your time @morrownr! I'll close this issue because I think is really straightforward where to problem is. 

Have a nice day! :)

josepepito-upm, 2025-03-19:
I tried to install the newest kernel bc I had an update pending and the problem persisted, so I guess it's a problem that hasn't been patched yet. 

Also, I think the problem also happens on the latest LTS kernel, as I installed it on a CachyOS distro (arch-based distribution) and I still had the same problem. However, I need to experiment more with this as the kernels dropped with this distribution are slightly modified.

Then, I'll try to make more experiments with the lts kernel to see if it works or not and see where the problem is

Neo-29, 2025-03-21:
Hi.
I'm a the same card.
I can confirm at the moment that kernel 6.13.5 is ok.
The newest causes drop, loss of connection, slow speed on `rtw_8821ce`

josepepito-upm, 2025-03-22:
Hi, I'll bring an update on this because I thought I posted the message, but it seems that it didn't:

right now I'm using Linux Mint Debian Edition (which I think it uses the 6.1 kernel) and everything works fine. If I have any problem it's mainly caused because there's no important network activity on that moment and the network suspends, but turning off an on the wifi menu or docking and undocking the laptop fixes the issue right away. It doesn't always happen, but when it does I do that and problem solved. It is using the module `rtw_8822ce`.

The problem mentioned above it isn't something I loose the sleep with, mainly because it has happened to me before and at least I can use my laptop. Why I suddenly changed that? Well, I don't know why but after changing to the lts kernel back to the `6.13.5` one, the problem reappeared so I don't know what could have happened with that.

josepepito-upm, 2025-05-11:
I'll close this issue due to not having any further problems in the recent versions of the kernel `6.14.*`

closed 2025-05-11 by josepepito-upm