lwfinger/rtw88 #334: [rtw_8812au] Direct firmware load for rtw88/rtw8812a_fw.bin failed with error -2

gbrld, 2025-03-24:
Hi, I'm trying to use an Alfa AWUS036ACH USB attached RTL8812AU WiFi adapter with this driver, but receive the below error when the module loads.  I tried patch https://github.com/lwfinger/rtw88/issues/261#issuecomment-2517790993, but my system never reaches rtw_parse_efuse_map, so there's no debug output.

```
dmesg| grep -i rtw
[    3.945878] rtw_core: loading out-of-tree module taints kernel.
[    3.966823] rtw_8812au 1-1.4:1.0: Direct firmware load for rtw88/rtw8812a_fw.bin failed with error -2
[    3.966851] rtw_8812au 1-1.4:1.0: failed to request firmware
[    3.991108] rtw_8812au 1-1.4:1.0: failed to load firmware
[    3.991123] rtw_8812au 1-1.4:1.0: failed to setup chip efuse info
[    3.991127] rtw_8812au 1-1.4:1.0: failed to setup chip information
[    3.999869] rtw_8812au: probe of 1-1.4:1.0 failed with error -22
[    3.999975] usbcore: registered new interface driver rtw_8812au
```

```
lsusb| grep -i rtw
Bus 001 Device 003: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
```

dubhater, 2025-03-24:
-2 means "file not found". You are missing the firmware. You can find it in this repository or in the linux-firmware repository.

gbrld, 2025-03-24:
That did the trick, thanks!