lwfinger/rtw88 #340: [RTL8812AU][Linux Mint] DKMS installation may have missing steps

Dudly01, 2025-04-13:
I wanted to use an Alfa AWUS036ACHM (RTL8812AU chipset) with Linux Mint 22.1 (Linux Kernel 6.8.0-57-generic) and used the DKMS installation approach for later convenience. After following the installation steps (and rebooting), the Wifi adapter was not recognized. `lsusb` showed that the device is indeed available.

```
Bus 003 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
```

A comment on the [rtw88-dkms-git package's website](https://aur.archlinux.org/packages/rtw88-dkms-git) gave me some pointers:

> 
> If your wifi adapter doesn't work and you see a line like this when running `sudo dmesg | grep -i firmware`,
> ```
> [24672.978460] rtw_8814au 5-2:1.2: Direct firmware load for rtw88/rtw8814a_fw.bin failed with error -2
> ```
> that means your system lacks the firmware needed, try upgrading linux-firmware to 20250408.c1a774f3-1 or later.
> 
> If your distro doesn't provide these newer versions of linux-firmware, run these commands to install the firmware manually.
> ```
> git clone https://github.com/lwfinger/rtw88
> cd rtw88
> sudo make install_fw
> ```

Running `sudo dmesg | grep -i firmware` indeed showed the error.

```
[   17.183494] rtw_8812au 3-1:1.0: Direct firmware load for rtw88/rtw8812a_fw.bin failed with error -2
[   17.183503] rtw_8812au 3-1:1.0: failed to request firmware
[   17.190974] rtw_8812au 3-1:1.0: failed to load firmware
```

After installing the firmware with `sudo make install_fw` and rebooting the system, the error goes away.

```
[   12.447899] rtw_8812au 3-1:1.0: Firmware version 52.14.0, H2C version 0
```

`sudo airmon-ng` confirms that the device is recognised.

```
phy20	wlx00c0cab49cb7	rtw_8812au	Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
```

Based on some very minimal testing, the Wifi adapter works without issues. (As long as I do not try to connect to the same network with this and my built-in Wifi adapter.)

**Is this issue present indeed because of my Linux kernel version? Do I need to redo `sudo make install_fw` with each kernel update?**

a5a5aa555oo, 2025-04-15:
Yes, the step `sudo make install_fw` is missing.

> Do I need to redo sudo make install_fw with each kernel update?

You only need to do that step one time.

SijmenHuizenga, 2025-04-29:
Had the same problem. Learnt that `sudo make install_fw` needs to be run after `dkms build` and before `dkms install`.