lwfinger/rtw88 #341: Will RTL8188ETV/EU be supported?

Mastiffen, 2025-04-14:
I ran into a snag for an 8188ETV card, and for me it's not easy to replace the card. I have the USB wifi glued in place in my boat, as the AP for ESP32's and other stuff. You may or may not remember that I am using an Asus 8814 based card as the link to my cabin, where I had some signal issues. But this is something else, I do not get the card up at all. Does that mean that it is not supported, or is there a change I need to do somewhere? Or will it be supported?

dubhater, 2025-04-14:
rtw88 doesn't support that chip. rtl8xxxu supports it starting from kernel 6.3, but no AP mode.

Mastiffen, 2025-04-15:
Aha, I see. Thanks! Then I guess I will have to use the old driver and HostAPD.