lwfinger/rtw88 #343: [RTL8814AU]: Annoying issue that requires cutting power to system

Sammy-0, 2025-04-16:
Kernel: 6.14.2-arch1-1
Device: NetGear, Inc. Nighthawk A7000 802.11ac Wireless Adapter AC1900 [Realtek 8814AU]
Module: rtw_8814au

At seemingly random my networking will die and certain processes like NetworkManager will die or hang indefinitely, requiring a restart, because I can't work without internet, and trying to do things like kill NetworkManager directly causes the terminal to hang.

These processes also become immune to the system's SIGTERM and SIGKILL, which causes shutdowns or reboots to hang indefinitely. So while troubleshooting this issue I've had to shut it down the old fashioned way every time.

Here's the only journalctl entries I could find that seemed related:
```
Apr 15 20:59:03 dallas kernel: ---[ end trace 0000000000000000 ]---
Apr 15 20:59:03 dallas kernel:  </TASK>
Apr 15 20:59:03 dallas kernel:  common_startup_64+0x13e/0x141
Apr 15 20:59:03 dallas kernel:  start_secondary+0x11e/0x140
Apr 15 20:59:03 dallas kernel:  cpu_startup_entry+0x29/0x30
Apr 15 20:59:03 dallas kernel:  do_idle+0x1ad/0x210
Apr 15 20:59:03 dallas kernel:  cpuidle_enter+0x2d/0x40
Apr 15 20:59:03 dallas kernel: R13: 000001a5409ed666 R14: 0000000000000002 R15: 0000000000000000
Apr 15 20:59:03 dallas kernel: R10: 0000000000000008 R11: ffff9201cf13538c R12: ffffffffbc7f4b80
Apr 15 20:59:03 dallas kernel: RBP: ffff91fec0eb5000 R08: 0000000000000002 R09: 0000000000000025
Apr 15 20:59:03 dallas kernel: RDX: 000001a5409ed666 RSI: fffffffbaadf35c9 RDI: 0000000000000000
Apr 15 20:59:03 dallas kernel: RAX: ffff9201cf100000 RBX: 0000000000000002 RCX: 0000000000000000
Apr 15 20:59:03 dallas kernel: RSP: 0018:fffface7401efe80 EFLAGS: 00000246
Apr 15 20:59:03 dallas kernel: Code: 00 00 e8 bd 3f 13 ff e8 98 f1 ff ff 49 89 c5 0f 1f 44 00 00 31 ff e8 79 d0 11 ff 45 84 ff 0f 85 40 02 00 00 fb 0f 1f 44 00 00 <45>
 85 f6 0f 88 84 01 00 00 49 63 d6 48 8d 04 52 48 8d 04 82 49 8d
Apr 15 20:59:03 dallas kernel: RIP: 0010:cpuidle_enter_state+0xc6/0x420
Apr 15 20:59:03 dallas kernel:  asm_common_interrupt+0x26/0x40
Apr 15 20:59:03 dallas kernel:  <TASK>
Apr 15 20:59:03 dallas kernel:  </IRQ>
Apr 15 20:59:03 dallas kernel:  common_interrupt+0x85/0xa0
Apr 15 20:59:03 dallas kernel:  __irq_exit_rcu+0xc2/0xe0
Apr 15 20:59:03 dallas kernel:  handle_softirqs+0xeb/0x2b0
Apr 15 20:59:03 dallas kernel:  tasklet_action_common+0xcf/0x240
Apr 15 20:59:03 dallas kernel:  ieee80211_handle_queued_frames+0xbe/0xd0 [mac80211 682e229732a6bfe53cb0bbcde81ec2801d27c374]
Apr 15 20:59:03 dallas kernel:  ieee80211_rx_napi+0x54/0xe0 [mac80211 682e229732a6bfe53cb0bbcde81ec2801d27c374]
Apr 15 20:59:03 dallas kernel:  <IRQ>
Apr 15 20:59:03 dallas kernel: Call Trace:
Apr 15 20:59:03 dallas kernel: CR2: 00003d7c011ed000 CR3: 00000001599d6000 CR4: 00000000003506f0
Apr 15 20:59:03 dallas kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Apr 15 20:59:03 dallas kernel: FS:  0000000000000000(0000) GS:ffff9201cf100000(0000) knlGS:0000000000000000
Apr 15 20:59:03 dallas kernel: R13: fffface74036cee0 R14: 0000000000000000 R15: 0000000000000000
Apr 15 20:59:03 dallas kernel: R10: d81916e70f3c6ca3 R11: ffffffffc1817e50 R12: 0000000000000000
Apr 15 20:59:03 dallas kernel: RBP: ffff91fee5140900 R08: 0000000000000000 R09: ffff91fec1ae6bba
Apr 15 20:59:03 dallas kernel: RDX: 0000000000000001 RSI: 0000000000000004 RDI: 0000000000000000
Apr 15 20:59:03 dallas kernel: RAX: 0000000000020102 RBX: ffff91ffeaeb3600 RCX: 0000000000000000
Apr 15 20:59:03 dallas kernel: RSP: 0018:fffface74036ce00 EFLAGS: 00010202
Apr 15 20:59:03 dallas kernel: Code: ff ff 0f b6 73 4a 8d 4e ff 80 f9 07 0f 87 fe fa ff ff e9 7a ff ff ff 41 8d 76 04 c7 04 24 04 00 00 00 39 ce 0f 82 62 fb ff ff <0f> 0b 48 89
 df e8 d2 84 fb f8 e9 4e fd ff ff 48 89 54 24 58 4c 89
Apr 15 20:59:03 dallas kernel: RIP: 0010:ieee80211_rx_list+0x7a4/0xbd0 [mac80211]
Apr 15 20:59:03 dallas kernel: Hardware name: ###########################################################
Apr 15 20:59:03 dallas kernel: Tainted: [D]=DIE, [W]=WARN, [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
Apr 15 20:59:03 dallas kernel: CPU: 14 UID: 0 PID: 0 Comm: swapper/14 Tainted: G      D W  OE      6.14.2-arch1-1 #1 51440b8a0cc8bb91764dac94f6c2b53455e5a907
Apr 15 20:59:03 dallas kernel:  crypto_user loop dm_mod nfnetlink zram 842_decompress 842_compress lz4hc_compress lz4_compress ip_tables x_tables hid_generic nvme nvme_core sr_
mod nvme_auth usbhid cdrom wmi
Apr 15 20:59:03 dallas kernel: Modules linked in: udf crc_itu_t snd_ctl_led uinput ccm snd_seq_dummy snd_hrtimer snd_seq vfat fat amd_atl intel_rapl_msr intel_rapl_common rtw_8
821ce(OE) snd_hda_codec_realtek rtw_8814au(OE) rtw_8821c(OE) snd_hda_codec_generic rtw_8814a(OE) polyval_clmulni snd_hda_scodec_component snd_hda_codec_hdmi polyval_generic rtw_pci(OE) rtw_usb(OE) snd_usb_audio s
nd_hda_intel ghash_clmulni_intel sha512_ssse3 btusb snd_intel_dspcfg snd_usbmidi_lib rtw_core(OE) sha256_ssse3 btrtl snd_intel_sdw_acpi pkcs8_key_parser snd_hda_codec snd_ump mac80211 ee1004 sha1_ssse3 nvidia_drm
(OE) btintel snd_hda_core snd_rawmidi aesni_intel btbcm libarc4 snd_hwdep snd_seq_device nvidia_uvm(OE) crypto_simd btmtk nvidia_modeset(OE) hp_wmi r8169 snd_pcm cryptd platform_profile sp5100_tco rapl cfg80211 b
luetooth mc sparse_keymap snd_timer wmi_bmof drm_ttm_helper realtek acpi_cpufreq pcspkr mdio_devres i2c_piix4 snd ttm k10temp gpio_amdpt joydev mousedev i2c_smbus soundcore video libphy ccp rfkill gpio_generic ac
pi_tad mac_hid nvidia(OE)
Apr 15 20:59:03 dallas kernel: WARNING: CPU: 14 PID: 0 at net/mac80211/rx.c:808 ieee80211_rx_list+0x7a4/0xbd0 [mac80211]
Apr 15 20:59:03 dallas kernel: ------------[ cut here ]------------
Apr 15 20:58:23 dallas NetworkManager[650]: <info>  [1744775903.8505] device (p2p-dev-wlan0): supplicant management interface state: interface_disabled -> disconnected
Apr 15 20:58:23 dallas NetworkManager[650]: <info>  [1744775903.8504] device (wlan0): supplicant interface state: interface_disabled -> disconnected
Apr 15 20:58:23 dallas NetworkManager[650]: <info>  [1744775903.8351] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> interface_disabled
Apr 15 20:58:23 dallas NetworkManager[650]: <info>  [1744775903.8350] device (wlan0): supplicant interface state: disconnected -> interface_disabled
Apr 15 20:58:23 dallas NetworkManager[650]: <info>  [1744775903.3976] device (wlan0): set-hw-addr: set MAC address to ################# (scanning

```

Sidenote: RTL8821CE is slow as hell on Linux. What gives?

dubhater, 2025-04-16:
That portion of the log only contains a warning, which can't cause the unkillable processes. There must be something more serious in the journal.

Sammy-0, 2025-04-16:
Those were the only entries prior to the connection dying that were somewhat related that I saw, unfortunately.

dubhater, 2025-04-16:
Maybe show the rest of it anyway.

Sammy-0, 2025-04-24:
Unfortunately still happening, and the journal is as undescriptive as ever. 

#344 
This issue seems very similar. I'll close this until that one gets closed for whatever reason. 
Thanks anyway

dubhater, 2025-04-24:
That issue is fixed now.

dubhater, 2025-04-24:
Try it now, at least the warning should be gone.

Sammy-0, 2025-04-24:
Okay this time it died and I had to replug it to get it to show up. Will wait to see if it locks up again.
Here's what dmesg spit out:
[ 2625.667039] rtw_8814au 4-2:1.0: read register 0x60 failed with -71
[ 2625.693044] rtw_8814au 4-2:1.0: read register 0x60 failed with -71
[ 2625.719045] rtw_8814au 4-2:1.0: read register 0x60 failed with -71
[ 2625.745049] rtw_8814au 4-2:1.0: read register 0x60 failed with -71
[ 2627.076483] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2629.060340] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2631.044188] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2633.092046] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2635.075878] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2637.059731] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2639.107577] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2641.091429] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2643.075292] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2645.059128] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2647.042980] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2649.090822] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2651.074684] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2653.058535] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2655.106369] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2657.091217] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2659.074086] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2661.057922] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2663.105770] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2665.089618] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2667.073468] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2669.057315] rtw_8814au 4-2:1.0: failed to get tx report from firmware
[ 2670.096420] rtw_8814au 4-2:1.0: write register 0x3 failed with -71
[ 2670.096662] rtw_8814au 4-2:1.0: write register 0x80 failed with -71
[ 2670.096902] rtw_8814au 4-2:1.0: write register 0x81 failed with -71

dubhater, 2025-04-24:
What if you disable the LED blinking early on?

```sh
echo none > /sys/class/leds/rtw88*/trigger
```

Sammy-0, 2025-04-26:
Nope, still seems to happen. What I noticed is that it's not consistent, so it may be dependent on usage and the amount of traffic I put through it, dunno.

dubhater, 2025-04-28:
The commit from the other day was not good. Please try again. Is the WARNING still gone?

Sammy-0, 2025-05-15:
Sorry for the wait, been busy and haven't been wanting to deal with my computer breaking randomly.
Yes, the warning is gone from what I can tell, but the issue with the system locking up is still present.

Sammy-0, 2025-05-15:
I should've added that the old repo: https://github.com/morrownr/8814au used to work fine on my machine with absolutely no issue.

dubhater, 2025-05-15:
No worries.

Here's something you could try:

```diff
diff --git a/drivers/net/wireless/realtek/rtw88/usb.c b/drivers/net/wireless/realtek/rtw88/usb.c
index 43759d34b700..a9df3ed88606 100644
--- a/usb.c
+++ b/usb.c
@@ -979,7 +979,7 @@ static int rtw_usb_init_rx(struct rtw_dev *rtwdev)
 	struct sk_buff *rx_skb;
 	int i;
 
-	rtwusb->rxwq = alloc_workqueue("rtw88_usb: rx wq", WQ_BH, 0);
+	rtwusb->rxwq = create_singlethread_workqueue("rtw88_usb: rx wq");
 	if (!rtwusb->rxwq) {
 		rtw_err(rtwdev, "failed to create RX work queue\n");
 		return -ENOMEM;
```

This is just for testing. I'm hoping your computer won't freeze anymore and you'll see something useful in dmesg.

Sammy-0, 2025-05-16:
Nope, sorry. Same thing. :(

I got a few entries about skipping big packets. But again, nothing concrete.

dubhater, 2025-05-16:
In that case I guess you need [netconsole](https://docs.kernel.org/networking/netconsole.html) to see what goes wrong.

How big are the skipped packets?

Sammy-0, 2025-08-15:
The version of this that was built into the kernel worked fine for a couple months up until yesterday when I updated to the latest kernel, now this issue is back again. :/

Doesn't seem like netconsole showed anything either unfortunately, still have nothing I can give you to weed this out. Whole thing just craps out after a while.