lwfinger/rtw88 #344: System lock-up after plugging in adapter.

merll002, 2025-04-18:
I have installed the driver successfully on my Raspberry Pi Zero W but after rebooting, when plugging in the USB WIFI adapter (TP-Link T2U Plus), all sudo commands stop working - I can't use CTRL+C to interrupt after running a sudo command - I can't SSH, login VIA a TTY, or shutdown.

Here is the dmesg log:

<details>

<summary>dmesg log</summary>

```log
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.6.62+rpt-rpi-v6 (serge@raspberrypi.com) (gcc-12 (Raspbian 12.2.0-14+rpi1) 12.2.0, GNU ld (GNU Binutils for Raspbian) 2.40) #1 Raspbian 1:6.6.62-1+rpt1 (2024-11-25)
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] random: crng init done
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Reserved memory: created CMA memory pool at 0x0e400000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0e400000..0x1e3fffff (262144 KiB) map reusable linux,cma
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000001effffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001effffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001effffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 cgroup_disable=memory snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=B8:27:EB:D5:1A:95 vc_mem.mem_base=0x1fa00000 vc_mem.mem_size=0x20000000  console=ttyS0,115200 console=tty1 root=PARTUUID=a3486cb0-02 rootfstype=ext4 fsck.repair=yes rootwait modules-load=dwc2,g_ether
[    0.000000] cgroup: Disabling memory control group subsystem
[    0.000000] Unknown kernel command line parameters "modules-load=dwc2,g_ether", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 125860
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] Memory: 213880K/507904K available (10033K kernel code, 1478K rwdata, 3124K rodata, 460K init, 581K bss, 31880K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 35022 entries in 103 pages
[    0.000000] ftrace: allocated 103 pages with 5 groups
[    0.000000] trace event string verifier disabled
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000006] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000061] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000152] bcm2835: system timer (irq = 27)
[    0.000967] Console: colour dummy device 80x30
[    0.001005] printk: console [tty1] enabled
[    0.001794] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060294] CPU: Testing write buffer coherency: ok
[    0.060408] pid_max: default: 32768 minimum: 301
[    0.060582] LSM: initializing lsm=capability,integrity
[    0.060899] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.060974] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.064003] RCU Tasks Rude: Setting shift to 0 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=1.
[    0.064341] RCU Tasks Trace: Setting shift to 0 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=1.
[    0.064644] Setting up static identity map for 0x8220 - 0x8258
[    0.065984] devtmpfs: initialized
[    0.079217] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.079641] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.079737] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.115407] pinctrl core: initialized pinctrl subsystem
[    0.117174] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.119760] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.126974] audit: initializing netlink subsys (disabled)
[    0.127530] audit: type=2000 audit(0.120:1): state=initialized audit_enabled=0 res=1
[    0.128659] thermal_sys: Registered thermal governor 'step_wise'
[    0.129030] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.129126] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.129600] Serial: AMBA PL011 UART driver
[    0.139114] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.160751] raspberrypi-firmware soc:firmware: Attached to firmware from 2024-11-26T12:54:33, variant start_cd
[    0.170779] raspberrypi-firmware soc:firmware: Firmware hash is 2ae30f53898ae2f1ba77ff570a92991bedfb0398
[... 501 lines cut ...]
[ 2455.570689] 1fc0: 00120488 00120540 b6ed0000 00000128 bee65b54 00000001 00000003 00120540
[ 2455.570741] 1fe0: b6ed00d4 bee65b08 b6eafe34 b6e241cc
[ 2455.570769] INFO: task (udev-worker):4214 blocked for more than 245 seconds.
[ 2455.570795]       Tainted: G         C O       6.6.62+rpt-rpi-v6 #1 Raspbian 1:6.6.62-1+rpt1
[ 2455.570813] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 2455.570826] task:(udev-worker)   state:D stack:0     pid:4214  ppid:201    flags:0x00800005
[ 2455.570890]  __schedule from schedule+0x54/0xec
[ 2455.570947]  schedule from schedule_preempt_disabled+0x14/0x18
[ 2455.571006]  schedule_preempt_disabled from __mutex_lock.constprop.0+0x248/0x514
[ 2455.571058]  __mutex_lock.constprop.0 from rtnetlink_rcv_msg+0x20c/0x2fc
[ 2455.571130]  rtnetlink_rcv_msg from netlink_rcv_skb+0xc0/0x120
[ 2455.571185]  netlink_rcv_skb from netlink_unicast+0x18c/0x26c
[ 2455.571240]  netlink_unicast from netlink_sendmsg+0x1f0/0x480
[ 2455.571286]  netlink_sendmsg from __sock_sendmsg+0x44/0x78
[ 2455.571352]  __sock_sendmsg from __sys_sendto+0x118/0x15c
[ 2455.571402]  __sys_sendto from __sys_trace_return+0x0/0x10
[ 2455.571436] Exception stack(0xdfac9fa8 to 0xdfac9ff0)
[ 2455.571476] 9fa0:                   bee0956c 00000080 00000006 0243aae8 00000020 00000000
[ 2455.571507] 9fc0: bee0956c 00000080 02417648 00000122 0243eb78 0243eb78 00000005 00000000
[ 2455.571531] 9fe0: 00000001 bee094f8 004eac44 b6d24290
[ 2455.571550] Future hung task reports are suppressed, see sysctl kernel.hung_task_warnings

```

</details>

ZerBea, 2025-04-19:
I can confirm this. The entire USB subsystem freezes.
More information is here:
https://github.com/ZerBea/hcxdumptool/discussions/361#discussioncomment-12866068
and here:
https://github.com/morrownr/8821au-20210708/issues/133#issuecomment-2813688315

merll002, 2025-04-19:
Do you have an idea of when this will be fixed?

morrownr, 2025-04-19:
@merll002 

> Do you have an idea of when this will be fixed?

Need to figure out what the problem is first. I am planning to burn a clean sd card card for my raspi4b this afternoon. If appears that you are using the 32 bit version of RasPiOS?

If you RasPi is capable of 64 bit, would you consider burning a sd card with the current 64 bit version of RaspPiOS so as to test?

As you know, I tested on Ubuntu 24.10 and did not see what you are seeing. We are just going to have to figure out what the problem is.

morrownr, 2025-04-19:
@merll002 @ZerBea 

I just spent the last 45 minutes burning an sd card for my RasPi4B. I burned the 64 bit RaspBerry Pi OS and then updated it. I then installed from this repo using the Basic Installation. My USB adapter is an Alfa ACS, rtl8811au based.

I have beat it up at this point and can find no problems whatsoever.

rtw_8821au 1-1.4:1.0: Firmware version 42.4.0, H2C version 0

Given that I tested on Ubuntu 24.10 yesterday found no problems, I am pondering whether this is a driver problem or a distro problem. One way or another, it would be good to get this figured out but more info is needed. Maybe you guys can compare notes to see if we can figure out what direction to go.

dubhater, 2025-04-19:
@ZerBea Your report has a clue about why things hang:

```
[  809.256440] rtw_8821au 5-2.1:1.0: Firmware version 42.4.0, H2C version 0
[  812.759142] rtw_8821au 5-2.1:1.0 wlp48s0f4u2u1: renamed from wlan0
[  837.315388] rtw_8821au 1-4:1.0: write register 0x1ef4 failed with -110
[  867.524259] rtw_8821au 1-4:1.0: write register 0x1ef8 failed with -110
[  868.930976] rtw_8821au 5-2.1:1.0 wlp48s0f4u2u1: entered promiscuous mode
[  897.730952] rtw_8821au 1-4:1.0: write register 0x1efc failed with -110
```

Each of those register writes takes 30 seconds to fail. From 0x1ef4 to 0x2000 there are about 67 writes, times 30 seconds each equals 2010 seconds.

@merll002 Try it now, I pushed something that should help with the lock-up.

morrownr, 2025-04-19:
@merll002 

Do you know how to use `git pull` to update the code on your system?

ZerBea, 2025-04-20:
@dubhater Thanks.
I'm not at home until 6th of May.I'll test it (TP-Link T2U Plus - 8821au) when I'm back home.

merll002, 2025-04-20:
I'm out all day today, so I might have to test this evening or tomorrow... Thanks for your help!

merll002, 2025-04-20:
Also, although this is unrelated, the same issue sometimes occurs when the adapter is in monitor mode and NetworkManager is running. (I've observed this on multiple PCs)

morrownr, 2025-04-22:
@dubhater 

For what it is worth, I have been continuing to test with the patch you merged and I see no problem...however, I was not seeing the problem in the first place so we'll have to wait for @merll002 and @ZerBea to test.

merll002, 2025-04-22:
Sorry for the delay... Compiling now. (Will take a few hours)

merll002, 2025-04-22:
Is `sudo make install_fw` required?

merll002, 2025-04-22:
**UPDATE:** Driver compiled, installed and working with both bands and in monitor mode with the TP-Link T2U Plus network adapter!
Thanks @morrownr and @dubhater!

One caveat: When using `airmon-ng` to enable monitor mode on the adapter, the new interface is created (`wlan1mon`) but the original is not deleted (`wlan1`). This might not be a problem with the driver but I'm mentioning it just in case. (Will do some research)

morrownr, 2025-04-22:
@merll002 

> One caveat: When using airmon-ng to enable monitor mode on the adapter, the new interface is created (wlan1mon) but the original is not deleted (wlan1). 

That is how it is supposed to work. VIF. The Realtek out of kernel drivers never quite got that right. 

Also, if you decide to get a WiFi 6 adapter you'll need to get a Mediatek based adapter because the only available drivers still don't get that right.

merll002, 2025-04-22:
Thanks! 
Should I close this issue?

PS: What does the `install_fw` target do exactly?

morrownr, 2025-04-22:
> Should I close this issue?

No

> PS: What does the install_fw target do exactly?

It installs the firmware. Since this driver is actually just here to allow for dev and testing, it needs to be made like drivers in the kernel. That means the firmware and driver files go in different places. Keep in mind that once you get to kernel 6.14 or later, you do not need this driver anymore.

Thanks for reporting and testing.

naalit, 2025-04-22:
I'm having the same problem (on kernel `6.14.3-arch1-1.1`), using the newest Git version didn't fix it - still causes anything touching USB or network to take forever, and it sees networks but can't actually connect to them. I'm using an Archer T2U Nano, 8811au.

naalit, 2025-04-22:
Hmm, it's possible my issue is slightly different, actually - dmesg looks like:
```
[  635.778750] rtw_8821au 1-12:1.0: Firmware version 42.4.0, H2C version 0
[  663.265131] usbcore: registered new interface driver rtw_8821au
[  663.267328] rtw_8821au 1-12:1.0 wlp2s0f0u12: renamed from wlan0
[  834.515764] rtw_8821au 1-12:1.0: write register 0x1d50 failed with -71
```
(not repeated, it ends here, unless I unplug and replug the adapter.) And `dmesg`/`ip link`/anything using USB does run *eventually*, it just takes >30s. Similarly, it takes 30s to open settings, 30s for it to register the wifi adapter, 30s for it to search for networks, then it keeps trying to connect but after 30s or so it says "failed to activate network connection" and tries again.

dubhater, 2025-04-22:
@naalit What kind of computer is it? If you plug the T2U Nano 10 times in a row do you get this problem every time?

I pushed something that may help. Please test again.

naalit, 2025-04-22:
Okay, on the newest revision the register writes only fail when I disconnect it, and commands sometimes hang for a bit but it's less often and only like ~5s. So dmesg when unplugging and replugging a few times:
```
[    9.514631] rtw_8821au 1-12:1.0: Firmware version 42.4.0, H2C version 0
[   14.522028] usbcore: registered new interface driver rtw_8821au
[   14.529784] rtw_8821au 1-12:1.0 wlp2s0f0u12: renamed from wlan0
[  347.613087] rtw_8821au 1-12:1.0: write register 0xb0 failed with -71
[  349.403202] rtw_8821au 1-12:1.0: Firmware version 42.4.0, H2C version 0
[  354.417701] rtw_8821au 1-12:1.0 wlp2s0f0u12: renamed from wlan0
[  458.178391] rtw_8821au 1-12:1.0: Firmware version 42.4.0, H2C version 0
[  463.196121] rtw_8821au 1-12:1.0 wlp2s0f0u12: renamed from wlan0
[  520.678846] rtw_8821au 1-12:1.0: write register 0x668 failed with -71
[  526.113355] rtw_8821au 1-12:1.0: Firmware version 42.4.0, H2C version 0
[  531.138040] rtw_8821au 1-12:1.0 wlp2s0f0u12: renamed from wlan0
[  550.469902] rtw_8821au 1-12:1.0: read register 0x8ac failed with -71
[  557.898615] rtw_8821au 1-12:1.0: Firmware version 42.4.0, H2C version 0
[  562.915615] rtw_8821au 1-12:1.0 wlp2s0f0u12: renamed from wlan0
```
The register writes and reads failing are only when I unplug it. It still can't connect to networks though, it gets stuck on "configuring" and then fails and tries again.

naalit, 2025-04-22:
Interestingly, if I try to use the adapter to make a hotspot (while USB tethering from my phone), I can see it from other devices but not actually successfully connect.

dubhater, 2025-04-22:
Check dmesg and/or journalctl without filtering them.

naalit, 2025-04-22:
Okay, here's dmesg when trying to connect to a network:
```
[ 1604.120686] usb 1-12: new high-speed USB device number 15 using xhci_hcd
[ 1604.333982] usb 1-12: New USB device found, idVendor=2357, idProduct=011e, bcdDevice= 2.00
[ 1604.333986] usb 1-12: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1604.333988] usb 1-12: Product: 802.11ac WLAN Adapter 
[ 1604.333989] usb 1-12: Manufacturer: Realtek 
[ 1604.333991] usb 1-12: SerialNumber: 00e04c000001
[ 1604.366196] rtw_8821au 1-12:1.0: Firmware version 42.4.0, H2C version 0
[ 1609.384573] rtw_8821au 1-12:1.0 wlp2s0f0u12: renamed from wlan0
[ 1687.281527] wlp2s0f0u12: authenticate with 34:8a:12:e9:4a:d1 (local address=8c:90:2d:a2:9b:94)
[ 1689.287704] wlp2s0f0u12: send auth to 34:8a:12:e9:4a:d1 (try 1/3)
[ 1689.288981] wlp2s0f0u12: authenticated
[ 1689.290551] wlp2s0f0u12: associate with 34:8a:12:e9:4a:d1 (try 1/3)
[ 1689.414964] wlp2s0f0u12: RX AssocResp from 34:8a:12:e9:4a:d1 (capab=0x11 status=0 aid=4)
[ 1689.770755] wlp2s0f0u12: associated
[ 1690.219164] wlp2s0f0u12: deauthenticating from 34:8a:12:e9:4a:d1 by local choice (Reason: 3=DEAUTH_LEAVING)
[ 1713.206204] wlp2s0f0u12: authenticate with 34:8a:12:e9:4a:d1 (local address=8c:90:2d:a2:9b:94)
[ 1715.123370] wlp2s0f0u12: send auth to 34:8a:12:e9:4a:d1 (try 1/3)
[ 1715.218068] wlp2s0f0u12: aborting authentication with 34:8a:12:e9:4a:d1 by local choice (Reason: 3=DEAUTH_LEAVING)
```
There's also this error when I tried the hotspot thing:
```
[ 1441.370242] rtw_8821au 1-12:1.0: error beacon valid
[ 1441.376231] rtw_8821au 1-12:1.0: failed to download drv rsvd page
[ 1441.376234] rtw_8821au 1-12:1.0: failed to download beacon
```

dubhater, 2025-04-22:
So it's connecting and then disconnecting half a second later:

```
[ 1689.770755] wlp2s0f0u12: associated
[ 1690.219164] wlp2s0f0u12: deauthenticating from 34:8a:12:e9:4a:d1 by local choice (Reason: 3=DEAUTH_LEAVING)
```

You should see these two messages in journalctl as well, with something more between them. What's between them may explain why it's disconnecting.

Those errors you get with the hotspot are probably unrelated.

naalit, 2025-04-22:
Here's the full `journalctl -r` output:
```
Apr 22 17:01:25 archbook NetworkManager[603]: <info>  [1745359285.6750] device (wlp2s0f0u12): set-hw-addr: set MAC address to 26:00:5B:8E:AA:70 (scanning)
Apr 22 17:01:25 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-DSCP-POLICY clear_all
Apr 22 17:01:25 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-DISCONNECTED bssid=34:8a:12:e9:4a:d1 reason=3 locally_generated=1
Apr 22 17:01:25 archbook wpa_supplicant[1552]: nl80211: send_event_marker failed: Source based routing not supported
Apr 22 17:01:24 archbook kernel: wlp2s0f0u12: aborting authentication with 34:8a:12:e9:4a:d1 by local choice (Reason: 3=DEAUTH_LEAVING)
Apr 22 17:01:24 archbook NetworkManager[603]: <info>  [1745359284.9985] device (wlp2s0f0u12): state change: failed -> disconnected (reason 'none', managed-type: 'full')
Apr 22 17:01:24 archbook NetworkManager[603]: <warn>  [1745359284.9984] device (wlp2s0f0u12): Activation: failed for connection 'eduroam'
Apr 22 17:01:24 archbook NetworkManager[603]: <info>  [1745359284.9984] manager: NetworkManager state is now DISCONNECTED
Apr 22 17:01:24 archbook NetworkManager[603]: <info>  [1745359284.9983] device (wlp2s0f0u12): state change: config -> failed (reason 'ssid-not-found', managed-type: 'full')
Apr 22 17:01:24 archbook NetworkManager[603]: <warn>  [1745359284.9983] device (wlp2s0f0u12): Activation: (wifi) association took too long, failing activation
Apr 22 17:01:24 archbook kernel: wlp2s0f0u12: send auth to 34:8a:12:e9:4a:d1 (try 1/3)
Apr 22 17:01:24 archbook NetworkManager[603]: <info>  [1745359284.9429] device (p2p-dev-wlp2s0f0u12): supplicant management interface state: scanning -> authenticating
Apr 22 17:01:24 archbook NetworkManager[603]: <info>  [1745359284.9428] device (wlp2s0f0u12): supplicant interface state: scanning -> authenticating
Apr 22 17:01:23 archbook kernel: wlp2s0f0u12: authenticate with 34:8a:12:e9:4a:d1 (local address=8c:90:2d:a2:9b:94)
Apr 22 17:01:16 archbook wpa_supplicant[1552]: wlp2s0f0u12: SME: Trying to authenticate with 34:8a:12:e9:4a:d1 (SSID='eduroam' freq=5240 MHz)
Apr 22 17:01:15 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-REGDOM-BEACON-HINT after freq=5805 max_tx_power=2000
Apr 22 17:01:15 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-REGDOM-BEACON-HINT before freq=5805 max_tx_power=2000 no_ir=1
Apr 22 17:01:15 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
Apr 22 17:01:15 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-REGDOM-BEACON-HINT after freq=5785 max_tx_power=2000
Apr 22 17:01:15 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-REGDOM-BEACON-HINT before freq=5785 max_tx_power=2000 no_ir=1
Apr 22 17:01:15 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
Apr 22 17:01:15 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-REGDOM-BEACON-HINT after freq=5765 max_tx_power=2000
Apr 22 17:01:15 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-REGDOM-BEACON-HINT before freq=5765 max_tx_power=2000 no_ir=1
Apr 22 17:01:15 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
Apr 22 17:01:11 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-REGDOM-BEACON-HINT after freq=5240 max_tx_power=2000
Apr 22 17:01:11 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-REGDOM-BEACON-HINT before freq=5240 max_tx_power=2000 no_ir=1
Apr 22 17:01:11 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
Apr 22 17:01:10 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-REGDOM-BEACON-HINT after freq=5220 max_tx_power=2000
Apr 22 17:01:10 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-REGDOM-BEACON-HINT before freq=5220 max_tx_power=2000 no_ir=1
Apr 22 17:01:10 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
Apr 22 17:01:07 archbook NetworkManager[603]: <info>  [1745359267.4646] device (p2p-dev-wlp2s0f0u12): supplicant management interface state: disconnected -> scanning
Apr 22 17:01:07 archbook NetworkManager[603]: <info>  [1745359267.4645] device (wlp2s0f0u12): supplicant interface state: disconnected -> scanning
Apr 22 17:01:00 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Apr 22 17:01:00 archbook NetworkManager[603]: <info>  [1745359260.9398] device (p2p-dev-wlp2s0f0u12): supplicant management interface state: associated -> disconnected
Apr 22 17:01:00 archbook NetworkManager[603]: <info>  [1745359260.9398] device (wlp2s0f0u12): supplicant interface state: associated -> disconnected
Apr 22 17:01:00 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-DSCP-POLICY clear_all
Apr 22 17:01:00 archbook wpa_supplicant[1552]: wlp2s0f0u12: CTRL-EVENT-DISCONNECTED bssid=34:8a:12:e9:4a:d1 reason=3 locally_generated=1
Apr 22 17:01:00 archbook wpa_supplicant[1552]: nl80211: send_event_marker failed: Source based routing not supported
Apr 22 17:01:00 archbook NetworkManager[603]: <info>  [1745359260.0004] Config: added 'proactive_key_caching' value '1'
Apr 22 17:01:00 archbook NetworkManager[603]: <info>  [1745359260.0004] Config: added 'identity' value 'dirkswagers@carleton.edu'
Apr 22 17:01:00 archbook NetworkManager[603]: <info>  [1745359260.0003] Config: added 'phase2' value 'auth=MSCHAPV2'
[... cut ...]
Apr 22 17:00:27 archbook NetworkManager[603]: <info>  [1745359227.7791] device (wlp2s0f0u12): Activation: starting connection 'eduroam' (aac57064-bbda-480e-84f8-7e2ac44d4fb8)
Apr 22 17:00:27 archbook systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
Apr 22 17:00:25 archbook wpa_supplicant[1552]: wlp2s0f0u12: Reject scan trigger since one is already pending
Apr 22 17:00:17 archbook systemd[1]: Started Network Manager Script Dispatcher Service.
Apr 22 17:00:17 archbook systemd[1]: Starting Network Manager Script Dispatcher Service...
```
Looks like it timed out associating?

dubhater, 2025-04-22:
Hmm, I wonder if it's something to do with the eduroam thing. Can you create a hotspot on your phone and connect to that?

naalit, 2025-04-22:
Oh interesting, that did actually work! It didn't when I tried before so I didn't bother trying again. Any idea what the problem would be with eduroam specifically?

dubhater, 2025-04-22:
I guess something about the encryption. Unfortunately I don't know what to do about it right now. Please open a new issue so the problem doesn't get lost.

naalit, 2025-04-22:
Oh, now it stopped connecting to anything again. Not sure why, will try again in an hour

naalit, 2025-04-23:
I also wonder if it might be timing related, since the journalctl output mentions the wifi requesting new secrets when it takes too long to associate - connecting to my phone hotspot or my university guest wifi works but takes significantly longer than it does on a different wifi card (30s-1min instead of ~instant). Maybe if the connection takes too long eduroam rejects it for some sort of security reason?

rabidcopy, 2025-04-27:
Haven't been able to test any recent merges here, but did also want to say I've also had system freeze issues when using the current in-kernel driver and this repo's driver specifically for the Archer T2U Plus. The strange thing is that before the in-kernel/rtw88 updates, this commit https://github.com/morrownr/8812au-20210820/commit/c01259932bfdee01e871b1468d4e75153fef1a12 was what worked for me with no system hanging or freezing. (And still works on more recent kernels that it successfully compiles for.) IIRC the https://github.com/aircrack-ng/rtl8812au driver also had this same lock up behavior that last time I tested it last year.

morrownr, 2025-04-28:
@rabidcopy 

>  this commit morrownr/8821au-20210708@0b12ea5 was what worked for me with no system hanging or freezing. (And still works on more recent kernels that it successfully compiles for.)

My intent was to freeze the vendor driver that you provided the link for with kernel 6.14 as hopefull it won't be needed going forward. Are you having trouble with it compliling on anything kernel 6.14 and below?

> Haven't been able to test any recent merges here,

It would be good if you can do so and provide log details or other details that night help if the issue is still a problem.

rabidcopy, 2025-04-28:
@morrownr 
> My intent was to freeze the vendor driver that you provided the link for with kernel 6.14 as hopefull it won't be needed going forward. Are you having trouble with it compliling on anything kernel 6.14 and below?
Never had an issues compiling up to 6.14 (or after). Sorry I worded that a bit weird.
> It would be good if you can do so and provide log details or other details that night help if the issue is still a problem.
Was able to find time today to compile the latest commit, (made sure the standalone 8821au driver was uninstalled and unloaded after a reboot) Installed rtw88, rebooted, plugged in the Archer T2U Plus. Took a bit for it to become active and show  available access points, then auto-connected to the existing saved connection. So far it hasn't hung the system or disconnected. Speeds are as expected.
dmesg output upon plugging it in and it connecting:
```
[   54.747199] rtw_8821au 1-10:1.0: Firmware version 42.4.0, H2C version 0
[   59.747512] usbcore: registered new interface driver rtw_8821au
[   96.065838] wlan0: authenticate with XX:XX:XX:XX:XX:XX (local address=YY:YY:YY:YY:YY:YY)
[   98.024546] wlan0: RX AssocResp from XX:XX:XX:XX:XX:XX (capab=0x1011 status=0 aid=18)
[   98.385567] wlan0: Limiting TX power to 30 (30 - 0) dBm as advertised by XX:XX:XX:XX:XX:XX
[  116.636722] warning: `ThreadPoolForeg' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
```
Assuming whenever https://github.com/lwfinger/rtw88/commit/9fee632533e1abd12e9660a160a2580161afadb6 makes it to 6.14+ in-kernel drivers, I should be able to just use that? (on Arch Linux)

morrownr, 2025-04-28:
> Assuming whenever 9fee632 makes it to 6.14+ in-kernel drivers, I should be able to just use that? (on Arch Linux, so figure that won't be far off)

@dubhater has been waiting for feedback from one of the original reporters but I'm sure it will go in soon unless there is a problem. He will likely address the folks that do the backports so that the patch is applied back to 6.14 as that kernel is in use by some of the big distros. This is probably a bug that could be taken directly to Linus to go into the kernel that is currently in work, 6.15. It is a rather ugly bug for those that experience it. 

Please continue testing and report your results.

ZerBea, 2025-05-07:
@dubhater , I forgot to mention that there are a lot of gcc 15 warnings in "regd.c" like this one:
```
/tmp/git_rtw88/regd.c:11:20: warning: initializer-string for array of 'char' truncates NUL terminator but destination lacks 'nonstring' attribute (3 chars into 2 available) [-Wunterminated-string-initialization]
   11 |         {.alpha2 = (_alpha2), \
      |                    ^
```

a5a5aa555oo, 2025-05-07:
> [@dubhater](https://github.com/dubhater) , I forgot to mention that there are a lot of gcc 15 warnings in "regd.c" like this one:
>
> ```
> /tmp/git_rtw88/regd.c:11:20: warning: initializer-string for [...]

@ZerBea @dubhater 

Yesterday, I found this problem on my Arch Linux, too.

kernel 6.12 ==> no warning
kernel 6.13 ==> no warning
kernel 6.14.arch1-1 ==> no warning
kernel 6.14.1.arch1-1 ==> no warning
kernel 6.14.2.arch1-1 ==> no warning
kernel 6.14.3.arch1-1 ==> no warning
kernel 6.14.4.arch1-1 ==> no warning (compiled by gcc 14)
kernel 6.14.4.arch1-2 ==> warnings appear ([kernel was recompiled by gcc 15](https://gitlab.archlinux.org/archlinux/packaging/packages/linux/-/commit/3144b858334ff444ca0d4706ebbda2fbdaffa2dc))
[linux-mainline 6.15rc5-1](https://aur.archlinux.org/packages/linux-mainline) ==> no warning ...

not sure what happened, maybe it was caused by gcc 15.
I see there are 2 commits related to this problem.
[gcc-15: make 'unterminated string initialization' just a warning](https://github.com/torvalds/linux/commit/d5d45a7f26194460964eb5677a9226697f7b7fdd)
[gcc-15: disable '-Wunterminated-string-initialization' entirely for now](https://github.com/torvalds/linux/commit/9d7a0577c9db35c4cc52db90bc415ea248446472)

dubhater, 2025-05-07:
There is a patch for that upstream: https://github.com/torvalds/linux/commit/d58ad77d5cc2a6d04db622a898e54d93fc7002a2

I left it out in the latest update because I wasn't sure how it would work with the older kernels/compilers.

a5a5aa555oo, 2025-05-08:
@dubhater 

> There is a patch for that upstream: https://github.com/torvalds/linux/commit/d58ad77d5cc2a6d04db622a898e54d93fc7002a2

I compiled the rtw88 driver with the above patch applied on Linux Mint 20.3 and no warning/error appears, driver also works.

OS: Linux Mint 20.3
kernel: 5.4.0-91-generic
gcc: 9.3.0

[image]

dubhater, 2025-05-08:
@a5a5aa555oo Thank you for testing that.