lwfinger/rtw88 #346: Fails to connect to eduroam

naalit, 2025-04-22:
(see https://github.com/lwfinger/rtw88/issues/344#issuecomment-2822580594)

Repeatedly tries and fails to connect to eduroam. The profile I'm using works with a different wifi adapter, so in theory it should be configured correctly. A snippet of `journalctl` output mentions that associating took too long:
```
Apr 22 17:01:00 archbook kernel: wlp2s0f0u12: deauthenticating from 34:8a:12:e9:4a:d1 by local choice (Reason: 3=DEAUTH_LEAVING)
Apr 22 17:00:59 archbook NetworkManager[603]: <info>  [1745359259.9999] device (wlp2s0f0u12): state change: prepare -> config (reason 'none', managed-type: 'full')
Apr 22 17:00:59 archbook NetworkManager[603]: <info>  [1745359259.9998] device (wlp2s0f0u12): state change: need-auth -> prepare (reason 'none', managed-type: 'full')
Apr 22 17:00:59 archbook NetworkManager[603]: <warn>  [1745359259.9994] device (wlp2s0f0u12): Activation: (wifi) asking for new secrets
Apr 22 17:00:59 archbook NetworkManager[603]: <info>  [1745359259.9992] device (wlp2s0f0u12): state change: config -> need-auth (reason 'none', managed-type: 'full')
Apr 22 17:00:59 archbook NetworkManager[603]: <warn>  [1745359259.9992] device (wlp2s0f0u12): Activation: (wifi) association took too long
Apr 22 17:00:59 archbook NetworkManager[603]: <info>  [1745359259.0796] device (p2p-dev-wlp2s0f0u12): supplicant management interface state: associating -> associated
Apr 22 17:00:59 archbook NetworkManager[603]: <info>  [1745359259.0796] device (wlp2s0f0u12): supplicant interface state: associating -> associated
Apr 22 17:00:59 archbook kernel: wlp2s0f0u12: associated
```

I'm on Arch Linux, using the latest dkms revision of this driver.

dubhater, 2025-04-23:
Try to run ``iw event -T`` before you connect to eduroam. It may print a bit more information.

naalit, 2025-04-23:
All I get is scanning messages:
```
[2025-04-23 13:21:47.601469]: wlp2s0f0u3 (phy #0): scan started
[2025-04-23 13:21:51.219851]: phy #0: beacon hint:
phy0 5240 MHz [48]:
	o Initiating radiation enabled
[2025-04-23 13:21:55.328423]: phy #0: beacon hint:
phy0 5765 MHz [153]:
	o Initiating radiation enabled
[2025-04-23 13:21:56.640371]: wlp2s0f0u3 (phy #0): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, ""
[2025-04-23 13:22:09.454639]: wlp2s0f0u3 (phy #0): scan started
[2025-04-23 13:22:09.871682]: wlp2s0f0u3 (phy #0): scan aborted: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, "eduroam" ""
[2025-04-23 13:22:22.713950]: wlp2s0f0u3 (phy #0): scan started
[2025-04-23 13:22:23.185002]: wlp2s0f0u3 (phy #0): scan aborted: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, ""
[2025-04-23 13:22:42.396903]: wlp2s0f0u3 (phy #0): scan started
[2025-04-23 13:22:42.852953]: wlp2s0f0u3 (phy #0): scan aborted: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, ""
[2025-04-23 13:22:55.687214]: wlp2s0f0u3 (phy #0): scan started
[2025-04-23 13:22:59.057626]: phy #0: beacon hint:
phy0 5220 MHz [44]:
	o Initiating radiation enabled
[2025-04-23 13:23:04.704118]: wlp2s0f0u3 (phy #0): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, "eduroam" ""
```

naalit, 2025-04-23:
I also wonder if it might be timing related, since the journalctl output mentions the wifi requesting new secrets when it takes too long to associate - connecting to my phone hotspot or my university guest wifi works but takes significantly longer than it does on a different wifi card (30s-1min instead of ~instant). Maybe if the connection takes too long eduroam rejects it for some sort of security reason?

dubhater, 2025-04-24:
It's strange, according to that iw event output it's not even trying to connect.

Anyway, I made one more change. If this doesn't help either, I can only suggest trying a [different driver](https://github.com/morrownr/8821au-20210708) or a different adapter.

naalit, 2025-04-24:
Ok, now `iw event -T` shows it connecting and disconnecting right after again. Here's a couple of attempts:
```
[2025-04-24 14:22:47.043651]: wlp2s0f0u3 (phy #1): scan started
[2025-04-24 14:22:56.121571]: wlp2s0f0u3 (phy #1): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, "eduroam" ""
[2025-04-24 14:23:02.660089]: wlp2s0f0u3: new station 34:8a:12:e9:4a:d1
[2025-04-24 14:23:04.570004]: wlp2s0f0u3 (phy #1): auth 34:8a:12:e9:4a:d1 -> 8c:90:2d:a2:9b:94 status: 0: Successful
[2025-04-24 14:23:04.933227]: wlp2s0f0u3 (phy #1): assoc 34:8a:12:e9:4a:d1 -> 8c:90:2d:a2:9b:94 status: 0: Successful
[2025-04-24 14:23:04.933264]: wlp2s0f0u3 (phy #1): connected to 34:8a:12:e9:4a:d1
[2025-04-24 14:23:06.109311]: wlp2s0f0u3: del station 34:8a:12:e9:4a:d1
[2025-04-24 14:23:06.883355]: wlp2s0f0u3 (phy #1): deauth 8c:90:2d:a2:9b:94 -> 34:8a:12:e9:4a:d1 reason 3: Deauthenticated because sending station is leaving (or has left) the IBSS or ESS
[2025-04-24 14:23:06.883408]: wlp2s0f0u3 (phy #1): disconnected (local request) reason: 3: Deauthenticated because sending station is leaving (or has left) the IBSS or ESS
[2025-04-24 14:23:06.943336]: regulatory domain change: set to world roaming by the wireless core upon initialization request
[2025-04-24 14:23:13.461719]: wlp2s0f0u3 (phy #1): scan started
[2025-04-24 14:23:16.903077]: phy #1: beacon hint:
phy1 5220 MHz [44]:
	o Initiating radiation enabled
[2025-04-24 14:23:17.086518]: phy #1: beacon hint:
phy1 5240 MHz [48]:
	o Initiating radiation enabled
[2025-04-24 14:23:22.579152]: wlp2s0f0u3 (phy #1): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, "eduroam" ""
[2025-04-24 14:23:29.141411]: wlp2s0f0u3: new station 34:8a:12:e9:4a:d1
[2025-04-24 14:23:31.163232]: wlp2s0f0u3: del station 34:8a:12:e9:4a:d1
[2025-04-24 14:23:44.523819]: wlp2s0f0u3 (phy #1): scan started
[2025-04-24 14:23:53.605015]: wlp2s0f0u3 (phy #1): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, ""
[2025-04-24 14:24:06.469216]: wlp2s0f0u3 (phy #1): scan started
[2025-04-24 14:24:15.582339]: wlp2s0f0u3 (phy #1): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, "eduroam" ""
[2025-04-24 14:24:22.150432]: wlp2s0f0u3: new station 34:8a:12:e9:4a:d1
[2025-04-24 14:24:24.058709]: wlp2s0f0u3 (phy #1): auth 34:8a:12:e9:4a:d1 -> 8c:90:2d:a2:9b:94 status: 0: Successful
[2025-04-24 14:24:24.540438]: wlp2s0f0u3 (phy #1): assoc 34:8a:12:e9:4a:d1 -> 8c:90:2d:a2:9b:94 status: 0: Successful
[2025-04-24 14:24:24.540475]: wlp2s0f0u3 (phy #1): connected to 34:8a:12:e9:4a:d1
[2025-04-24 14:24:25.104445]: wlp2s0f0u3: del station 34:8a:12:e9:4a:d1
[2025-04-24 14:24:25.881447]: wlp2s0f0u3 (phy #1): deauth 8c:90:2d:a2:9b:94 -> 34:8a:12:e9:4a:d1 reason 3: Deauthenticated because sending station is leaving (or has left) the IBSS or ESS
[2025-04-24 14:24:25.881482]: wlp2s0f0u3 (phy #1): disconnected (local request) reason: 3: Deauthenticated because sending station is leaving (or has left) the IBSS or ESS
[2025-04-24 14:24:25.938441]: regulatory domain change: set to world roaming by the wireless core upon initialization request
[2025-04-24 14:24:32.449515]: wlp2s0f0u3 (phy #1): scan started
[2025-04-24 14:24:36.052252]: phy #1: beacon hint:
phy1 5220 MHz [44]:
	o Initiating radiation enabled
[2025-04-24 14:24:36.131750]: phy #1: beacon hint:
phy1 5240 MHz [48]:
	o Initiating radiation enabled
[2025-04-24 14:24:41.655588]: wlp2s0f0u3 (phy #1): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, "eduroam" ""
[2025-04-24 14:24:48.229661]: wlp2s0f0u3: new station 34:8a:12:e9:4a:d1
[2025-04-24 14:24:50.340667]: wlp2s0f0u3 (phy #1): auth 34:8a:12:e9:4a:d1 -> 8c:90:2d:a2:9b:94 status: 0: Successful
[2025-04-24 14:24:50.350231]: wlp2s0f0u3: del station 34:8a:12:e9:4a:d1
[2025-04-24 14:25:03.742766]: wlp2s0f0u3 (phy #1): scan started
[2025-04-24 14:25:11.406090]: phy #1: beacon hint:
phy1 5765 MHz [153]:
	o Initiating radiation enabled
[2025-04-24 14:25:12.740849]: wlp2s0f0u3 (phy #1): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, ""
[2025-04-24 14:25:25.582910]: wlp2s0f0u3 (phy #1): scan started
[2025-04-24 14:25:34.571972]: wlp2s0f0u3 (phy #1): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, "eduroam" ""
[2025-04-24 14:25:41.140034]: wlp2s0f0u3: new station 34:8a:12:e9:4a:d1
[2025-04-24 14:25:43.064251]: wlp2s0f0u3 (phy #1): auth 34:8a:12:e9:4a:d1 -> 8c:90:2d:a2:9b:94 status: 0: Successful
[... cut ...]
Apr 24 14:26:08 archbook NetworkManager[598]: <info>  [1745522768.9944] device (p2p-dev-wlp2s0f0u3): supplicant management interface state: scanning -> authenticating
Apr 24 14:26:08 archbook NetworkManager[598]: <info>  [1745522768.9944] device (wlp2s0f0u3): supplicant interface state: scanning -> authenticating
Apr 24 14:26:08 archbook kernel: wlp2s0f0u3: send auth to 34:8a:12:e9:4a:d1 (try 1/3)
Apr 24 14:26:06 archbook kernel: wlp2s0f0u3: authenticate with 34:8a:12:e9:4a:d1 (local address=8c:90:2d:a2:9b:94)
Apr 24 14:26:00 archbook wpa_supplicant[1133]: wlp2s0f0u3: SME: Trying to authenticate with 34:8a:12:e9:4a:d1 (SSID='eduroam' freq=5240 MHz)
```

naalit, 2025-05-04:
Actually, I moved the adapter to a USB port on my motherboard instead of the case, and suddenly it can connect to eduroam. I guess the authentication timeout was just due to the USB port being too slow? Weird that it would work with other networks, though. This also fixed the issue where it took >30s to connect to networks, so this driver (the one built into the 6.14 kernel, even) works perfectly for me now. I can close the issue now, unless it not working on slow USB ports is a problem that needs to be fixed?

dubhater, 2025-05-04:
That's great, I'm glad it's working.

I expect the device to work at the same speed in both ports (480 Mbps). You can check with ``lsusb -t``. Your computer is not old enough to have USB 1.1, so if lsusb shows less than 480 Mbps then something is wrong with the port, or the USB controller, or the controller's driver, etc.

My guess is the reception happens to be better at the back of the case. I had a computer where it was the opposite: a different nano-sized wifi adapter had very low speed when plugged in the back but it was decent in the front.

closed 2025-12-23 by dubhater