lwfinger/rtw88 #347: USB ID for Epson OT-WL06 rtl8821cu

ibishop100, 2025-04-24:
Hi, I'm using your repo/driver to support these devices and I need to patch rtw8821cu.c with this:

{ USB_DEVICE_AND_INTERFACE_INFO(0x04b8, 0x0537, 0xff, 0xff, 0xff),
          .driver_info = (kernel_ulong_t)&(rtw8821c_hw_spec) }, /* Epson */

I've done some basic testing with it and haven't seen any problems.

Am I right to assume that this patch would make it back up to the kernel eventually, or should I create this issue on the kernel or wireless-next projects?

dubhater, 2025-04-24:
Please send the patch to the linux-wireless mailing list.

morrownr, 2025-04-25:
@ibishop100 

If you have never sent a patch to the linux-wireless mailing list, you probably need some mentoring. Would you like some help?

ibishop100, 2025-04-25:
@morrownr 

Heh, oldschool, I haven't used a mailing list in years.

https://wireless.docs.kernel.org/en/latest/en/developers/documentation/submittingpatches.html

I guess as long as I follow what's on there I should be OK right?  Or submit as a bug with the patch as well?

I see some drivers have more specific bug reporting guidelines like: https://wireless.docs.kernel.org/en/latest/en/users/drivers/ath11k/bugreport.html

That process seems to make a little more sense to me to follow, but I guess this being a new module the doc hasn't caught up...

morrownr, 2025-04-26:
> I guess as long as I follow what's on there I should be OK right? 

Yes, but this is like death by a thousand paper cuts until you do it a few times. Here is an example. It is the last patch I submitted that does the same thing you are trying to do.

https://patchwork.kernel.org/project/linux-wireless/patch/e797f105-9ca8-41e9-96de-7d25dec09943@gmail.com/

Two things that you should do first in my opinion:

1. You email app needs to be in text mode, not html.
2. You probably need to subscribe to linux-wireless and watch for a while. It is best if your email app can filter into tabs or folders.

Let me know when you have more questions.

morrownr, 2025-04-26:
Forgot to mention: Usually it is best to add in the comment brand and model name instead of just brand.

a5a5aa555oo, 2025-05-06:
Hi, @ibishop100 

I can take the USB ID for your device to linux-wireless mailing list.
Please email your full name and email address to zenmchen@gmail.com if you need help. :)
(Your name and email will be displayed in the "patch author" field.)

dubhater, 2025-12-23:
@ibishop100 Any luck sending the patch?

ibishop100, 2025-12-24:
No.  I have been working on a different project.  For that one, we're
resigned to maintaining our own driver package since timing was never going
to work with Debian stable.  It looks like this device was not intended to
be widely distributed, but hey maybe in the wasteland in 20 years when some
grizzled survivor with no coding skills plugs in the wifi adapter he found
in a walmart into his crusty old laptop he'll be thrilled that someone
submitted the necessary patch now.