lwfinger/rtw88 #353: 8822BE not woking on Ubuntu 25.04

AmirHosseinNiknezhad, 2025-05-05:
Hello everyone

I followed the basic installation route and, I believe, installed the driver successfully; but I still can't really use it as it either doesn't work or works for a few minutes only after boot. Setting either of the kernel options pcie_aspm.policy=performance or pcie_aspm=off seems to make a difference though. I'm not very savvy, so I'm not sure how to diagnose the issue. I'm using a Lenovo Legion Y530-15ICH and the chip is built in (Bus 001 Device 005: ID 0bda:b023 Realtek Semiconductor Corp. RTL8822BE Bluetooth 4.2 Adapter). I'm on kernel 6.14.0-15-generic.

Here is what `dmesg | grep rtw` returns:

<details><summary>Details</summary>
<p>

[    4.500056] rtw_core: loading out-of-tree module taints kernel.
[    4.500062] rtw_core: module verification failed: signature and/or required key missing - tainting kernel
[    4.586752] rtw_8822be 0000:07:00.0: Firmware version 30.20.0, H2C version 14
[    4.680778] rtw_8822be 0000:07:00.0 wlp7s0: renamed from wlan0
[   52.111740] WARNING: CPU: 4 PID: 299 at pci.c:1280 rtw_dbi_read8.constprop.0+0xba/0xe0 [rtw_pci]
[   52.111750] Modules linked in: snd_seq_dummy snd_hrtimer ccm qrtr rfcomm cmac algif_hash algif_skcipher af_alg bnep snd_ctl_led snd_hda_codec_realtek snd_hda_codec_generic snd_hda_scodec_component intel_uncore_frequency intel_uncore_frequency_common binfmt_misc intel_tcc_cooling x86_pkg_temp_thermal intel_powerclamp coretemp snd_sof_pci_intel_cnl snd_sof_intel_hda_generic soundwire_intel soundwire_cadence snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_intel_hda_mlink snd_sof_intel_hda snd_sof_pci snd_sof_xtensa_dsp snd_sof snd_sof_utils snd_soc_acpi_intel_match snd_soc_acpi_intel_sdca_quirks soundwire_generic_allocation snd_soc_acpi rtw_8822be(OE) soundwire_bus rtw_8822b(OE) snd_soc_sdca rtw_pci(OE) snd_soc_avs kvm_intel snd_soc_hda_codec snd_hda_ext_core rtw_core(OE) snd_hda_codec_hdmi snd_soc_core kvm snd_compress ac97_bus snd_pcm_dmaengine rtl8xxxu uvcvideo snd_hda_intel videobuf2_vmalloc polyval_clmulni uvc snd_intel_dspcfg snd_intel_sdw_acpi polyval_generic videobuf2_memops snd_usb_audio ghash_clmulni_intel
[   52.111973] Workqueue: phy1 rtw_watch_dog_work [rtw_core]
[   52.111988] RIP: 0010:rtw_dbi_read8.constprop.0+0xba/0xe0 [rtw_pci]
[   52.112018]  ? rtw_pci_link_ps.part.0+0xa6/0xe0 [rtw_pci]
[   52.112026]  ? rtw_dbi_read8.constprop.0+0xba/0xe0 [rtw_pci]
[   52.112031]  ? rtw_dbi_read8.constprop.0+0xba/0xe0 [rtw_pci]
[   52.112049]  ? rtw_dbi_read8.constprop.0+0xba/0xe0 [rtw_pci]
[   52.112051]  rtw_pci_link_ps.part.0+0xa6/0xe0 [rtw_pci]
[   52.112054]  rtw_pci_link_ps+0x1b/0x30 [rtw_pci]
[   52.112057]  rtw_leave_lps+0x82/0x1d0 [rtw_core]
[   52.112070]  rtw_watch_dog_work+0x1d7/0x310 [rtw_core]
[   52.112109] rtw_8822be 0000:07:00.0: failed to read ASPM, ret=-5
[   52.115245] rtw_8822be 0000:07:00.0: failed to send h2c command
[   52.219735] rtw_8822be 0000:07:00.0: firmware failed to leave lps state
[   52.222863] rtw_8822be 0000:07:00.0: failed to send h2c command
[   52.226035] rtw_8822be 0000:07:00.0: failed to send h2c command
[   52.229173] rtw_8822be 0000:07:00.0: failed to send h2c command
[   52.232328] rtw_8822be 0000:07:00.0: failed to send h2c command
[   52.662614] rtw_8822be 0000:07:00.0: failed to send h2c command
[   52.666001] rtw_8822be 0000:07:00.0: failed to send h2c command
[   52.669544] rtw_8822be 0000:07:00.0: failed to send h2c command
[   54.830148] rtw_8822be 0000:07:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[   54.830474] WARNING: CPU: 0 PID: 94 at pci.c:1280 rtw_dbi_read8.constprop.0+0xba/0xe0 [rtw_pci]
[   54.830482] Modules linked in: snd_seq_dummy snd_hrtimer ccm qrtr rfcomm cmac algif_hash algif_skcipher af_alg bnep snd_ctl_led snd_hda_codec_realtek snd_hda_codec_generic snd_hda_scodec_component intel_uncore_frequency intel_uncore_frequency_common binfmt_misc intel_tcc_cooling x86_pkg_temp_thermal intel_powerclamp coretemp snd_sof_pci_intel_cnl snd_sof_intel_hda_generic soundwire_intel soundwire_cadence snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_intel_hda_mlink snd_sof_intel_hda snd_sof_pci snd_sof_xtensa_dsp snd_sof snd_sof_utils snd_soc_acpi_intel_match snd_soc_acpi_intel_sdca_quirks soundwire_generic_allocation snd_soc_acpi rtw_8822be(OE) soundwire_bus rtw_8822b(OE) snd_soc_sdca rtw_pci(OE) snd_soc_avs kvm_intel snd_soc_hda_codec snd_hda_ext_core rtw_core(OE) snd_hda_codec_hdmi snd_soc_core kvm snd_compress ac97_bus snd_pcm_dmaengine rtl8xxxu uvcvideo snd_hda_intel videobuf2_vmalloc polyval_clmulni uvc snd_intel_dspcfg snd_intel_sdw_acpi polyval_generic videobuf2_memops snd_usb_audio ghash_clmulni_intel
[   54.830691] RIP: 0010:rtw_dbi_read8.constprop.0+0xba/0xe0 [rtw_pci]
[   54.830720]  ? rtw_pci_link_ps.part.0+0xa6/0xe0 [rtw_pci]
[   54.830728]  ? rtw_dbi_read8.constprop.0+0xba/0xe0 [rtw_pci]
[   54.830733]  ? rtw_dbi_read8.constprop.0+0xba/0xe0 [rtw_pci]
[   54.830766]  ? rtw_dbi_read8.constprop.0+0xba/0xe0 [rtw_pci]
[   54.830769]  ? rtw_dbi_read8.constprop.0+0xba/0xe0 [rtw_pci]
[   54.830772]  rtw_pci_link_ps.part.0+0xa6/0xe0 [rtw_pci]
[   54.830774]  rtw_pci_link_ps+0x1b/0x30 [rtw_pci]
[   54.830777]  rtw_enter_ips+0x47/0x60 [rtw_core]
[   54.830792]  rtw_ops_config+0xb6/0xd0 [rtw_core]
[   54.831528] rtw_8822be 0000:07:00.0: failed to read ASPM, ret=-5
[   54.932356] WARNING: CPU: 7 PID: 1644 at pci.c:1280 rtw_dbi_read8.constprop.0+0xba/0xe0 [rtw_pci]
[   54.932363] Modules linked in: snd_seq_dummy snd_hrtimer ccm qrtr rfcomm cmac algif_hash algif_skcipher af_alg bnep snd_ctl_led snd_hda_codec_realtek snd_hda_codec_generic snd_hda_scodec_component intel_uncore_frequency intel_uncore_frequency_common binfmt_misc intel_tcc_cooling x86_pkg_temp_thermal intel_powerclamp coretemp snd_sof_pci_intel_cnl snd_sof_intel_hda_generic soundwire_intel soundwire_cadence snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_intel_hda_mlink snd_sof_intel_hda snd_sof_pci snd_sof_xtensa_dsp snd_sof snd_sof_utils snd_soc_acpi_intel_match snd_soc_acpi_intel_sdca_quirks soundwire_generic_allocation snd_soc_acpi rtw_8822be(OE) soundwire_bus rtw_8822b(OE) snd_soc_sdca rtw_pci(OE) snd_soc_avs kvm_intel snd_soc_hda_codec snd_hda_ext_core rtw_core(OE) snd_hda_codec_hdmi snd_soc_core kvm snd_compress ac97_bus snd_pcm_dmaengine rtl8xxxu uvcvideo snd_hda_intel videobuf2_vmalloc polyval_clmulni uvc snd_intel_dspcfg snd_intel_sdw_acpi polyval_generic videobuf2_memops snd_usb_audio ghash_clmulni_intel
[   54.932513] RIP: 0010:rtw_dbi_read8.constprop.0+0xba/0xe0 [rtw_pci]
[   54.932555]  ? rtw_pci_link_ps.part.0+0xa6/0xe0 [rtw_pci]
[   54.932564]  ? rtw_dbi_read8.constprop.0+0xba/0xe0 [rtw_pci]
[   54.932568]  ? rtw_dbi_read8.constprop.0+0xba/0xe0 [rtw_pci]
[   54.932584]  ? rtw_dbi_read8.constprop.0+0xba/0xe0 [rtw_pci]
[   54.932587]  ? rtw_dbi_read8.constprop.0+0xba/0xe0 [rtw_pci]
[   54.932589]  rtw_pci_link_ps.part.0+0xa6/0xe0 [rtw_pci]
[   54.932592]  rtw_pci_link_ps+0x1b/0x30 [rtw_pci]
[   54.932596]  rtw_leave_ips+0x28/0x80 [rtw_core]
[   54.932612]  rtw_ops_config+0x95/0xd0 [rtw_core]
[   54.933302] rtw_8822be 0000:07:00.0: failed to read ASPM, ret=-5
[   57.081636] rtw_8822be 0000:07:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[   57.081647] rtw_8822be 0000:07:00.0: mac power on failed
[   57.081649] rtw_8822be 0000:07:00.0: failed to power on mac
[   57.081651] rtw_8822be 0000:07:00.0: leave idle state failed
[   57.084753] rtw_8822be 0000:07:00.0: failed to send h2c command
[   57.087875] rtw_8822be 0000:07:00.0: failed to send h2c command
[   57.090973] rtw_8822be 0000:07:00.0: failed to send h2c command
[   57.091198] rtw_8822be 0000:07:00.0: failed to leave ips state
[... 99 lines cut ...]
[  357.410079] rtw_8822be 0000:07:00.0: failed to leave idle state
[  477.454626] rtw_8822be 0000:07:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[  477.454637] rtw_8822be 0000:07:00.0: mac power on failed
[  477.454639] rtw_8822be 0000:07:00.0: failed to power on mac
[  477.454641] rtw_8822be 0000:07:00.0: leave idle state failed
[  477.457777] rtw_8822be 0000:07:00.0: failed to send h2c command
[  477.460930] rtw_8822be 0000:07:00.0: failed to send h2c command
[  477.464064] rtw_8822be 0000:07:00.0: failed to send h2c command
[  477.464209] rtw_8822be 0000:07:00.0: failed to leave ips state
[  477.464212] rtw_8822be 0000:07:00.0: failed to leave idle state
[  597.432285] rtw_8822be 0000:07:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[  597.432297] rtw_8822be 0000:07:00.0: mac power on failed
[... cut ...]
[  597.432301] rtw_8822be 0000:07:00.0: leave idle state failed
[  597.435430] rtw_8822be 0000:07:00.0: failed to send h2c command
[  597.438603] rtw_8822be 0000:07:00.0: failed to send h2c command
[  597.441749] rtw_8822be 0000:07:00.0: failed to send h2c command
[  597.441891] rtw_8822be 0000:07:00.0: failed to leave ips state
[  597.441894] rtw_8822be 0000:07:00.0: failed to leave idle state

</p>
</details> 

Feel free to ask for any relevant information I might have left out
Thanks for any help in advance!

dubhater, 2025-05-05:
Create ``/etc/modprobe.d/rtw88_pci.conf`` and put these lines in it:

```
options rtw88_pci disable_aspm=1
options rtw_pci disable_aspm=1
```

Then shut down the computer. Rebooting is not enough, you have to shut down.

Maybe that will help.

josectheone, 2025-05-19:
> Create `/etc/modprobe.d/rtw88_pci.conf` and put these lines in it:
> 
> ```
> options rtw88_pci disable_aspm=1
> options rtw_pci disable_aspm=1
> ```
> 
> Then shut down the computer. Rebooting is not enough, you have to shut down.
> 
> Maybe that will help.

This solution on top of the driver instalation from this repo solved the issue, thanks ;)

AmirHosseinNiknezhad, 2025-05-20:
> Create `/etc/modprobe.d/rtw88_pci.conf` and put these lines in it:
> 
> ```
> options rtw88_pci disable_aspm=1
> options rtw_pci disable_aspm=1
> ```
> 
> Then shut down the computer. Rebooting is not enough, you have to shut down.
> 
> Maybe that will help.

That totally fixed the issue!
Thanks a lot and sorry for the late reply