lwfinger/rtw88 #355: Glitches Occur When Using Two Realtek Adapters in Monitor Mode with this driver

reaveruser, 2025-05-08:
When using two Realtek adapters in Monitor Mode (tested with two rtw8814au, or one rtw8814au and one rtw8822b) with this driver, multiple glitches occur. These issues are inconsistent but tend to happen with varying frequency across different channels and bands. The glitches prevent proper scanning, injection, and successful attack execution.

Steps to Reproduce:

- Set both Realtek adapters in monitor mode.
- Use tools like airodump-ng and reaver to test functionality.

Observe the glitches:

- airodump-ng detects only one AP, despite many being available.
- Injection may only work on channels 1–11, or may not work at all.
- Reaver may crash or fail to proceed with the attack.

_**Note:** The issue only occurs when enabling monitor mode for two Realtek adapters (either two rtw8814au or one rtw8814au + one rtw8822b) using this driver._

**Expected Behavior:**

- Both Realtek adapters should function correctly in monitor mode without causing glitches.
- Tools like airodump-ng should display all available APs, and injection should work on all channels.
- No crashes or errors should occur when using tools like Reaver.

**Actual Behavior:**

_Glitches:_

airodump-ng only sees one AP, despite many being available.
Injection works intermittently or only on channels 1–11, or doesn’t work at all.
Reaver crashes or fails to execute correctly.

**No Issues When:**

- Using only one Realtek adapter with this driver (no glitches).
- Using one Mediatek adapter and one Realtek adapter (no glitches).

**Additional Notes:**

When combining the Morrownr 8814au driver for rtw8814au and Kali’s native driver for rtw8822b, there are no issues enabling monitor mode for two Realtek adapters. This combination works as expected without glitches. It is only with this driver.

dubhater, 2025-05-08:
I wonder if it's the same problem that was recently fixed in rtw89: https://lore.kernel.org/linux-wireless/2goskmst4na36v42p2bs47uernp6kh3gzpadhr3u3r2yvyoxlg@bfprgq2qae7p/

reaveruser, 2025-05-11:
This could explain why the injection didn't work, even on channels 36 to 48.

This issue has been driving me crazy—not only because it is inconsistent, but also because it only occurs when two Realtek adapters are connected. When only one adapter is attached, the issue does not occur. Disconnecting one adapter immediately resolves the problem. It took me a while to realize it.

If you need assistance with logs or debugging, please let me know, and I’ll be happy to help.