lwfinger/rtw88 #356: USB register read/write failures (-71, -19) in rtw_8814au driver

Watchdog0x, 2025-05-08:
### Description:
The rtw_8814au driver repeatedly fails to read/write registers over USB, resulting in errors -71 (EPROTO) and -19 (ENODEV). The issue persists across reboots and occurs during both initialization (h2c packet failures) and general operation.

### Error Logs:
```bash
[   92.972428] rtw_8814au 6-4:1.0: write register 0xc60 failed with -71
[   93.197328] rtw_8814au 6-4:1.0: failed to do USB write, ret=-19
[   93.197345] rtw_8814au 6-4:1.0: failed to send h2c packet
[  129.208860] rtw_8814au 6-4:1.0: read register 0x60 failed with -71
[  236.574200] rtw_8814au 6-4:1.0: read register 0x454 failed with -71
[  236.574314] rtw_8814au 6-4:1.0: read register 0x860 failed with -71
[  236.574556] rtw_8814au 6-4:1.0: read register 0x2860 failed with -71
```

###  System Information:
**Kernel Version:** ```6.1.0-34-amd64```
**Driver:** ```8814au```
**USB Device:** ```0b05:1817 ASUSTek Computer, Inc. USB-AC68 802.11a/b/g/n/ac (4x4) Wireless Adapter [Realtek RTL8814AU]```
**Distribution:**  ```Debian 12 (bookworm)```

### Steps to Reproduce:
1. Connect the USB WiFi adapter using the rtw_8814au driver.
2. Reboot the system
3. Monitor kernel logs for register read/write errors.

dubhater, 2025-05-08:
Please attach the full dmesg. Or the output from ``journalctl -b -X`` is fine too, you don't need to try to reproduce it again.

Watchdog0x, 2025-05-08:
[journalctl_current.log](https://github.com/user-attachments/files/20104531/journalctl_current.log)

dubhater, 2025-05-08:
I see a lot of "usb 6-4: reset SuperSpeed USB device number 4 using xhci_hcd". Do you know what causes that?

Watchdog0x, 2025-05-08:
I can try disabling USB autosuspend ```usbcore.autosuspend=-1```

dubhater, 2025-05-08:
Sure, try that. Also try to disable the automatic switching to USB 3. rtw_usb has the parameter "switch_usb_mode" for that.

Watchdog0x, 2025-05-08:
> I see a lot of "usb 6-4: reset SuperSpeed USB device number 4 using xhci_hcd". Do you know what causes that?

I think it might be a misbehaving driver.  
When I load my old driver, the "reset SuperSpeed USB device" messages stop. 

But I'll try what you suggested and see how it goes.

a5a5aa555oo, 2025-05-08:
Hi @Watchdog0x,

> I think it might be a misbehaving driver. When I load my old driver, the "reset SuperSpeed USB device" messages stop.

There is a difference between old/new driver: the new driver switches the adapter to USB 3 mode automatically, but the old one doesn't.

> But I'll try what you suggested and see how it goes.

Please see [Q2](https://github.com/lwfinger/rtw88?tab=readme-ov-file#q2-my-wi-fi-adapter-is-plugged-in-an-usb-3-port-how-can-i-keep-it-in-usb-2-mode-to-avoid-potential-interference-in-24-ghz-band) which contains detailed instructions to make your adapter stay in USB 2 mode.

Watchdog0x, 2025-05-08:
The funny part?? I just checked, and the WiFi dongle is actually USB 3.0.

Watchdog0x, 2025-05-08:
I will try 2 sek

Watchdog0x, 2025-05-08:
I changed the config from:

```
options 8814au rtw_switch_usb_mode=0 rtw_led_ctrl=1
```
to: 
```
options 8814au rtw_switch_usb_mode=1 rtw_led_ctrl=1
```
just like it says in the 8814au.conf
Note: To activate USB3 mode, change rtw_switch_usb_mode above to rtw_switch_usb_mode=1

And yes, it did switch now see:
```
Class=Vendor Specific Class, Driver=rtl8814au, 5000M <- USB 3
```

But nope, it still didn’t work,  I’m getting the same error:
```
reset SuperSpeed USB device number 2 using xhci_hcd
```

Watchdog0x, 2025-05-08:
@dubhater 

Haha, that’s actually pretty funny, switched my adapter to a different USB 3 port, and now the old driver is working on USB 3:
```
Class=Vendor Specific Class, Driver=rtl8814au, 5000M <- USB 3
```
I’m going to try the new driver using the same USB port.

[logForOldDriverOnUSB3.log](https://github.com/user-attachments/files/20106771/gg.log)

Watchdog0x, 2025-05-08:
## Full Report - What I Found Out

### Old Driver (on USB 3.0):

1. The device constantly flips on and off.
2. Oddly, if I move it to another USB 3.0 port while the machine is running, it starts working normally.
3. However, I have to do this every time I reboot, which is obviously not ideal.

### New Driver (on USB 3.0):

1. Same issue: the device keeps flipping on and off.
2. I tried disabling USB autosuspend with usbcore.autosuspend=-1, but it made no difference.
3. If I switch it to another USB 3.0 port while the machine is running, it appears connected (I see internet access), but actual usage fails (no response from the router, no pings, no traffic)

For now, the only reliable setup is forcing the adapter to use USB 2.0.
This works consistently with both drivers.
Unfortunately, I still don’t know why USB 3.0 causes these issues.

[newDriverTest.log](https://github.com/user-attachments/files/20108277/newDriverTest.log)

dubhater, 2025-05-08:
I see that ModemManager is trying to do something with the wifi. What if you turn it off?

Watchdog0x, 2025-05-10:
Hi @dubhater 

After spending some time digging today, I finally had a chance to figure out what the issue was about.

The problem was that Realtek had originally downgraded the USB connection to USB 2.0 due to lock-ups. 
I managed to get full USB 3.0 throughput working without those lock-ups.

To fix it, I added this function to usb.c

```C
static void rtw_usb_disable_lpm(struct usb_device *udev)
{
	usb_disable_lpm(udev);
}
```

Then I modified rtw_usb_probe to include:

```C
struct usb_device *udev = interface_to_usbdev(intf);

if (udev->speed == USB_SPEED_SUPER || udev->speed == USB_SPEED_SUPER_PLUS) {
	dev_info(&intf->dev, "rtw88_usb: Super-Speed link – disabling U1/U2\n");
	rtw_usb_disable_lpm(udev);
}
```
It seems to be working very well now.

Watchdog0x, 2025-05-10:
Never mind nothing works 🦖

dubhater, 2025-05-10:
If you have other USB devices, try unplugging them all. If that helps, add them back one by one.