lwfinger/rtw88 #357: Conflicting file (firmware) during system update

MR-Diamond, 2025-05-10:
Hi, since 5 February 2025 I'm using this module for my USB WI-Fi adapter: `TP-Link Archer T9UH v1 [Realtek RTL8814AU]`

So, on Manjaro, I installed the package `rtw88-dkms-git`, which has always works perfectly (and still is working perfectly, without issues.)

On May 6 2025 I applied a system update of Manjaro, which included the package `linux-firmware` version `20250408.c1a774f3-1`

During the update process, I received an error message which has stopped the update process:
`the file /usr/lib/firmware/rtw88/rtw8814a_fw.bin already exists on the file system`, so I had to unistall the package `rtw88-dkms-git`, continue to apply the system update, and at the next boot (since the adapter, obviously, wasn't recognized), using the USB tethering of my smartphone, I have been able to reinstall the package `rtw88-dkms-git`.

Who is the culprit? I repeat: since 5 February 2025 I never encountered this issue.
Maybe, to avoid this error for the next system update, should I add
`NoExtract = /usr/lib/firmware/rtw88/rtw8814a_fw.bin` in the config file `/etc/pacman.conf` ?

Thank you for the attention and keep up the good work!

a5a5aa555oo, 2025-05-10:
Hi,

im the maintainer of rtw88-dkms-git, let me tell you what happended as possible as i can:

linux-firmware older than 20250408.c1a774f3-1 did not include the firmware necessary for 8814AU. To make 8814AU adapters be driven properly by the rtw88 driver, rtw88-dkms-git installs both the firmware/driver for 8814AU to your system.

This is the reason why you saw that error. When you upgrade linux-firmware to 20250408.c1a774f3-1 which also contains 8814AU firmware, pacman found that /etc/modprobe.d/rtw8814a_fw.bin exist already and spited out an error.

[This problem had been fixed](https://aur.archlinux.org/cgit/aur.git/commit/?h=rtw88-dkms-git&id=b4cbd2b96eaf61dceab85e0991a23f3c70c3413c) after linux-firmware 20250408.c1a774f3-1 is out, now rtw88-dkms-git doesn't  copy rtw_8814a_fw.bin to /etc/modprobe.d, so conflict won't happen anymore. :)

> Maybe, to avoid this error for the next system update, should I add
> NoExtract = /usr/lib/firmware/rtw88/rtw8814a_fw.bin in the config file /etc/pacman.conf ?

This is not needed, you won't see that error anymore.

MR-Diamond, 2025-05-10:
> [This problem had been fixed](https://aur.archlinux.org/cgit/aur.git/commit/?h=rtw88-dkms-git&id=b4cbd2b96eaf61dceab85e0991a23f3c70c3413c) after linux-firmware 20250408.c1a774f3-1 is out

... Which I received with the system update of May 6, 2025; so: I can consider all fixed?

Many thanks not only for the detailed answer, but also and expecially for your good work on this driver/module! For me, works like a charme :-)

a5a5aa555oo, 2025-05-10:
> ... Which I received with the system update of May 6, 2025; so: I can consider all fixed?

Yes, all fixed.

> Many thanks not only for the detailed answer, but also and expecially for your good work on this driver/module! For me, works like a charme :-)

The person you should thank is @dubhater, he wrote an excellent driver for us! I'm only a packager of rtw88-dkms-git.

MR-Diamond, 2025-05-10:
> The person you should thank is [@dubhater](https://github.com/dubhater), he wrote an excellent driver for us!

So: [@dubhater](https://github.com/dubhater), many thanks!

MR-Diamond, 2025-05-17:
On May 13, I received an update of `linux-firmware` to `20250508.788aadc8-1`, and this time I didn't had any issue.
I close this ticket.

Cheers.

closed 2025-05-17 by MR-Diamond