lwfinger/rtw88 #358: rtl8811au nixos not adding a device via `ip a`

Redhawk18, 2025-05-11:
Hello everyone I was trying to get my spare wifi adapter to work. 

This is the product
https://www.netgear.com/support/product/a6100/

This is my dmesg log
```
[ 4286.013115] usb 3-1: USB disconnect, device number 3
[ 4288.838658] usb 3-1: new high-speed USB device number 4 using xhci_hcd
[ 4288.965107] usb 3-1: New USB device found, idVendor=0846, idProduct=9052, bcdDevice= 2.00
[ 4288.965121] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4288.965127] usb 3-1: Product: 802.11ac WLAN Adapter 
[ 4288.965132] usb 3-1: Manufacturer: Realtek 
[ 4288.965137] usb 3-1: SerialNumber: 00e04c000001
[ 4288.968267] rtw_8821au 3-1:1.0: Firmware version 42.4.0, H2C version 0
```

This is my `ip a` log
```
❯ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: enp5s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 70:85:c2:49:14:d5 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.143/24 brd 192.168.1.255 scope global dynamic noprefixroute enp5s0
       ...
5: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether 44:a5:6e:51:00:ce brd ff:ff:ff:ff:ff:ff
```

It is odd that in the nixos installer it gives me more details, that its a Realtek RTL8811AU. The driver version says 8821au, I'm not sure if that's a problem but fyi. I'm not sure what else the problem could be.

morrownr, 2025-05-12:
> The driver version says 8821au, I'm not sure if that's a problem

That is not a problem. They are basically the same chip and use the same driver. The 21au has bluetooth support and the 11au does not.

I don't see a problem. Is it working?

Redhawk18, 2025-05-12:
> > The driver version says 8821au, I'm not sure if that's a problem
> 
> That is not a problem. They are basically the same chip and use the same driver. The 21au has bluetooth support and the 11au does not.
> 
> I don't see a problem. Is it working?

No

Redhawk18, 2025-05-12:
Ah I was confused by the device names, my bad

closed 2025-05-12 by Redhawk18