lwfinger/rtw88 #363: dkms install fails on RPi 2

acarl005, 2025-06-06:
I installed the pre-requisites and then ran `sudo dkms install $PWD` inside the cloned directory, and got this:

```
Sign command: /lib/modules/6.12.25+rpt-rpi-v7/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
KVER=6.12.25+rpt-rpi-v7 'make' all....................................................................
Error! Build of rtw_8723de.ko failed for: 6.12.25+rpt-rpi-v7 (armv71)
Make sure the name of the generated module is correct and at the root of the
build directory, or consult make.log in the build directory
/var/lib/dkms/rtw88/0.6/build for more information.
```

```
uname -a
Linux raspberrypi 6.12.25+rpt-rpi-v7 #1 SMP Raspbian 1:6.12.25-1+rpt1 (2025-04-30) armv71 GNU/Linux
```

I checked `/var/lib/dkms/rtw88/0.6/build/make.log` and no errors as far as I can tell.

acarl005, 2025-06-06:
I only needed the `rtw_8821au` driver, which did successfully build. Running `sudo make install` directly seems to have worked, but when I ran `sudo dkms install $PWD` as suggested in the README, the process short-circuited.

sashablue, 2025-08-16:
Confirming for RPI 2B with USB device 
`Bus 001 Device 004: ID 2001:3314 D-Link Corp. DWA-171 AC600 DB Wireless Adapter(rev.A1) [Realtek RTL8811AU]`
Removing from dkms.conf the PCI dependent modules found under `ifeq ($(CONFIG_PCI), y)` sections in `Makefile `solved it for me as well:
dkms.conf after removal:

```
PACKAGE_NAME="rtw88"
PACKAGE_VERSION="0.6"
MAKE="KVER=${kernelver} 'make' all"
CLEAN="KVER=${kernelver} 'make' clean"
AUTOINSTALL=yes
BUILT_MODULE_NAME[0]="rtw_8703b"
BUILT_MODULE_NAME[1]="rtw_8723cs"
BUILT_MODULE_NAME[2]="rtw_8723d"
BUILT_MODULE_NAME[3]="rtw_8723ds"
BUILT_MODULE_NAME[4]="rtw_8723du"
BUILT_MODULE_NAME[5]="rtw_8723x"
BUILT_MODULE_NAME[6]="rtw_8812a"
BUILT_MODULE_NAME[7]="rtw_8812au"
BUILT_MODULE_NAME[8]="rtw_8814a"
BUILT_MODULE_NAME[9]="rtw_8814au"
BUILT_MODULE_NAME[10]="rtw_8821a"
BUILT_MODULE_NAME[11]="rtw_8821au"
BUILT_MODULE_NAME[12]="rtw_8821c"
BUILT_MODULE_NAME[13]="rtw_8821cs"
BUILT_MODULE_NAME[14]="rtw_8821cu"
BUILT_MODULE_NAME[15]="rtw_8822b"
BUILT_MODULE_NAME[16]="rtw_8822bs"
BUILT_MODULE_NAME[17]="rtw_8822bu"
BUILT_MODULE_NAME[18]="rtw_8822c"
BUILT_MODULE_NAME[19]="rtw_8822cs"
BUILT_MODULE_NAME[20]="rtw_8822cu"
BUILT_MODULE_NAME[21]="rtw_88xxa"
BUILT_MODULE_NAME[22]="rtw_core"
BUILT_MODULE_NAME[23]="rtw_sdio"
BUILT_MODULE_NAME[24]="rtw_usb"
DEST_MODULE_LOCATION[0]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[1]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[2]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[3]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[4]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[5]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[6]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[7]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[8]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[9]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[10]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[11]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[12]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[13]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[14]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[15]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[16]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[17]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[18]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[19]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[20]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[21]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[22]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[23]="/updates/dkms/rtw88"
DEST_MODULE_LOCATION[24]="/updates/dkms/rtw88"

# POST_INSTALL="dkms/post_install.sh"
```

randomcoder67, 2025-08-28:
> Confirming for RPI 2B with USB device `Bus 001 Device 004: ID 2001:3314 D-Link Corp. DWA-171 AC600 DB Wireless Adapter(rev.A1) [Realtek RTL8811AU]` Removing from dkms.conf the PCI dependent modules [...]

Can confirm this fixed it for me, note that I had to run `sudo rm -rf /var/lib/rtw88` otherwise the build would fail at the start

a5a5aa555oo, 2025-08-29:
@acarl005 @Z0n1 @sashablue @randomcoder67 

Could you help testing if [this PR](https://github.com/lwfinger/rtw88/pull/388) helps?

Steps:

1. Remove the installed rtw88 drivers

```
sudo dkms remove rtw88/0.6 --all
```

2. Remove the source code located at /usr/src/

```
sudo rm -rf /usr/src/rtw88-0.6/
```

3. Clone the rtw88 driver in my repo

```
git clone https://github.com/a5a5aa555oo/rtw88
```

4. Try to install the driver

```
cd rtw88
sudo dkms install $PWD
```

Z0n1, 2025-08-31:
> [@acarl005](https://github.com/acarl005) [@Z0n1](https://github.com/Z0n1) [@sashablue](https://github.com/sashablue) @randomcoder67
> 
> Could you help testing if [this PR](https://github.com/lwfinger/rtw88/pull/388) helps?
> 
> Steps:
> 
> 1. Remove the installed rtw88 drivers
> 
> ```
> sudo dkms remove rtw88/0.6 --all
> ```
> 
> 2. Remove the source code located at /usr/src/
> 
> ```
> sudo rm -rf /usr/src/rtw88-0.6/
> ```
> 
> 3. Clone the rtw88 driver in my repo
> 
> ```
> git clone https://github.com/a5a5aa555oo/rtw88
> ```
> 
> 4. Try to install the driver to see if this problem is solved.
> 
> ```
> cd rtw88
> sudo dkms install $PWD
> ```

I can happily confirm! Installation went through without any issues. Device is working as expected.
Happy to do further tests if needed. 
Thanks for the efforts in supporting older hardware!

`Sign command: /lib/modules/6.12.25+rpt-rpi-v7/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/rtw88/0.6/source -> /usr/src/rtw88-0.6

Building module:
Cleaning build area...
KVER=6.12.25+rpt-rpi-v7 'make' all...............................................................................................
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_core.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8703b.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8723d.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8723x.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8812a.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8814a.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8821a.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8821c.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8822b.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8822c.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_88xxa.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_usb.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8723du.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8812au.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8814au.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8821au.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8821cu.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8822bu.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8822cu.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_sdio.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8723cs.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8723ds.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8821cs.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8822bs.ko
Signing module /var/lib/dkms/rtw88/0.6/build/rtw_8822cs.ko
Cleaning build area...

[... 149 lines cut ...]
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.12.25+rpt-rpi-v7/updates/dkms/

Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.12.25+rpt-rpi-v7/updates/dkms/

rtw_8822bs.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.12.25+rpt-rpi-v7/updates/dkms/

rtw_8822cs.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.12.25+rpt-rpi-v7/updates/dkms/
depmod.......`

a5a5aa555oo, 2025-08-31:
> Note this wasn't a fresh install, I had already installed using the modified .conf file, but I did follow your steps to remove installed drivers. Would it be helpful for me to try on a fresh install?

@randomcoder67-2 

This is sufficient, thank you for your testing! 👍