lwfinger/rtw88 #365: Drivers for wifi adapter with RTL8812AU chipset wont activate on Linux 6.15.2-arch1-1

BusterLundgaard, 2025-06-16:
Hi. For some time, to use a wifi adapter with a RTL8812AU chipset you had to use an unofficial driver, like the one at [here](https://github.com/morrownr/8812au-20210820). A few months ago, an official plug and play driver was released. Unfortunately, I can't get it to work, and a recent update to Arch broke the unofficial driver I had been relying on for my TP-Link RTL8812AU Archer T4U 802.11ac. 

I created an issue at the Github of the unofficial driver. After a bit of back and forth, they suspect the issue is with some header files in Arch. More details [here](https://github.com/morrownr/8812au-20210820/issues/49#issuecomment-2972590183).

**Potentially useful information:**
```
> sudo ls -l /lib/firmware/rtw88
total 412
-rw-r--r-- 1 root root   610 May 22 20:06 README.zst
-rw-r--r-- 1 root root 13117 May 22 20:06 rtw8703b_fw.bin.zst
-rw-r--r-- 1 root root 14647 May 22 20:06 rtw8703b_wow_fw.bin.zst
-rw-r--r-- 1 root root 17918 May 22 20:06 rtw8723d_fw.bin.zst
-rw-r--r-- 1 root root 16756 May 22 20:06 rtw8812a_fw.bin.zst
-rw-r--r-- 1 root root 36290 May 22 20:06 rtw8814a_fw.bin.zst
-rw-r--r-- 1 root root 19570 May 22 20:06 rtw8821a_fw.bin.zst
-rw-r--r-- 1 root root 53954 May 22 20:06 rtw8821c_fw.bin.zst
-rw-r--r-- 1 root root 77712 May 22 20:06 rtw8822b_fw.bin.zst
-rw-r--r-- 1 root root 85925 May 22 20:06 rtw8822c_fw.bin.zst
-rw-r--r-- 1 root root 64058 May 22 20:06 rtw8822c_wow_fw.bin.zst
```
```
> iw dev
(nothing)
```
```
> ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eno1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether 04:42:1a:2b:bd:3b brd ff:ff:ff:ff:ff:ff
    altname enp0s31f6
    altname enx04421a2bbd3b
```
When I plug out the wifi adapter and plug it in again, `sudo dmesg` shows me:
```
  970.688815] usb 2-1: USB disconnect, device number 13
[  986.002645] usb 1-1: new high-speed USB device number 19 using xhci_hcd
[  986.126279] usb 1-1: New USB device found, idVendor=2357, idProduct=0101, bcdDevice= 0.00
[  986.126292] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  986.126298] usb 1-1: Product: 802.11n NIC
[  986.126303] usb 1-1: Manufacturer: Realtek
[  986.126307] usb 1-1: SerialNumber: 123456
[  986.130140] rtw_8812au 1-1:1.0: Firmware version 52.14.0, H2C version 0
[  986.493786] usb 1-1: USB disconnect, device number 19
[ 1000.621555] xhci_hcd 0000:00:14.0: **WARN** Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
**[ 1000.833435] rtw_8812au 2-1:1.0: failed to validate firmware
[ 1000.833446] rtw_8812au 2-1:1.0: failed to download firmware**
[ 1001.090446] usb 2-1: new SuperSpeed USB device number 14 using xhci_hcd
[ 1001.101799] usb 2-1: Int endpoint with wBytesPerInterval of 512 in config 1 interface 0 altsetting 0 ep 133: setting to 64
[ 1001.101970] usb 2-1: New USB device found, idVendor=2357, idProduct=0101, bcdDevice= 0.00
[ 1001.101973] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1001.101976] usb 2-1: Product: 802.11n NIC
[ 1001.101978] usb 2-1: Manufacturer: Realtek
[ 1001.101980] usb 2-1: SerialNumber: 123456
[ 1001.104940] rtw_8812au 2-1:1.0: Firmware version 52.14.0, H2C version 0
[ 1008.330227] wlan0: authenticate with e4:f7:5b:24:d3:74 (local address=14:cc:20:16:a8:50)
[ 1008.431536] wlan0: send auth to e4:f7:5b:24:d3:74 (try 1/3)
[ 1008.433943] wlan0: authenticated
[ 1008.434279] wlan0: associate with e4:f7:5b:24:d3:74 (try 1/3)
[ 1008.454579] wlan0: RX AssocResp from e4:f7:5b:24:d3:74 (capab=0x11 status=0 aid=4)
[ 1008.458814] wlan0: associated
[ 1017.093239] usb 2-1: USB disconnect, device number 14
[ 1017.103226] wlan0: deauthenticating from e4:f7:5b:24:d3:74 by local choice (Reason: 3=DEAUTH_LEAVING)
[ 1017.325220] xhci_hcd 0000:00:14.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
```

BusterLundgaard, 2025-06-17:
I've tried a different USB port yeah, same story. The device has no cable. 

I doubt the hardware itself is broken though, cause it works on Windows for me right now.

dubhater, 2025-06-17:
Is it the same if you go back to kernel 6.14.x?

dubhater, 2025-06-19:
xone seems to be some Xbox driver. You can just delete it temporarily.

dubhater, 2025-06-19:
So it's the same with 6.14.

Can you try some patches?

```diff
diff --git a/drivers/net/wireless/realtek/rtw88/mac.c b/drivers/net/wireless/realtek/rtw88/mac.c
index 011b81c82f3b..6b883061463b 100644
--- a/mac.c
+++ b/mac.c
@@ -447,10 +447,10 @@ static void wlan_cpu_enable(struct rtw_dev *rtwdev, bool enable)
 		rtw_write8_set(rtwdev, REG_SYS_FUNC_EN + 1, BIT_FEN_CPUEN);
 	} else {
 		/* cpu io interface disable */
-		rtw_write8_clr(rtwdev, REG_SYS_FUNC_EN + 1, BIT_FEN_CPUEN);
+		rtw_write8_clr(rtwdev, REG_RSV_CTRL + 1, BIT_WLMCU_IOIF);
 
 		/* cpu disable */
-		rtw_write8_clr(rtwdev, REG_RSV_CTRL + 1, BIT_WLMCU_IOIF);
+		rtw_write8_clr(rtwdev, REG_SYS_FUNC_EN + 1, BIT_FEN_CPUEN);
 	}
 }
 
```

BusterLundgaard, 2025-06-19:
Okay, a couple of unfortunately stupid questions but:

With `a/drivers/net/wireless/realtek/rtw88/mac.c b/drivers/net/wireless/realtek/rtw88/mac.c` what directories are you referring to exactly? Like, just to make sure we're on the same page: I haven't explicitly cloned this github repository here. I was just told this was the right place to create this issue, as the driver was plug and play. 

So should I clone this repository, save the noted patches to .patch files and use git to apply them, or how exactly does this work?

Thanks a ton for all the help by the way.

BusterLundgaard, 2025-06-19:
dmesg --follow yields:

```
[  406.786857] usb 1-8: Manufacturer: Xiaomi
[  406.786861] usb 1-8: SerialNumber: a4e0911a233f
[  413.178209] usb 1-1: new high-speed USB device number 15 using xhci_hcd
[  413.301751] usb 1-1: New USB device found, idVendor=2357, idProduct=0101, bcdDevice= 0.00
[  413.301767] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  413.301773] usb 1-1: Product: 802.11n NIC
[  413.301778] usb 1-1: Manufacturer: Realtek
[  413.301782] usb 1-1: SerialNumber: 123456
[  413.306471] rtw_8812au 1-1:1.0: Firmware version 52.14.0, H2C version 0
[  413.422764] usb 1-1: USB disconnect, device number 15
[  413.567506] rtw88_core: exports duplicate symbol check_hw_ready (owned by rtw_core)
[  413.728458] usb 2-1: new SuperSpeed USB device number 2 using xhci_hcd
[  413.739786] usb 2-1: Int endpoint with wBytesPerInterval of 512 in config 1 interface 0 altsetting 0 ep 133: setting to 64
[  413.739943] usb 2-1: New USB device found, idVendor=2357, idProduct=0101, bcdDevice= 0.00
[  413.739952] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  413.739958] usb 2-1: Product: 802.11n NIC
[  413.739962] usb 2-1: Manufacturer: Realtek
[  413.739967] usb 2-1: SerialNumber: 123456
[  413.743968] rtw_8812au 2-1:1.0: Firmware version 52.14.0, H2C version 0
[  414.784949] xhci_hcd 0000:00:14.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[  414.784992] rtw_8812au 2-1:1.0: read register 0x5 failed with -110
[  414.998870] rtw_8812au 2-1:1.0: failed to check fw download ready
[  415.209954] rtw_8812au 2-1:1.0: failed to validate firmware
[  415.209964] rtw_8812au 2-1:1.0: failed to download firmware
[  415.210025] rtw_8812au 2-1:1.0: failed to setup chip efuse info
[  415.210030] rtw_8812au 2-1:1.0: failed to setup chip information
[  415.210192] rtw_8812au 2-1:1.0: probe with driver rtw_8812au failed with error -22
[  415.676117] usb 2-1: USB disconnect, device number 2
[  415.893273] usb 2-1: new SuperSpeed USB device number 3 using xhci_hcd
[  415.904608] usb 2-1: Int endpoint with wBytesPerInterval of 512 in config 1 interface 0 altsetting 0 ep 133: setting to 64
[  415.904765] usb 2-1: New USB device found, idVendor=2357, idProduct=0101, bcdDevice= 0.00
[  415.904774] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  415.904780] usb 2-1: Product: 802.11n NIC
[  415.904784] usb 2-1: Manufacturer: Realtek
[  415.904789] usb 2-1: SerialNumber: 123456
[  415.908984] rtw_8812au 2-1:1.0: Firmware version 52.14.0, H2C version 0
[  415.914028] rtw_8812au 2-1:1.0: write register 0x28 failed with -71
[  416.417510] xhci_hcd 0000:00:14.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[  416.417538] rtw_8812au 2-1:1.0: write register 0x100 failed with -110
[  416.631462] rtw_8812au 2-1:1.0: failed to check fw download ready
[  416.841543] rtw_8812au 2-1:1.0: failed to validate firmware
[  416.841553] rtw_8812au 2-1:1.0: failed to download firmware
[  416.841613] rtw_8812au 2-1:1.0: failed to setup chip efuse info
[  416.841618] rtw_8812au 2-1:1.0: failed to setup chip information
[  416.841778] rtw_8812au 2-1:1.0: probe with driver rtw_8812au failed with error -22
[  417.300668] usb 2-1: USB disconnect, device number 3
[  417.518821] usb 2-1: new SuperSpeed USB device number 4 using xhci_hcd
[  417.530236] usb 2-1: Int endpoint with wBytesPerInterval of 512 in config 1 interface 0 altsetting 0 ep 133: setting to 64
[  417.530394] usb 2-1: New USB device found, idVendor=2357, idProduct=0101, bcdDevice= 0.00
[  417.530403] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  417.530409] usb 2-1: Product: 802.11n NIC
[  417.530414] usb 2-1: Manufacturer: Realtek
[  417.530418] usb 2-1: SerialNumber: 123456
[  417.534455] rtw_8812au 2-1:1.0: Firmware version 52.14.0, H2C version 0
[  417.662068] rtw88_core: exports duplicate symbol check_hw_ready (owned by rtw_core)
[  417.712945] rtw88_core: exports duplicate symbol check_hw_ready (owned by rtw_core)
[  417.777484] rtw88_core: exports duplicate symbol check_hw_ready (owned by rtw_core)
[  418.597182] usb 2-1: USB disconnect, device number 4
[  418.634301] xhci_hcd 0000:00:14.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[  418.634400] rtw_8812au 2-1:1.0: write register 0xe90 failed with -110
[  418.906360] usb 2-1: new SuperSpeed USB device number 5 using xhci_hcd
[  418.917648] usb 2-1: Int endpoint with wBytesPerInterval of 512 in config 1 interface 0 altsetting 0 ep 133: setting to 64
[  418.917818] usb 2-1: New USB device found, idVendor=2357, idProduct=0101, bcdDevice= 0.00
[  418.917827] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  418.917833] usb 2-1: Product: 802.11n NIC
[  418.917837] usb 2-1: Manufacturer: Realtek
[  418.917842] usb 2-1: SerialNumber: 123456
[... 164 lines cut ...]
[  456.974156] rtw_8812au 2-1:1.0: Firmware version 52.14.0, H2C version 0
[  457.101338] rtw88_core: exports duplicate symbol check_hw_ready (owned by rtw_core)
[  458.087553] rtw_8812au 2-1:1.0: failed to validate firmware
[  458.087566] rtw_8812au 2-1:1.0: failed to download firmware
[  458.088648] rtw_8812au 2-1:1.0: leave idle state failed
[  458.089042] rtw_8812au 2-1:1.0: failed to leave ips state
[  458.089047] rtw_8812au 2-1:1.0: failed to leave idle state
[  458.907891] usb 2-1: USB disconnect, device number 20
[  459.181989] usb 2-1: new SuperSpeed USB device number 21 using xhci_hcd
[... cut ...]
[  459.953298] xhci_hcd 0000:00:14.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[  460.892612] usb 2-1: new SuperSpeed USB device number 22 using xhci_hcd
[  460.903992] usb 2-1: Int endpoint with wBytesPerInterval of 512 in config 1 interface 0 altsetting 0 ep 133: setting to 64
[  460.904157] usb 2-1: New USB device found, idVendor=2357, idProduct=0101, bcdDevice= 0.00
[  460.904166] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
```
It more or less keeps going on and on. Maybe the randomly dying thing you were talking about?

BusterLundgaard, 2025-06-19:
No wait it works!
Wonderful!
Thanks a ton for the help!

Do say if you'd like me to run any other tests on my machine that might be of info/importance to you/the project : )

BusterLundgaard, 2025-06-20:
The blacklisting seemed to help. `dmesg` now only writes:

```
[  102.379416] usb 1-1: USB disconnect, device number 15
[  112.066152] usb 1-1: new high-speed USB device number 16 using xhci_hcd
[  112.189636] usb 1-1: New USB device found, idVendor=2357, idProduct=0101, bcdDevice= 0.00
[  112.189650] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  112.189656] usb 1-1: Product: 802.11n NIC
[  112.189661] usb 1-1: Manufacturer: Realtek
[  112.189665] usb 1-1: SerialNumber: 123456
```

If i run the insmod commands now, however, I get:

```
insmod: ERROR: could not insert module ./rtw_core.ko: Unknown symbol in module
insmod: ERROR: could not insert module ./rtw_usb.ko: Unknown symbol in module
insmod: ERROR: could not insert module ./rtw_88xxa.ko: Unknown symbol in module
insmod: ERROR: could not insert module ./rtw_8812a.ko: Unknown symbol in module
insmod: ERROR: could not insert module ./rtw_8812au.ko: Unknown symbol in module
```
I then have no internet. If i remove the blacklist and follow the same commands you sent me, i get the same dmesg errors as before, but I do have internet.

On reboot, I need to run the commands again to get internet.

dubhater, 2025-06-20:
> insmod: ERROR: could not insert module ./rtw_core.ko: Unknown symbol in module

Right, that's because mac80211 is not loaded. Load it with ``modprobe mac80211``, then you can insert the rtw_* modules.

BusterLundgaard, 2025-06-25:
Tried the first patch now and it seems to work. I've only been using it for half an hour so far, but dmesg reports no issues and internet has been stable. I'll keep this open for now in case i get issues in the upcoming days, then close it.

Thanks a lot for the help :)

dubhater, 2025-06-25:
I'm glad it's working, but I'm confused about your other message:

> Hi, sorry for the long wait. Anyways, following the instructions now, I don't get the same error messages as before, but dmesg frequently reports:

What were you testing there? This repository without any patches?

BusterLundgaard, 2025-06-29:
So its unfortunately not been particularily stable. It will mostly last at a maximum for about half an hour before it goes out again. `dmesg` reports a lot of:

```
[  660.874581] rtw_8812au 2-1:1.0: mac power on failed
[  660.874587] rtw_8812au 2-1:1.0: failed to power on mac
[  660.874592] rtw_8812au 2-1:1.0: failed to setup chip efuse info
[  660.874596] rtw_8812au 2-1:1.0: failed to setup chip information
[  660.874747] rtw_8812au 2-1:1.0: probe with driver rtw_8812au failed with error -114
[  661.333379] usb 2-1: USB disconnect, device number 117
[  662.117393] usb 2-1: new SuperSpeed USB device number 118 using xhci_hcd
[  662.128694] usb 2-1: Int endpoint with wBytesPerInterval of 512 in config 1 interface 0 altsetting 0 ep 133: setting to 64
[  662.128865] usb 2-1: New USB device found, idVendor=2357, idProduct=0101, bcdDevice= 0.00
[  662.128874] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  662.128880] usb 2-1: Product: 802.11n NIC
[  662.128885] usb 2-1: Manufacturer: Realtek
[  662.128889] usb 2-1: SerialNumber: 123456
[  662.132937] rtw_8812au 2-1:1.0: Firmware version 52.14.0, H2C version 0
[  662.156908] rtw_8812au 2-1:1.0: write register 0x1fd8 len 8 failed with -71
[  662.658368] xhci_hcd 0000:00:14.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[  662.658400] rtw_8812au 2-1:1.0: write register 0x1fe0 len 8 failed with -110
[  662.869301] rtw_8812au 2-1:1.0: failed to validate firmware
[  662.869310] rtw_8812au 2-1:1.0: failed to download firmware
[  662.869325] rtw_8812au 2-1:1.0: failed to setup chip efuse info
[  662.869330] rtw_8812au 2-1:1.0: failed to setup chip information
[  662.869483] rtw_8812au 2-1:1.0: probe with driver rtw_8812au failed with error -22
[  664.789171] usb 2-1: USB disconnect, device number 118
[  665.893134] usb 2-1: new SuperSpeed USB device number 119 using xhci_hcd
[  665.904384] usb 2-1: Int endpoint with wBytesPerInterval of 512 in config 1 interface 0 altsetting 0 ep 133: setting to 64
[  665.904553] usb 2-1: New USB device found, idVendor=2357, idProduct=0101, bcdDevice= 0.00
[  665.904562] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  665.904567] usb 2-1: Product: 802.11n NIC
[  665.904572] usb 2-1: Manufacturer: Realtek
[  665.904576] usb 2-1: SerialNumber: 123456
[  665.908572] rtw_8812au 2-1:1.0: Firmware version 52.14.0, H2C version 0
[  666.837022] usb 2-1: USB disconnect, device number 119
[  667.996871] usb 2-1: new SuperSpeed USB device number 120 using xhci_hcd
[  668.008149] usb 2-1: Int endpoint with wBytesPerInterval of 512 in config 1 interface 0 altsetting 0 ep 133: setting to 64
[  668.008303] usb 2-1: New USB device found, idVendor=2357, idProduct=0101, bcdDevice= 0.00
[  668.008305] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  668.008306] usb 2-1: Product: 802.11n NIC
[  668.008306] usb 2-1: Manufacturer: Realtek
[  668.008308] usb 2-1: SerialNumber: 123456
[  668.011162] rtw_8812au 2-1:1.0: Firmware version 52.14.0, H2C version 0
[  668.636845] usb 2-1: USB disconnect, device number 120
[  668.673912] xhci_hcd 0000:00:14.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[  668.673970] rtw_8812au 2-1:1.0: write register 0x17a8 len 196 failed with -110
[  670.134779] usb 1-1: new high-speed USB device number 14 using xhci_hcd
[  670.258430] usb 1-1: New USB device found, idVendor=2357, idProduct=0101, bcdDevice= 0.00
[  670.258443] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  670.258450] usb 1-1: Product: 802.11n NIC
[  670.258454] usb 1-1: Manufacturer: Realtek
[  670.258459] usb 1-1: SerialNumber: 123456
[  670.263263] rtw_8812au 1-1:1.0: Firmware version 52.14.0, H2C version 0
[  670.479714] rtw_8812au 1-1:1.0: failed to check fw download ready
[  670.480562] rtw_8812au 1-1:1.0: write register 0x1000 len 196 failed with -71
[  670.480691] rtw_8812au 1-1:1.0: write register 0x10c4 len 196 failed with -71
[  670.480807] rtw_8812au 1-1:1.0: write register 0x1188 len 196 failed with -71
[  670.480927] rtw_8812au 1-1:1.0: write register 0x124c len 196 failed with -71
[  670.481047] rtw_8812au 1-1:1.0: write register 0x1310 len 196 failed with -71
[  670.481178] rtw_8812au 1-1:1.0: write register 0x13d4 len 196 failed with -71
[  670.481296] rtw_8812au 1-1:1.0: write register 0x1498 len 196 failed with -71
[  670.481415] rtw_8812au 1-1:1.0: write register 0x155c len 196 failed with -71
[  670.481535] rtw_8812au 1-1:1.0: write register 0x1620 len 196 failed with -71
[  670.481675] rtw_8812au 1-1:1.0: write register 0x16e4 len 196 failed with -71
[  670.481790] rtw_8812au 1-1:1.0: write register 0x17a8 len 196 failed with -71
[  670.481910] rtw_8812au 1-1:1.0: write register 0x186c len 196 failed with -71
[  670.482030] rtw_8812au 1-1:1.0: write register 0x1930 len 196 failed with -71
[  670.482151] rtw_8812au 1-1:1.0: write register 0x19f4 len 196 failed with -71
[  670.482270] rtw_8812au 1-1:1.0: write register 0x1ab8 len 196 failed with -71
[  670.482390] rtw_8812au 1-1:1.0: write register 0x1b7c len 196 failed with -71
[... 249 lines cut ...]
[  670.515167] rtw_8812au 1-1:1.0: write register 0x1938 len 8 failed with -71
[  670.515275] rtw_8812au 1-1:1.0: write register 0x1940 len 8 failed with -71
[  670.515394] rtw_8812au 1-1:1.0: write register 0x1948 len 8 failed with -71
[  670.515529] rtw_8812au 1-1:1.0: write register 0x1950 len 8 failed with -71
[... cut ...]
[  670.730976] rtw_8812au 1-1:1.0: failed to setup chip information
[  670.731113] rtw_8812au 1-1:1.0: probe with driver rtw_8812au failed with error -22
[  670.731398] usb 1-1: USB disconnect, device number 14
[  672.587584] usb 1-8: new high-speed USB device number 15 using xhci_hcd
[  672.900625] usb 1-8: device descriptor read/64, error -71
[  673.124941] usb 1-8: New USB device found, idVendor=2357, idProduct=0101, bcdDevice= 0.00
[  673.124946] usb 1-8: New USB device strings: Mfr=1, Product=2, SerialNumber=3

```

149segolte, 2025-09-03:
Well I have a alfa network's adapter with rtl8812au I am using it with an RPi 4 on NixOS. Previously I was using `config.boot.kernelPackages.rtl88xxau-aircrack` it worked fine but caused issues with hostapd. So I switched to `config.boot.kernelPackages.rtw88`.
The system is using linux kernel LTS:
```
Linux rpinix 6.12.44 #1-NixOS SMP Thu Aug 28 14:31:16 UTC 2025 aarch64 GNU/Linux
```
I am having similar issue with `rtw88`.

- When plugging into USB3 ports the device immediately disconnects (and does not come back until replugging).
`dmesg` output:
```
[  951.349074] usb 1-1.1: new high-speed USB device number 13 using xhci_hcd
[  951.443734] rtw_8812au 1-1.1:1.0: Firmware version 52.14.0, H2C version 0
[  951.790991] usb 1-1.1: USB disconnect, device number 13
```
`lsusb -vt` output:
```
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/1p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 001: Dev 002, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 2109:3431 VIA Labs, Inc. Hub
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
```

- When plugging into USB2 ports, it works but is prone to errors on slightest touch of cables.
`dmesg` output:
```
[  918.244721] usb 1-1.4: new high-speed USB device number 8 using xhci_hcd
[  918.337584] rtw_8812au 1-1.4:1.0: Firmware version 52.14.0, H2C version 0
[  918.555568] rtw_8812au 1-1.4:1.0 wlp1s0u1u4: renamed from wlan1
```
`lsusb -vt` output:
```
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/1p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 001: Dev 002, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 2109:3431 VIA Labs, Inc. Hub
        |__ Port 003: Dev 016, If 0, Class=Vendor Specific Class, Driver=rtw_8812au, 480M
            ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
```
`ip a` output:
```
7: wlp1s0u1u3: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether 7a:3e:24:68:98:34 brd ff:ff:ff:ff:ff:ff permaddr 00:c0:ca:ad:8e:0f
    altname wlx00c0caad8e0f
```

I would like to get this working with USB3 ports and be a bit more stable. On NixOS I don't know how to build the custom patches mentioned above, but are there any config options or kernel options that I can try to make this work? I am hoping the solution is going to be simple enough as the adapter is fully functional in the USB2 ports.