lwfinger/rtw88 #366: rtw_8822bu: write RF mode table fail

icegood, 2025-06-16:
I added some debug log in my version
https://github.com/icegood/rtw88
 and this is what i obtained:

[127326.123525] rtw_8822bu 2-2:1.0: Firmware version 30.20.0, H2C version 14
[127326.324076] usbcore: registered new interface driver rtw_8822bu
[127452.120070] rtw_8822bu 2-2:1.0: LUT write attempt 5: rf_reg33=0xaeaea
[127452.127490] rtw_8822bu 2-2:1.0: LUT write attempt 4: rf_reg33=0xaeaea
[127452.134942] rtw_8822bu 2-2:1.0: LUT write attempt 3: rf_reg33=0xaeaea
[127452.142392] rtw_8822bu 2-2:1.0: LUT write attempt 2: rf_reg33=0xaeaea
[127452.150047] rtw_8822bu 2-2:1.0: LUT write attempt 1: rf_reg33=0xaeaea
[127452.157336] ------------[ cut here ]------------
[127452.162574] write RF mode table fail
[127452.166668] WARNING: CPU: 8 PID: 25957 at 0xffffffffa0958037 [rtw_8822b@00000000f3511045+0x5000]
[127452.176601] Modules linked in: rtw_8822bu(O) rtw_8822b(O) rtw_usb(O) rtw_core(O) mac80211(O) cfg80211(O) pppoe ppp_async i915 video pppox ppp_generic nft_fib_inet nf_flow_table_inet ipt_REJECT zstd xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_addrtype xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY wmi slhc nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_reject_ipv4 nf_log_syslog nf_flow_table nf_conntrack_netlink nf_conncount lzo_rle lzo iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables forcedeth e1000e drm_suballoc_helper drm_exec drm_display_helper drm_buddy crc_ccitt br_netfilter bnx2 act_connmark
[127452.176692]  sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact drivetemp i2c_dev ledtrig_usbport compat(O) it87(O) xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink dwmac_intel ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 dwmac_generic stmmac_platform stmmac ixgbe e1000 amd_xgbe ifb mdio veth tun autofs4 nls_utf8 pcs_xpcs ena xxhash_generic sha512_ssse3 sha512_generic seqiv sha3_generic jitterentropy_rng drbg hmac geniv rng cmac blake2b_generic crypto_acompress nls_iso8859_1 nls_cp437 r8169 xhci_plat_hcd fsl_mph_dr_of ehci_platform ehci_fsl igb igc vfat fat btrfs
[127452.277362]  zstd_decompress zstd_compress zstd_common xxhash xor raid6_pq lzo_decompress lzo_compress libcrc32c button_hotplug(O) tg3 realtek phylink mii [last unloaded: cfg80211(O)]

....

dubhater, 2025-06-17:
This "write RF mode table fail" was also reported with RTL8822BE (PCI) but the maintainers couldn't reproduce it.

Is it very easy to trigger this warning?

Is there any problem other than the warning? I mean, does the device work?

What kind of computer are you using? What kernel version?

0xaeaea is a clue. Hardware registers have the value 0xeaeaeaea when some part of the hardware is not powered on. (Here it's truncated because the RF registers are only 20 bits.)

Can you modify the warning to print the values of tx_path, rx_path, is_tx2_path, and efuse->rfe_option?

icegood, 2025-06-17:
Warning leads to failure and nothing works after that

hbalint2100, 2025-11-24:
I've received a similar warning also in openwrt:

[  406.705972] ------------[ cut here ]------------
[  406.710517] WARNING: CPU: 0 PID: 10287 at backports-6.12.44/drivers/net/wireless/realtek/rtw88/rtw8822b.c:824 rtw8822b_set_antenna+0x3c/0x74 [rtw88_8822b]
[  406.724296] write RF mode table fail
[  406.727768] Modules linked in: rtw88_8822cu(O) rtw88_8822c(O) rtw88_8822bu(O) rtw88_8822b(O) pppoe ppp_async wireguard rtw88_usb(O) rtw88_core(O) qcserial pppox ppp_generic option nft_fib_inet nf_flow_table_inet mac80211(O) libchacha20poly1305 ipt_REJECT cfg80211(O) xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_policy xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY xfrm_interface usb_wwan slhc rtc_ds1307 rndis_host rfcomm qmi_wwan poly1305_arm nft_xfrm nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_compat nft_chain_nat nf_tables nf_reject_ipv4 nf_nat nf_log_syslog nf_flow_table nf_conntrack_bridge nf_conncount mdio_netlink(O) lm75 libcurve25519_generic libcrc32c iptable_raw
[  406.728658]  iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables hwmon hidp hci_uart crc_ccitt compat(O) chacha_neon cdc_ether btusb btrtl btmtk btintel bnep bluetooth act_connmark nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact configs hid ledtrig_pattern ledtrig_oneshot cryptodev(O) xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 swconfig qmi_wwan_q(O) usbnet cdc_wdm ip6_gre ip_gre gre ifb dummy ip6_vti ip_vti ip6_udp_tunnel udp_tunnel ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ipip ip6_tunnel tunnel6 tunnel4 ip_tunnel vrf
[  406.818527]  tun xfrm_user xfrm_ipcomp af_key xfrm_algo ecdh_generic ecc crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_arm sha1_generic sha1_arm md5 kpp echainiv ecb des_generic libdes cmac cbc authencesn authenc gpio_keys_polled ledtrig_transient fsl_mph_dr_of ehci_platform ehci_fsl microchip mii
[  406.937051] CPU: 0 PID: 10287 Comm: iw Tainted: G        W  O       6.6.104 #0
[  406.944163] Hardware name: Atmel SAMA5
[  406.947867]  unwind_backtrace from show_stack+0x10/0x14
[  406.953043]  show_stack from dump_stack_lvl+0x24/0x2c
[  406.958033]  dump_stack_lvl from __warn+0x98/0xc0
[  406.962668]  __warn from warn_slowpath_fmt+0x68/0x78
[  406.967561]  warn_slowpath_fmt from rtw8822b_set_antenna+0x3c/0x74 [rtw88_8822b]
[  406.974877]  rtw8822b_set_antenna [rtw88_8822b] from rtw_ops_set_antenna+0x4c/0x6c [rtw88_core]
[  406.983566]  rtw_ops_set_antenna [rtw88_core] from ieee80211_set_antenna+0x34/0x5c [mac80211]
[  406.992524]  ieee80211_set_antenna [mac80211] from nl80211_set_wiphy+0x37c/0x634 [cfg80211]
[  407.001418]  nl80211_set_wiphy [cfg80211] from genl_rcv_msg+0x24c/0x390
[  407.008146]  genl_rcv_msg from netlink_rcv_skb+0xb8/0x11c
[  407.013465]  netlink_rcv_skb from genl_rcv+0x28/0x34
[  407.018358]  genl_rcv from netlink_unicast+0x220/0x314
[  407.023423]  netlink_unicast from netlink_sendmsg+0x1cc/0x434
[  407.029089]  netlink_sendmsg from ____sys_sendmsg+0x1f0/0x284
[  407.034758]  ____sys_sendmsg from ___sys_sendmsg+0x6c/0xa4
[  407.040165]  ___sys_sendmsg from sys_sendmsg+0x44/0x74
[  407.045230]  sys_sendmsg from ret_fast_syscall+0x0/0x54
[  407.050380] Exception stack(0xc242bfa8 to 0xc242bff0)
[  407.055370] bfa0:                   00000000 00000000 00000003 beed34b8 00000000 00000000
[  407.063442] bfc0: 00000000 00000000 b6fbca20 00000128 b6fbc910 b6fbc980 0002c348 00000046
[  407.071507] bfe0: beed3468 beed3458 b6f9cdf8 b6f9c044
[  407.077491] ---[ end trace 0000000000000000 ]---
[  412.936540] switch: port 6(wlan0) entered blocking state
[  412.941943] switch: port 6(wlan0) entered disabled state
[  412.947251] rtw_8822bu 1-2.1:1.2 wlan0: entered allmulticast mode
[  412.953822] rtw_8822bu 1-2.1:1.2 wlan0: entered promiscuous mode
[  412.960275] switch: port 6(wlan0) entered blocking state
[  412.965594] switch: port 6(wlan0) entered forwarding state
[  413.036701] switch: port 6(wlan0) entered disabled state

dubhater, 2025-11-25:
This gets rid of the warning for me:

```diff
diff --git a/drivers/net/wireless/realtek/rtw88/rtw8822b.c b/drivers/net/wireless/realtek/rtw88/rtw8822b.c
index 89b6485b229a..4d88cc2f4148 100644
--- a/drivers/net/wireless/realtek/rtw88/rtw8822b.c
+++ b/drivers/net/wireless/realtek/rtw88/rtw8822b.c
@@ -1005,7 +1005,8 @@ static int rtw8822b_set_antenna(struct rtw_dev *rtwdev,
 	hal->antenna_tx = antenna_tx;
 	hal->antenna_rx = antenna_rx;
 
-	rtw8822b_config_trx_mode(rtwdev, antenna_tx, antenna_rx, false);
+	if (test_bit(RTW_FLAG_POWERON, rtwdev->flags))
+		rtw8822b_config_trx_mode(rtwdev, antenna_tx, antenna_rx, false);
 
 	return 0;
 }
```

Please test it.