lwfinger/rtw88 #368: Alfa AWUS036ACH Stops Responding in Promiscuous Mode; `rtw_8812au`

electronsandstuff, 2025-06-23:
Hey folks, thank you for the work in backporting the Realtek drivers. I am using them on a Raspberry Pi 4 with Raspberry Pi OS (kernel 6.12.25+rpt-rpi-v8). When using the driver with the card in promiscuous mode and with packet injection, the card stops responding after approximately one minute of use. Upon exiting the tool, I can no longer see the wireless interface and I have to reboot for it to show up again. I receive some of the same error messages reported in #344.

Here is an excerpt from `dmesg`.
```
[ 5773.532855] rtw_8812au 2-1:1.0: failed to get tx report from firmware
[ 7075.147821] rtw_8812au 2-1:1.0: failed to get tx report from firmware
[ 8629.531824] rtw_8812au 2-1:1.0: failed to get tx report from firmware
...  (messages repeat)
[64884.848557] usb 2-1: USB disconnect, device number 2
[64885.324615] xhci_hcd 0000:01:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[64885.324643] rtw_8812au 2-1:1.0: write register 0xf008 failed with -110
[64885.325921] rtw_8812au 2-1:1.0 wlan1 (unregistering): left promiscuous mode
```

Here is some from `journalctl`.
```
Jun 23 11:07:04 mach kernel: usb 2-1: USB disconnect, device number 2
Jun 23 11:07:05 mach kernel: xhci_hcd 0000:01:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
Jun 23 11:07:05 mach kernel: rtw_8812au 2-1:1.0: write register 0xf008 failed with -110
Jun 23 11:07:05 mach kernel: rtw_8812au 2-1:1.0 wlan1 (unregistering): left promiscuous mode
Jun 23 11:07:05 mach systemd[1]: Starting systemd-rfkill.service - Load/Save RF Kill Switch Status...
Jun 23 11:07:05 mach avahi-daemon[414]: Withdrawing workstation service for wlan1.
Jun 23 11:07:05 mach systemd[1]: Started systemd-rfkill.service - Load/Save RF Kill Switch Status.
Jun 23 11:07:05 mach sudo[1321]: pam_unix(sudo:session): session closed for user root
Jun 23 11:07:05 mach NetworkManager[511]: <info>  [1750702025.2778] radio killswitch /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.>
Jun 23 11:07:10 mach systemd[1]: systemd-rfkill.service: Deactivated successfully.
```
Let me know what other log files I should be looking at or if there are any other tips you have for debugging.

dubhater, 2025-06-23:
The full output of dmesg could be helpful.

Are you sure the power supply can provide enough current?

Does it do the same in USB 2 mode? You can plug it in a USB 2 port, or use a module parameter:

```
options rtw_usb switch_usb_mode=n
```

electronsandstuff, 2025-06-24:
I am watching the Raspberry Pi undervoltage flag and don't see anything warning me about it.

I just moved over to a USB 2 port and am still experiencing the issue. Here is the full output of `dmesg` from this run.

<details>

<summary>Output from dmesg</summary>

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.12.25+rpt-rpi-v8 (serge@raspberrypi.com) (aarch64-linux-gnu-gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils f                                                       or Debian) 2.40) #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.5
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000e800000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000000e800000..0x000000002e7fffff (524288 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000003ef667c0..0x000000003ef667f5 (0 KiB) nomap non-reusable nvram@0
[    0.000000] OF: reserved mem: 0x000000003ef66580..0x000000003ef6677f (0 KiB) nomap non-reusable nvram@1
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Faking node 0 at [mem 0x0000000000000000-0x000000003b3fffff] (948MB)
[    0.000000] NUMA: Initialized distance table, cnt=1
[    0.000000] NODE_DATA(0) allocated [mem 0x3b1da300-0x3b1dcfff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0, zone DMA: 19456 pages in unavailable ranges
[    0.000000] percpu: Embedded 33 pages/cpu s95064 r8192 d31912 u135168
[    0.000000] pcpu-alloc: s95064 r8192 d31912 u135168 alloc=33*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 cgroup_disable=memory numa_policy=interleave s                                                       nd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  numa=fake=1 system_heap.max_order=0 smsc95xx.macaddr=E4:5F:01:B                                                       A:5B:FD vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=286e4e98-02 rootfstype=ext4 fsck.rep                                                       air=yes rootwait cfg80211.ieee80211_regdom=US
[    0.000000] cgroup: Disabling memory control group subsystem
[    0.000000] mempolicy: NUMA default policy overridden to 'interleave:0'
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] Fallback order for Node 0: 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 242688
[    0.000000] Policy zone: DMA
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 44909 entries in 176 pages
[    0.000000] ftrace: allocated 176 pages with 3 groups
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] RCU Tasks: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.000000] RCU Tasks Rude: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.000000] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[... 579 lines cut ...]
[   71.456192] 00000140: e0 48 e0 70 27 00 78 ea 60 d0 e1 20 32 63 4f 3c  .H.p'.x.`.. 2cO<
[   71.456195] 00000150: 12 40 92 15 8b 4a 4f 8a 15 d1 98 00 ff ff ff ff  .@...JO.........
[   71.456198] 00000160: ff ff ff ff ff ff ff ff 00 02 e1 63 12 29 03 00  ...........c.)..
[   71.456201] 00000170: cb 3f a0 e0 c2 f0 f7 cf 01 36 08 32 05 eb 94 d7  .?.......6.2....
[... cut ...]
[  155.811946] rtw_8812au 1-1.3:1.0: failed to get tx report from firmware
[  161.315988] rtw_8812au 1-1.3:1.0: failed to get tx report from firmware
[  179.172094] rtw_8812au 1-1.3:1.0: failed to get tx report from firmware
[  223.140262] rtw_8812au 1-1.3:1.0: failed to get tx report from firmware
[  233.060265] rtw_8812au 1-1.3:1.0: failed to get tx report from firmware
[  235.972273] rtw_8812au 1-1.3:1.0: failed to get tx report from firmware
[  239.236280] rtw_8812au 1-1.3:1.0: failed to get tx report from firmware
```

</details>

Snowy-Fluffy, 2025-06-29:
same problem on same device

Snowy-Fluffy, 2025-06-29:
The errors in dmesg are the same

dubhater, 2025-07-03:
@YipYup Unfortunately that's just a coincidence. switch_usb_mode and support_bf already have the value y by default, and disable_aspm, disable_msi, and disable_lps_deep have no effect on the USB devices.

What if you change your software setup to monitor the 2.4 GHz band with the two failing devices, leaving everything connected exactly the way it is now?

@YipYup and @Snowy-Fluffy Did you install the driver from this repository? What is your hardware?

Snowy-Fluffy, 2025-07-03:
> [@YipYup](https://github.com/YipYup) Unfortunately that's just a coincidence. switch_usb_mode and support_bf already have the value y by default, and disable_aspm, disable_msi, and disable_lps_deep [...]

I installed the driver from this repository via dkms, tried on my x86-64 laptop with arch linux and on my x86-64 computer with arch linux but everywhere my alfa awus036ach adapter did not work properly with this driver

Snowy-Fluffy, 2025-07-03:
It took about 20 minutes before the adapter stopped working in AC mode (these two lines repeated over a hundred times in dmesg [ 2244.329454] rtw_8812au 1-4:1.0: failed to download drv rsvd page
[ 2247.611441] rtw_8812au 1-4:1.0: error beacon valid)

electronsandstuff, 2025-07-06:
Update: I installed the deprecated drivers at https://github.com/aircrack-ng/rtl8812au and these seem to work on the Raspberry Pi OS. Happy to help continue debugging if there are suggestions as to what to try.

dubhater, 2025-07-06:
Try kernel 6.13 to see if it's a regression. It supports 0bda:8812 but it lacks some features and fixes.

G3zz, 2025-07-29:
I think I am also hitting this issue on a Raspberry Pi 5, running the latest stable raspberry pi kernel. I am trying to run a 5GHz AP, but find that the SSID drops after an indeterminate amount of time, and needs rebooting or unplugging.

The relevant dmesg lines are:

```txt
[  681.031976] rtw_8812au 4-1.1:1.0: read register 0x4c failed with -110
[  683.312968] rtw_8812au 3-1.1:1.0: write register 0x5 failed with -71
```

uname -a 
```txt
Linux rpi5 6.12.34+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.34-1+rpt1~bookworm (2025-06-26) aarch64 GNU/Linux
```

The dongle is connected to a Raspberry Pi branded powered USB 3 hub, so there should be enough power.

I have been experimenting with various `hostapd` configurations - all I can say is that `ieee80211ac=1` never works, but having it disabled, I get brief success for an indeterminate amount of time.

Is there anything that I can do to try and help debug the issue further? Should I just use the aircrack-ng driver for now?

andreas-lundgren-alsterit, 2026-09-03:
Hey, wanted to add a data point that lines up with this bug — running an Alfa AWUS036ACH v2 (RTL8812AU) inside a Kali Linux VM (VirtualBox on macOS, USB passthrough), kernel 7.1.5+kali-amd64, using the in-kernel `rtw88_8812au` driver.

Monitor mode enables fine and the interface looks healthy (`iw dev` shows `type monitor`, correct channel), but it captures zero traffic — not just my own packets, literally nothing, not even neighboring beacon frames. Tested with both `airodump-ng` and plain `tcpdump -i wlan0`, same result on multiple channels, both 2.4GHz and 5GHz.

I also hit `failed to get tx report from firmware` in dmesg early on, and separately had the USB device drop entirely after driver init (though that second part turned out to be VirtualBox passthrough flakiness, not this driver).

Normal client mode (associating to my home AP) works fine on this driver — it's specifically monitor-mode RX that's broken for me. Ended up switching to the DKMS driver (`realtek-rtl88xxau-dkms`) as a workaround, which restored monitor mode capture immediately, so it does look driver-specific rather than a hardware issue.

Happy to pull more logs/dmesg output if it's useful for tracking this down.