lwfinger/rtw88 #379: Bluetooth audio unusable with RTW88-8723DU

mya4k, 2025-08-01:
When connected to a 2.4GHz network, bluetooth audio gets cut off all the time for on average 5-10 seconds. Any way to fix this issue? Thanks.

dubhater, 2025-08-02:
If you run ``iw event -T`` do you see "scan started" when the audio cuts off?

dubhater, 2025-08-02:
Yes, there is a file in debugfs:

``watch -n 1 cat /sys/kernel/debug/ieee80211/phyX/rtw88/coex_info``

dubhater, 2025-08-03:
One thing to check is the TDMA line in the Mechanism section.

But post the entire file here because I'm curious about some of the details that don't change.

mya4k, 2025-08-03:
TDMA stays on case-13 if audio is uninterrupted. However case number changes around periods of interruptions, but doesn't change exclusively when the audio is not being played back.

Here's the content
```
**********************************************
                BT Coexist info 2
**********************************************
Mech/ RFE                                = Shared/ 0
Coex Ver/ BT Dez/ BT Rpt                 = 2007022f/ 0x2f/ 0x0000002f (Match)
Role/ RoleSwCnt/ IgnWL/ Feature          = Master/ 0/ 0
WL FW/ BT FW/ BT FW Desired/ KT          = 48.0/ 0x96f1/ 0x0/ D
AFH Map                                  = 0/ 0/ 0/ ch-(6)
RSSI/ STA Channel                        = -46dBm/ 6
TP {Tx, Rx} = {0, 0}Mbps
[Tx Rate] =  MCS7     (0x13)
[Rx Rate] =  CCK_1M   (0x0)
**********************************************
                BT Status
**********************************************
BT status/ rssi/ retry/ pop              = busy/ -46dBm/ 0/ 573
Profiles                                 = A2DP, (multi-link 0)
Reinit/ Relink/ IgnWl/ Feature           = 0/ 9/ 0/ 0x00000677
Page/ Inq/ iqk/ iqk fail                 = 11/ 0/ 0/ 0
0xae/ 0xac/ score board (W->B)/ (B->W)   = 0x0001/ 0x9000/ 0x0003/ 0x0012
Hi-Pri TX/RX, Lo-Pri TX/RX               = 0/0, 1396/134
WL_FW                                    = 00 00 00 00 00 00 00
BT_RSP                                   = 01 49 10 13 00 04 00
BT_ACT                                   = 02 49 10 16 00 04 00
**********************************************
                WiFi Status
**********************************************
Scanning                                 = 0
G_busy/ TX/ RX                           = 0/ TX 0 Mbps/ RX 0 Mbps
IPS/ Low Power/ PS mode                  = 0/ 0/ 0
Iface on Port (0)
        Beacon interval                  = 100
        Network Type                     = 2
        Peer   0
                RSSI                     = 54
                BW mode                  = 1
**********************************************
                Mechanism
**********************************************
TDMA                                     = 51 08 03 10 55 (case-14)
Coex Mode/Free Run/Timer base            = 2G1PORT/ No/ 3
Table/ 0x6c0/ 0x6c4/ 0x6c8               = 9(9)/ 0x66555555/ 0x5a5a5a5a/ 0xf0ffffff
0x778/ 0x6cc/ Run Count/ Reason          = 0x00000001/ 0x1800001b/ 67021/ reason (BTINFO)
AFH Map to BT                            = 00 00 00
AntDiv/ BtCtrlLPS/ g_busy                = On/ 0
Null All/ Retry/ Ack/ BT Empty/ BT Late  = 0/ 0/ 0/ 0/ 0
Cnt TDMA Toggle/ Lk 5ms/ Lk 5ms on/ FW   = 0/ 0/ No/ 0
WL_TxPw/ BT_TxPw/ WL_Rx/ BT_LNA_Lvl      = 0/ 0/ Off/ 7
**********************************************
                HW setting
**********************************************
LTE Coex/ Path Owner                     = OFF/ WL
GNT_WL_CTRL/ GNT_BT_CTRL/ Dbg            = RF:HW_BB:HW/ RF:HW_BB:HW/ Off
GNT_WL/ GNT_BT                           = 0/ 1
CRC OK CCK/ OFDM/ HT/ VHT                = 18/ 64/ 0/ 0
CRC ERR CCK/ OFDM/ HT/ VHT               = 1/ 24/ 71/ 0
HiPr/ Locking/ Locked/ Noisy             = N/ N/ N/ 0
948/ 67[7]                               = 0x99000080/ 0x1
964[1]/ 864[0]/ ab7[5]/ a01[7]           = 0x0/ 0x1/ 0x1/ 0x0
430/ 434/ 42a/ 426/ 45e[3]               = 0x1000000/ 0x4030201/ 0x808/ 0x3f/ 0x1
4c6[4]/ 40[5]/ 550/ 522/ 953[1]          = 0x1/ 0x1/ 0x1008/ 0x6f/ 0x0
EVM A/ EVM B/ SNR A/ SNR B               = 0/ 0/ -31/ 0
CCK-CCA/CCK-FA/OFDM-CCA/OFDM-FA          = 55/ 28/ 333/ 204
CRC OK CCK/11g/11n/11ac                  = 18/ 64/ 0/ 0
CRC Err CCK/11g/11n/11ac                 = 1/ 24/ 71/ 0
```

dubhater, 2025-08-03:
And what is the TDMA case when the audio gets interrupted?

mya4k, 2025-08-03:
Normal case: `51 08 03 10 54 (case-13)`
There was a 20 second period of no audio, before which TDMA value to siwtched to `51 08 03 10 55 (case-14)` for a bit and back to `51 08 03 10 54 (case-13)`, which is when the audio cut off. Then it switched back to case-14 briefly and back to case-13. Audio resumed shortly after.

dubhater, 2025-08-03:
Hmm, I think you have a different problem than mine. For me, the audio only gets interrupted during scans (TDMA case 11) and the interruptions last 1-2 seconds, no more. TDMA cases 13 and 14 work fine here.

Is your audio local or streamed through wifi?

mya4k, 2025-08-03:
My interruptions vary in length. Most of the time it is small audio drops but then it can do completely silent starting from a second upto 20 seconds.

I am simply using my bluetooth headset locally.

mya4k, 2025-08-03:
Yeah it was the same, but I have noticed something interesting. I tried my bluetooth earbuds for a change, and they weren't affected as heavily with interrupts as did my bluetooth headphones. And I have also noticed that long audio pauses are only affecting my headphones when they're on the AAC codec.