lwfinger/rtw88 #385: Commit 549f33c cuses IRQ issues with rtw_8821ce

daniviga, 2025-08-17:
Running latest master 52072d8 the following system hangs after few minutes after Wi-Fi gets connected

```
Operating System: Fedora Linux 42
KDE Plasma Version: 6.4.4
KDE Frameworks Version: 6.17.0
Qt Version: 6.9.1
Kernel Version: 6.15.9-201.fc42.x86_64 (64-bit)
Graphics Platform: Wayland
Processors: 4 × Intel® Celeron® J4125 CPU @ 2.00GHz
Memory: 8 GiB of RAM (7.6 GiB usable)
Graphics Processor: Intel® UHD Graphics 600
```

Involved modules:
```
rtw_8821ce
rtw_8821c
rtw_pci
rtw_core
```

Reverting commit 549f33c from master 52072d8 solved the issue and made the system back to a fully stable condition

Issue #383 might be related to this.

Due to the kind of failure I was unable most of the time to collect good logs but one:

```
Aug 13 23:59:44 touch-mini kernel: BUG: scheduling while atomic: swapper/2/0/0x00000102
Aug 13 23:59:44 touch-mini kernel: Modules linked in: rfcomm snd_seq_dummy snd_hrtimer nf_conntrack_netbios_ns nf_conntrack_broadcast nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables snd_sof_pci_intel_apl snd_sof_intel_hda_generic soundwire_intel snd_sof_intel_hda_sdw_bpt snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_intel_hda_mlink snd_sof_intel_hda soundwire_cadence snd_sof_pci snd_sof_xtensa_dsp snd_sof snd_sof_utils snd_soc_acpi_intel_match snd_soc_acpi_intel_sdca_quirks soundwire_generic_allocation snd_soc_acpi crc8 rtw_8821ce(OE) soundwire_bus rtw_8821c(OE) bnep snd_soc_sdca intel_pmc_bxt intel_telemetry_pltdrv snd_soc_avs intel_punit_ipc rtw_pci(OE) intel_telemetry_core x86_pkg_temp_thermal snd_soc_hda_codec intel_powerclamp coretemp snd_hda_ext_core rtw_core(OE) snd_hda_codec_hdmi kvm_intel snd_soc_core snd_hda_codec_realtek snd_compress snd_hda_codec_generic ac97_bus snd_pcm_dmaengine
Aug 13 23:59:44 touch-mini kernel:  snd_hda_scodec_component kvm snd_hda_intel mei_pxp mei_hdcp intel_rapl_msr snd_intel_dspcfg irqbypass mac80211 btusb btrtl btintel snd_intel_sdw_acpi btbcm snd_hda_codec rapl libarc4 intel_cstate snd_hda_core btmtk i2c_i801 processor_thermal_device_pci_legacy i2c_smbus lpc_ich snd_hwdep binfmt_misc processor_thermal_device bluetooth snd_seq processor_thermal_wt_hint processor_thermal_rfim cfg80211 idma64 processor_thermal_rapl snd_seq_device snd_pcm intel_rapl_common processor_thermal_wt_req rfkill snd_timer processor_thermal_power_floor vfat fat mei_me snd processor_thermal_mbox soundcore mei intel_soc_dts_iosf mxc4005 industrialio_triggered_buffer goodix_ts kfifo_buf industrialio intel_pmc_core pmt_telemetry soc_button_array pmt_class int3400_thermal intel_hid intel_vsec acpi_thermal_rel int3403_thermal int340x_thermal_zone sparse_keymap joydev fuse loop zram lz4hc_compress lz4_compress i915 mmc_block rpmb_core polyval_generic sdhci_pci i2c_algo_bit drm_buddy sdhci_uhs2 ghash_clmulni_intel ttm sdhci
Aug 13 23:59:44 touch-mini kernel:  sha512_ssse3 cqhci drm_display_helper sha256_ssse3 sha1_ssse3 mmc_core wdat_wdt spi_pxa2xx_platform dw_dmac spi_pxa2xx_core cec r8169 realtek video wmi pinctrl_geminilake serio_raw hid_logitech_hidpp uas usb_storage hid_logitech_dj i2c_dev nfnetlink
Aug 13 23:59:44 touch-mini kernel: CPU: 2 UID: 0 PID: 0 Comm: swapper/2 Tainted: G        W  OE       6.15.9-201.fc42.x86_64 #1 PREEMPT(lazy) 
Aug 13 23:59:44 touch-mini kernel: Tainted: [W]=WARN, [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
Aug 13 23:59:44 touch-mini kernel: Hardware name: Default string To be filled by O.E.M./To be filled by O.E.M., BIOS 1.18 08/08/2023
Aug 13 23:59:44 touch-mini kernel: Call Trace:
Aug 13 23:59:44 touch-mini kernel:  <IRQ>
Aug 13 23:59:44 touch-mini kernel:  dump_stack_lvl+0x5d/0x80
Aug 13 23:59:44 touch-mini kernel:  __schedule_bug.cold+0x42/0x4e
Aug 13 23:59:44 touch-mini kernel:  __schedule+0x61c/0x7a0
Aug 13 23:59:44 touch-mini kernel:  schedule+0x27/0x80
Aug 13 23:59:44 touch-mini kernel:  schedule_preempt_disabled+0x15/0x30
Aug 13 23:59:44 touch-mini kernel:  __mutex_lock.constprop.0+0x7bc/0x880
Aug 13 23:59:44 touch-mini kernel:  rtw_led_set+0x25/0x50 [rtw_core]
Aug 13 23:59:44 touch-mini kernel:  ? __pfx_led_timer_function+0x10/0x10
Aug 13 23:59:44 touch-mini kernel:  led_timer_function+0x70/0x100
Aug 13 23:59:44 touch-mini kernel:  call_timer_fn+0x26/0x120
Aug 13 23:59:44 touch-mini kernel:  ? __pfx_led_timer_function+0x10/0x10
Aug 13 23:59:44 touch-mini kernel:  __run_timers+0x1f5/0x280
Aug 13 23:59:44 touch-mini kernel:  run_timer_softirq+0x8c/0xf0
Aug 13 23:59:44 touch-mini kernel:  ? irqtime_account_irq+0x3c/0xc0
Aug 13 23:59:44 touch-mini kernel:  handle_softirqs+0xed/0x340
Aug 13 23:59:44 touch-mini kernel:  __irq_exit_rcu+0xcb/0xf0
Aug 13 23:59:44 touch-mini kernel:  sysvec_apic_timer_interrupt+0x71/0x90
Aug 13 23:59:44 touch-mini kernel:  </IRQ>
Aug 13 23:59:44 touch-mini kernel:  <TASK>
Aug 13 23:59:44 touch-mini kernel:  asm_sysvec_apic_timer_interrupt+0x1a/0x20
Aug 13 23:59:44 touch-mini kernel: RIP: 0010:cpuidle_enter_state+0xcc/0x660
Aug 13 23:59:44 touch-mini kernel: Code: 00 00 e8 57 f4 f5 fe e8 12 ee ff ff 49 89 c4 0f 1f 44 00 00 31 ff e8 f3 12 f4 fe 45 84 ff 0f 85 02 02 00 00 fb 0f 1f 44 00 00 <85> ed 0f 88 d3 01 00 00 4c 63 f5 49 83 fe 0a 0f 83 9f 04 00 00 49
Aug 13 23:59:44 touch-mini kernel: RSP: 0018:ffffd1f6c00ebe50 EFLAGS: 00000246
Aug 13 23:59:44 touch-mini kernel: RAX: ffff8d500cf70000 RBX: ffff8d4fb7d41040 RCX: 0000000000000000
Aug 13 23:59:44 touch-mini kernel: RDX: 00000030a35a810f RSI: 00000000401a41a4 RDI: 0000000000000000
Aug 13 23:59:44 touch-mini kernel: RBP: 0000000000000004 R08: fffffff40ed8a5a0 R09: ffff8d4fb7d2c7e0
Aug 13 23:59:44 touch-mini kernel: R10: 0000003c94b7892a R11: 0000000000000000 R12: 00000030a35a810f
Aug 13 23:59:44 touch-mini kernel: R13: ffffffffaa10c580 R14: 0000000000000004 R15: 0000000000000000
Aug 13 23:59:44 touch-mini kernel:  cpuidle_enter+0x31/0x50
Aug 13 23:59:44 touch-mini kernel:  cpuidle_idle_call+0xf5/0x160
Aug 13 23:59:44 touch-mini kernel:  do_idle+0x78/0xd0
Aug 13 23:59:44 touch-mini kernel:  cpu_startup_entry+0x29/0x30
[... 296 lines cut ...]
Aug 14 00:01:02 touch-mini kernel:  <TASK>
Aug 14 00:01:02 touch-mini kernel:  asm_sysvec_apic_timer_interrupt+0x1a/0x20
Aug 14 00:01:02 touch-mini kernel: RIP: 0010:cpuidle_enter_state+0xcc/0x660
Aug 14 00:01:02 touch-mini kernel: Code: 00 00 e8 57 f4 f5 fe e8 12 ee ff ff 49 89 c4 0f 1f 44 00 00 31 ff e8 f3 12 f4 fe 45 84 ff 0f 85 02 02 00 00 fb 0f 1f 44 00 00 <85> ed 0f 88 d3 01 00 00 4c 63 f5 49 83 fe 0a 0f 83 9f 04 00 00 49
Aug 14 00:01:02 touch-mini kernel: RSP: 0018:ffffd1f6c00f3e50 EFLAGS: 00000246
Aug 14 00:01:02 touch-mini kernel: RAX: ffff8d500cff0000 RBX: ffff8d4fb7dc1040 RCX: 0000000000000000
Aug 14 00:01:02 touch-mini kernel: RDX: 00000031bbd8aa8a RSI: 00000000401a41a4 RDI: 0000000000000000
Aug 14 00:01:02 touch-mini kernel: RBP: 0000000000000004 R08: fffffff40ed8a5a0 R09: ffff8d4fb7dac7e0
Aug 14 00:01:02 touch-mini kernel: R10: 0000003dad35b2a5 R11: 0000000000000000 R12: 00000031bbd8aa8a
Aug 14 00:01:02 touch-mini kernel: R13: ffffffffaa10c580 R14: 0000000000000004 R15: 0000000000000000
Aug 14 00:01:02 touch-mini kernel:  ? sched_balance_update_blocked_averages+0xdb/0x1a0
Aug 14 00:01:02 touch-mini kernel:  cpuidle_enter+0x31/0x50
Aug 14 00:01:02 touch-mini kernel:  cpuidle_idle_call+0xf5/0x160
Aug 14 00:01:02 touch-mini kernel:  do_idle+0x78/0xd0
Aug 14 00:01:02 touch-mini kernel:  cpu_startup_entry+0x29/0x30
Aug 14 00:01:02 touch-mini kernel:  start_secondary+0x126/0x170
Aug 14 00:01:02 touch-mini kernel:  common_startup_64+0x13e/0x141
Aug 14 00:01:02 touch-mini kernel:  </TASK>
Aug 14 00:01:02 touch-mini kernel: INFO: NMI handler (nmi_cpu_backtrace_handler) took too long to run: 1.124 msecs
Aug 14 00:01:14 touch-mini systemd[1]: systemd-localed.service: State 'stop-sigterm' timed out. Aborting.
Aug 14 00:01:14 touch-mini systemd[1]: systemd-localed.service: Killing process 1471 (systemd-localed) with signal SIGABRT.
Aug 14 00:01:27 touch-mini kernel: watchdog: BUG: soft lockup - CPU#3 stuck for 93s! [swapper/3:0]
Aug 14 00:01:27 touch-mini kernel: CPU#3 Utilization every 4s during lockup:
Aug 14 00:01:27 touch-mini kernel:         #1:   0% system,        100% softirq,          0% hardirq,          0% idle
```

dubhater, 2025-08-17:
Does it help?

```diff
diff --git a/led.c b/led.c
index 7f9ace351a5b..4cc62e49d167 100644
--- a/led.c
+++ b/led.c
@@ -6,8 +6,8 @@
 #include "debug.h"
 #include "led.h"
 
-static void rtw_led_set(struct led_classdev *led,
-			enum led_brightness brightness)
+static int rtw_led_set(struct led_classdev *led,
+		       enum led_brightness brightness)
 {
 	struct rtw_dev *rtwdev = container_of(led, struct rtw_dev, led_cdev);
 
@@ -16,12 +16,6 @@ static void rtw_led_set(struct led_classdev *led,
 	rtwdev->chip->ops->led_set(led, brightness);
 
 	mutex_unlock(&rtwdev->mutex);
-}
-
-static int rtw_led_set_blocking(struct led_classdev *led,
-				enum led_brightness brightness)
-{
-	rtw_led_set(led, brightness);
 
 	return 0;
 }
@@ -46,10 +40,7 @@ void rtw_led_init(struct rtw_dev *rtwdev)
 	if (!rtwdev->chip->ops->led_set)
 		return;
 
-	if (rtw_hci_type(rtwdev) == RTW_HCI_TYPE_PCIE)
-		led->brightness_set = rtw_led_set;
-	else
-		led->brightness_set_blocking = rtw_led_set_blocking;
+	led->brightness_set_blocking = rtw_led_set;
 
 	snprintf(rtwdev->led_name, sizeof(rtwdev->led_name),
 		 "rtw88-%s", dev_name(rtwdev->dev));
```

daniviga, 2025-08-19:
Thank you @dubhater , I applied the patch over 52072d8 and the system is stable so far

dubhater, 2025-08-20:
@daniviga @utzig That's good. Thanks for testing.

But I'm curious, why are the two of you using this repository instead of the driver built into the kernel?

dubhater, 2025-08-20:
@utzig Is your card RTL8821CE?