lwfinger/rtw88 #390: Geoclue causes network jitter on RTL8814au

haxtaxed, 2025-09-09:
Geoclue causes network jitter. Here's me pinging 1.1.1.1 with geoclue running:
```
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=60 time=1.98 ms
64 bytes from 1.1.1.1: icmp_seq=2 ttl=60 time=31.8 ms
64 bytes from 1.1.1.1: icmp_seq=3 ttl=60 time=63.1 ms
64 bytes from 1.1.1.1: icmp_seq=4 ttl=60 time=92.6 ms
64 bytes from 1.1.1.1: icmp_seq=5 ttl=60 time=134 ms
64 bytes from 1.1.1.1: icmp_seq=6 ttl=60 time=2.82 ms
64 bytes from 1.1.1.1: icmp_seq=7 ttl=60 time=2.28 ms
64 bytes from 1.1.1.1: icmp_seq=8 ttl=60 time=26.4 ms
64 bytes from 1.1.1.1: icmp_seq=9 ttl=60 time=1.79 ms
64 bytes from 1.1.1.1: icmp_seq=10 ttl=60 time=3.92 ms
64 bytes from 1.1.1.1: icmp_seq=11 ttl=60 time=2.33 ms
64 bytes from 1.1.1.1: icmp_seq=12 ttl=60 time=1.89 ms
^[[A64 bytes from 1.1.1.1: icmp_seq=13 ttl=60 time=2.56 ms
64 bytes from 1.1.1.1: icmp_seq=14 ttl=60 time=3.14 ms
^[
64 bytes from 1.1.1.1: icmp_seq=15 ttl=60 time=35.5 ms
64 bytes from 1.1.1.1: icmp_seq=16 ttl=60 time=83.2 ms
64 bytes from 1.1.1.1: icmp_seq=17 ttl=60 time=122 ms
64 bytes from 1.1.1.1: icmp_seq=18 ttl=60 time=3.28 ms
64 bytes from 1.1.1.1: icmp_seq=19 ttl=60 time=2.57 ms
64 bytes from 1.1.1.1: icmp_seq=20 ttl=60 time=1.33 ms
64 bytes from 1.1.1.1: icmp_seq=21 ttl=60 time=2.55 ms
64 bytes from 1.1.1.1: icmp_seq=22 ttl=60 time=3.45 ms
64 bytes from 1.1.1.1: icmp_seq=23 ttl=60 time=2.45 ms
64 bytes from 1.1.1.1: icmp_seq=24 ttl=60 time=2.37 ms
64 bytes from 1.1.1.1: icmp_seq=25 ttl=60 time=1.64 ms
64 bytes from 1.1.1.1: icmp_seq=26 ttl=60 time=2.31 ms
^C
--- 1.1.1.1 ping statistics ---
26 packets transmitted, 26 received, 0% packet loss, time 25040ms
rtt min/avg/max/mdev = 1.327/24.345/133.540/39.180 ms```
```

And the behaviour after stopping geoclue:
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=60 time=3.14 ms
64 bytes from 1.1.1.1: icmp_seq=2 ttl=60 time=1.35 ms
64 bytes from 1.1.1.1: icmp_seq=3 ttl=60 time=2.17 ms
64 bytes from 1.1.1.1: icmp_seq=4 ttl=60 time=1.41 ms
64 bytes from 1.1.1.1: icmp_seq=5 ttl=60 time=2.88 ms
64 bytes from 1.1.1.1: icmp_seq=6 ttl=60 time=1.54 ms
64 bytes from 1.1.1.1: icmp_seq=7 ttl=60 time=2.91 ms
64 bytes from 1.1.1.1: icmp_seq=8 ttl=60 time=1.80 ms
64 bytes from 1.1.1.1: icmp_seq=9 ttl=60 time=2.02 ms
64 bytes from 1.1.1.1: icmp_seq=10 ttl=60 time=2.00 ms
64 bytes from 1.1.1.1: icmp_seq=11 ttl=60 time=3.29 ms
64 bytes from 1.1.1.1: icmp_seq=12 ttl=60 time=3.10 ms
64 bytes from 1.1.1.1: icmp_seq=13 ttl=60 time=1.97 ms
64 bytes from 1.1.1.1: icmp_seq=14 ttl=60 time=1.96 ms
64 bytes from 1.1.1.1: icmp_seq=15 ttl=60 time=1.86 ms
64 bytes from 1.1.1.1: icmp_seq=16 ttl=60 time=3.44 ms
64 bytes from 1.1.1.1: icmp_seq=17 ttl=60 time=1.51 ms
64 bytes from 1.1.1.1: icmp_seq=18 ttl=60 time=1.45 ms
64 bytes from 1.1.1.1: icmp_seq=19 ttl=60 time=1.67 ms
64 bytes from 1.1.1.1: icmp_seq=20 ttl=60 time=2.66 ms
64 bytes from 1.1.1.1: icmp_seq=21 ttl=60 time=3.40 ms
64 bytes from 1.1.1.1: icmp_seq=22 ttl=60 time=1.61 ms
64 bytes from 1.1.1.1: icmp_seq=23 ttl=60 time=1.45 ms
64 bytes from 1.1.1.1: icmp_seq=24 ttl=60 time=2.20 ms
64 bytes from 1.1.1.1: icmp_seq=25 ttl=60 time=3.02 ms
64 bytes from 1.1.1.1: icmp_seq=26 ttl=60 time=1.63 ms
64 bytes from 1.1.1.1: icmp_seq=27 ttl=60 time=1.98 ms
64 bytes from 1.1.1.1: icmp_seq=28 ttl=60 time=2.81 ms
64 bytes from 1.1.1.1: icmp_seq=29 ttl=60 time=2.49 ms
64 bytes from 1.1.1.1: icmp_seq=30 ttl=60 time=3.31 ms
64 bytes from 1.1.1.1: icmp_seq=31 ttl=60 time=2.41 ms
64 bytes from 1.1.1.1: icmp_seq=32 ttl=60 time=1.98 ms
64 bytes from 1.1.1.1: icmp_seq=33 ttl=60 time=2.04 ms
64 bytes from 1.1.1.1: icmp_seq=34 ttl=60 time=3.86 ms
64 bytes from 1.1.1.1: icmp_seq=35 ttl=60 time=2.42 ms
64 bytes from 1.1.1.1: icmp_seq=36 ttl=60 time=3.46 ms
^C
--- 1.1.1.1 ping statistics ---
36 packets transmitted, 36 received, 0% packet loss, time 35054ms
rtt min/avg/max/mdev = 1.350/2.337/3.859/0.708 ms```

dubhater, 2025-09-09:
I wonder if it's scanning every 10 seconds? What does ``iw event -T`` print while geoclue is running and you're pinging something?

haxtaxed, 2025-09-10:
> I wonder if it's scanning every 10 seconds? What does `iw event -T` print while geoclue is running and you're pinging something?

iw event -T stays blank

The jitter did appear to happen at intervals before I masked the geoclue service.

haxtaxed, 2025-09-10:
iw event -T being blank was a small issue on my side.

the stuttering is starts a few seconds after a scanning event starts:

```[2025-09-11 00:36:49.439017]: wlp43s0f3u2 (phy #0): scan started```

and it stops immedietly after it's finished:

```[2025-09-11 00:37:02.030044]: wlp43s0f3u2 (phy #0): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, "Wu Tang LAN" ""```

closed 2025-09-10 by haxtaxed