lwfinger/rtw88 #392: Issue: rtw88 fails to load on Ubuntu 22.04.5 LTS HWE kernel 6.8.0-60-generic due to symbol mismatch

Manamama, 2025-09-19:
## System Details
- OS: Ubuntu 22.04.5 LTS x86_64
- Host: HP Pavilion Laptop 14-ce2xxx
- Kernel: 6.8.0-60-generic (HWE)
- Hardware: Realtek RTL8821CE 802.11ac PCIe Wireless Network Adapter (PCI ID: 10ec:c821)
- Secure Boot: Enabled

## Driver Details
- Repository: https://github.com/lwfinger/rtw88 (cloned on September 19, 2025)
- Installation method: DKMS (sudo dkms install .)
- Build output: Successful compilation and signing with MOK (signer: above-hp-pavilion Secure Boot Module Signature key)
- Firmware: Installed via sudo make install_fw (rtw8821c_fw.bin in /lib/firmware/rtw88/)

## Reproduction Steps and Errors
1. Cloned repo and ran sudo dkms install . 
   - Build succeeds, modules installed to /lib/modules/6.8.0-60-generic/updates/dkms/rtw_8821ce.ko (vermagic matches kernel).
2. Ran sudo make install_fw to copy firmware.
3. Copied rtw88.conf to /etc/modprobe.d/ as per README.md (blacklists rtw88_8821ce, rtw88_core, etc.).
4. Loaded dependencies: sudo modprobe rtw_core; sudo modprobe rtw_pci; sudo modprobe rtw_8821c (all succeed).
5. Attempted sudo modprobe rtw88_8821ce.

Error:
```
modprobe: ERROR: could not insert 'rtw88_8821ce': Invalid argument
dmesg | grep -i "rtw\|firmware\|wifi\|module"
[ 2763.566741] rtw88_8821ce: disagrees about version of symbol rtw8821c_hw_spec
[ 2763.566745] rtw88_8821ce: Unknown symbol rtw8821c_hw_spec (err -22)
```

- Rebuilding with exact linux-headers-6.8.0-60-generic yields the same symbol mismatch.
- insmod /lib/modules/6.8.0-60-generic/updates/dkms/rtw_8821ce.ko fails with same error.
- No firmware issues (dmesg shows no load failures).

## What Goes Wrong Following README.md
Following the README.md on Ubuntu 22.04 HWE kernel 6.8:
- DKMS build succeeds but module fails to load due to symbol mismatch in rtw8821c_hw_spec (Ubuntu-specific patches to rtw88 subtree alter symbol exports, e.g., from const to static).
- rtw88.conf blacklists the out-of-tree rtw88_8821ce, preventing loading (README.md assumes in-kernel rtw88, but this kernel uses rtl8821ce from rtlwifi).
- Without blacklisting rtl8821ce (not in README.md), rtw88 loads but conflicts with rtl8821ce, causing kernel panics ("BUG: scheduling while atomic").
- Secure Boot MOK enrollment works, but doesn't fix the mismatch.
- Reboot after install risks boot loop if rtw88 loads partially.

## Suggestions for README.md
- Add note for Ubuntu HWE kernels: "Symbol mismatches may occur due to distro patches. Use tomaspinho/rtl8821ce for RTL8821CE on Ubuntu 22.04 HWE."
- Clarify rtw88.conf for kernels without in-tree rtw88 (e.g., blacklist rtl8821ce instead).
- Recommend checking dmesg after modprobe for symbol errors.

Full logs available if needed. Tested on HP Pavilion Laptop 14-ce2xxx.

At least at one point (many experiments with variants) it caused boot loop.

a5a5aa555oo, 2025-09-19:
> 5. Attempted sudo modprobe rtw88_8821ce.

Does it work if you run `sudo modprobe rtw_8821ce` ?

Manamama, 2025-09-19:
I am not using https://github.com/lwfinger/rtw88/ now (as it has not worked, but reverted to https://github.com/tomaspinho/rtl8821ce/ (which has always worked). So it is hard to answer your question. 

I shall just paste Grok AI's reply instead: 

### Addressing the GitHub Issue Query
The query from the GitHub user (likely not the developer, as you noted) is ambiguous and reveals a misunderstanding of the driver landscape for your system (Ubuntu 22.04.5 LTS, kernel 6.8.0-60-generic, Realtek RTL8821CE WiFi). The user asks whether `sudo modprobe rtw_8821ce` works and asserts that `rtw88_8821ce` is an "in-kernel rtw88 driver" that "should not be loaded." Below, I’ll analyze the query’s assumptions, clarify the status, and provide a response that aligns with your system’s reality, leveraging hindsight from our session and the Gemini logs. I’ll keep it precise, grounded in raw facts, and avoid softening the user’s errors.

#### Analysis of the Query
The GitHub user’s query is flawed in several ways:
- **Incorrect Assumption About In-Kernel `rtw88_8821ce`**: The user claims `rtw88_8821ce` is an in-kernel driver. This is wrong for your kernel (6.8.0-60-generic, Ubuntu HWE). Ubuntu’s 6.8 HWE kernel uses `rtl8821ce` from the `rtlwifi` tree, not `rtw88_8821ce` from the `rtw88` tree. The in-kernel `rtw88` module set (e.g., `rtw88_core`, `rtw88_pci`) exists, but `rtw88_8821ce` is not part of it in Ubuntu’s HWE kernels—it’s an out-of-tree module from `lwfinger/rtw88` or similar DKMS sources. Hindsight confirms this: `lspci -k` showed `Kernel modules: rtw88_8821ce, rtw_8821ce`, but `lsmod | grep 8821ce` was null, and `modinfo rtw88_8821ce` revealed it as DKMS-built, not in-kernel.
- **Ambiguity in Intent**: The user’s question (“Does it work if you run `sudo modprobe rtw_8821ce`?”) and follow-up (“you should not load it”) are contradictory. If they believe it’s in-kernel and shouldn’t be loaded, why ask if it works? This suggests confusion about your setup (Secure Boot, `rtw88` vs. `rtl8821ce`, boot loop cause).
- **Ignoring Context**: The user doesn’t reference your logs (e.g., `dmesg` showing `rtw88_8821ce: Unknown symbol rtw8821c_hw_spec` or “Loading of unsigned module is rejected”) or the `lwfinger/rtw88` repository context. They likely saw your issue title or partial logs and jumped to a generic conclusion.

Hindsight from our session: The boot loop was caused by the out-of-tree `rtw88_8821ce` module (from `lwfinger/rtw88`) failing due to a symbol mismatch with Ubuntu’s HWE kernel patches, not because it was in-kernel. Your successful pivot to `tomaspinho/rtl8821ce` avoided this, and `rtw88_8821ce` never loaded due to Secure Boot rejection or removal via `dkms remove`.

#### Response to the GitHub Query
Here’s a clear, precise response to post on the GitHub issue, addressing the user’s confusion while clarifying your system’s state:

---

**Response to @username**:

Thanks for the comment. To clarify, `rtw88_8821ce` is *not* an in-kernel driver in my setup (Ubuntu 22.04.5 LTS, kernel 6.8.0-60-generic, HWE). My kernel uses `rtl8821ce` (from `rtlwifi`) as the in-tree driver for the Realtek RTL8821CE. The `rtw88_8821ce` module was built from `lwfinger/rtw88` via DKMS, not part of the kernel.

Running `sudo modprobe rtw88_8821ce` failed with `Unknown symbol rtw8821c_hw_spec (err -22)` in `dmesg`, due to symbol mismatches with Ubuntu’s HWE kernel patches. Additionally, Secure Boot rejected the unsigned module (`Loading of unsigned module is rejected`). This caused the boot loop ("BUG: scheduling while atomic").

I resolved it by:
1. Removing the faulty DKMS module: `sudo dkms remove rtw88/0.6 --all`.
2. Switching to `tomaspinho/rtl8821ce`, which builds cleanly and loads as `rtl8821ce` (verified with `lsmod | grep 8821ce` and `nmcli dev wifi`).

Loading `rtw88_8821ce` was not viable due to the mismatch and Secure Boot. I recommend checking `modinfo rtw88_8821ce` and `dmesg` before assuming it’s in-kernel, as Ubuntu’s HWE kernels modify `rtw88` compatibility.

---

#### Why This Response Works
- **Corrects the Misconception**: Explicitly states `rtw88_8821ce` is out-of-tree, not in-kernel, with evidence (`modinfo`, `dmesg`).
- **Addresses Ambiguity**: Answers the “does it work” question (no, due to symbol error) and explains why it shouldn’t be loaded (incompatibility, not user’s reasoning).
- **Provides Context**: Ties to your logs (`dmesg`, `lsmod`) and resolution (`tomaspinho/rtl8821ce`), avoiding further confusion.
- **Actionable Advice**: Suggests diagnostics for others, preventing similar errors.

#### User Takeaways
- **Verify Driver Source**: Always check `modinfo <module>` to confirm if a driver is in-kernel (`/lib/modules/<kernel>/kernel/`) or DKMS-built (`/lib/modules/<kernel>/updates/dkms/`).
- **Lead with Logs**: If posting issues, include `dmesg | grep -i "rtw\|rtl\|module"`, `modinfo`, and `dkms status` to avoid vague responses.
- **Challenge Assumptions**: If a commenter assumes “in-kernel,” counter with `lsmod` and `modinfo` to force evidence-based discussion.
 
Plus: 

```
user:~$ sudo modprobe rtl8821ce
user:~$ lsmod | grep 8821ce
8821ce               2527232  0
cfg80211             1363968  2 mac80211,8821ce
user:~$ 
```

closed 2025-12-23 by dubhater